pyocd/pyOCD #617: Question: why the get_probe_with_id() in pyocd/probe/cmsis_dap_probe.py doesn't return None?

anthrax-0, 2019-04-15:
I don't fully understand the usage of the DebugProbeAggregator.get_probe_with_id(board_id), for example, in the test/automated_test.py:164.
 As it is now, when I have only st-link probes connected and override the target in the pyocd.yaml, get_probe_with_id() still returns the CMSISDAPProbe probe even if it found no interface and has no probe with such id.
 Shouldn't it return None in that case?

anthrax-0, 2019-04-16:
Examples of the output:
```\pyOCD>pyocd list
## => Board Name | Unique ID
-- -- ----------------------
 0 => STM32 STLinkV2-1 | 066AFF393430533457055540
 1 => STM32 STLinkV2 | 53FF6D067865495422580467
```

```
>>> from pyocd.probe.aggregator import DebugProbeAggregator
>>> probe = DebugProbeAggregator.get_probe_with_id('53FF6D067865495422580467')
>>> print(probe)
<CMSISDAPProbe@19d76eabdd8  >
>>> print(probe.unique_id)
53FF6D067865495422580467
```

For me, these lines are the one I don't understand - [here](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py#L524) and [here](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/probe/cmsis_dap_probe.py#L80). 
Why if no interface is found pyOCD still creates some objects?

flit, 2019-04-16:
This is definitely a bug.

The code in `DAPAccessCMSISDAP.__init__()` is mostly correct. The constructor should only be called if it is known that a device exists. But it does need to handle the case where the device has disappeared between first identifying it and calling the constructor.

The primary bug here is in `CMSISDAPProbe.get_probe_with_id()`, as you have discovered.

Thanks for reporting this! I'll fix it right away.

PR #621 (2019-04-18): Fix get_probe_with_id() for CMSIS-DAP
Calling `CMSISDAPProbe.get_probe_with_id()` would always return an object even if there was no matching probe. This also caused the same method on `DebugProbeAggregator` to always return a `CMSISDAPProbe` even if the ID only matched an STLink.

Underlying this was original behaviour of `DAPAccessCMSISDAP`, where you could create an instance with any unique ID. It would only fail when you called `open()`. This patch changes this behaviour so it fails early. Attempting to construct a `DAPAccessCMSISDAP` with an unique ID that does not match a CMSIS-DAP inter [... cut]
commit d7e93275 (2019-04-18): Fix get_probe_with_id() for CMSIS-DAP.

- Calling CMSISDAPProbe.get_probe_with_id() would always return an object
  even if there was no matching probe. This also caused the same method
  on DebugProbeAggregator to always return a CMSISDAPProbe even if the
  ID only matched an STLink.
- This fixes #617.
- DAPAccessCMSISDAP constructor will raise DeviceError if it is given a
  unique ID and no interface, and it can't find an interface with the ID.
- DAPAccessCMSISDAP.get_device() will return None if no matching interface
  can be found. Previously it would always return an object, and the failu [... cut]
closed 2019-04-18 by flit