pyocd/pyOCD #623: STLink exception error while testing new target added

rgf97, 2019-04-19:
Hi, i added new stm32l476xg target. Everything seems OK and the Target is recognized:

pyocd-gdbserver -l
## => Board Name | Unique ID
-- -- ----------------------
 0 => STM32 STLinkV2-1 | 066EFF535550755187241812

(Note: I used flashalgo)

But, when i try one test, i get this error:

/pyOCD/test$ python ./flash_test.py
INFO:board:Target type is cortex_m
INFO:board:Target type is cortex_m
Exception STLink error (20): DP wait when testing board 066EFF535550755187241812
Traceback (most recent call last):
  File "./flash_test.py", line 109, in run
    result = self.test_function(board.unique_id)
  File "./flash_test.py", line 121, in flash_test
    with ConnectHelper.session_with_chosen_probe(board_id=board_id, **get_session_options()) as session:
  File "/home/osboxes/stage/pyOCD/pyocd/core/helpers.py", line 199, in session_with_chosen_probe
    session.open()
  File "/home/osboxes/stage/pyOCD/pyocd/core/session.py", line 281, in open
    self._board.init()
  File "/home/osboxes/stage/pyOCD/pyocd/board/board.py", line 66, in init
    self.target.init()
  File "/home/osboxes/stage/pyOCD/pyocd/core/coresight_target.py", line 152, in init
    seq.invoke()
  File "/home/osboxes/stage/pyOCD/pyocd/utility/sequencer.py", line 189, in invoke
    resultSequence = call()
  File "/home/osboxes/stage/pyOCD/pyocd/coresight/dap.py", line 98, in init
    self.link.connect()
  File "/home/osboxes/stage/pyOCD/pyocd/probe/stlink_probe.py", line 90, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "/home/osboxes/stage/pyOCD/pyocd/probe/stlink/stlink.py", line 215, in enter_debug
    self._check_status(response)
  File "/home/osboxes/stage/pyOCD/pyocd/probe/stlink/stlink.py", line 251, in _check_status
    raise self._ERROR_CLASSES[status](error_message)
pyocd.core.exceptions.TransferTimeoutError: STLink error (20): DP wait

------ Analyzer Performance ------
Target    Analyzer    Rate              Time              

------ Test Rate ------
Target    Chip Erase          Page Erase          Page Erase (Same data)

cortex_m  Fail                Fail                Fail      

Any idea? Thx

flit, 2019-04-19:
Hi @rgf97, it looks like the command to set the STLink to SWD mode is not succeeding. (I haven't seen this kind of issue before.) The "DP wait" error indicates that basic SWD communication to the DP isn't working; it gets an SWD "wait" protocol response. Although, I don't know what the STLink firmware is actually doing here.

First thing to check is whether you have the latest STLink firmware, V2J33M25 as of this writing. And I guess you're using the latest pyOCD code on master?

If it still happens, then it could be something with the target chip. Could it be in a bad state? Unfortunately, pyOCD currently doesn't support a "connect under reset" feature.

Btw, you will need to set the target type to use your new target. Either use the `--target` command line argument, or `target_override` user option (best set in a `pyocd.yaml` config file). When running the functional tests you have to use a config file to set the target type, as well as the `test_binary` option (set to just the name, not a path, of a file in the `binaries/` directory).

Also, you should try out the CMSIS-Pack support. Just run `pyocd pack -i stm32l476` to add that target. There are definitely some areas to improve and fix 😄 . And new builtin targets are always welcome.

rgf97, 2019-04-19:
@flit Thank you. I will try all your suggestions and give you feedbacks next week. (Yes it is the latest pyocd)

rgf97, 2019-04-23:
@flit Hi,  This error was caused by a specific error that I faced and solved in past days. I knew it was redundant, but I didn't notice it was at this point. I don't know why, but in my virtual machine (Linux VM on Windows), I need to flash (an application) to the board twice to make it work. And then, I can debug. I thought I needed to do that each time I launch the VM instead of each time I connect the board. So, flash an application twice into the board before making my tests solved this error. But i have another issue ... (Check New issue)

rgf97, 2019-04-23:
When i make test on a new board added, i get attribute or syntax errors like this: 

[image]

What did I did wrong?

flit, 2019-04-23:
We've previously noticed issues with running in a VM, specifically VirtualBox, where USB packets get lost. I've taken a note to add something to the docs about this.

closed 2019-04-24 by rgf97