pyocd/pyOCD #624: Loading the .elf file will give an warning，file could not be loaded

PpgpOs, 2019-04-23:
Use
DAPlink v1
STM32F407ZGT6

Python 3.7
MSYS2  
User@DESKTOP-U3LLAHV MINGW32 /c/Users/User/Desktop/MyProject/Stm32/Stm32f407zg/Project1
PyOCD 0.19
```
# pyocd gdbserver
0001226:INFO:board:Target type is cortex_m
0001256:INFO:dap:DP IDR = 0x2ba01477
0001273:INFO:dap:AP#0 IDR = 0x24770011
0001283:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=411)
0001293:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001299:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0001305:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0001312:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0001318:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0001325:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001329:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0001337:INFO:cortex_m:FPU present: FPv4-SP
0001346:INFO:dwt:4 hardware watchpoints
0001351:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001377:INFO:server:Semihost server started on port 4444
0001377:INFO:gdbserver:GDB server started on port 3333
```
The original Led flashing program stopped running，GPIO state retention
`# arm-none-eabi-gdb build/Project1.elf`
Then hit the Enter key twice to skip the message.
...
```
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/Project1.elf...done.
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
warning: while parsing target memory map (at line 3): Required element <memory> is missing
0x080012f6 in ?? ()
```
Report warning
```
(gdb) load
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0xe98 lma 0x8000188
Loading section .rodata, size 0x10 lma 0x8001020
Loading section .ARM, size 0x8 lma 0x8001030
Loading section .init_array, size 0x4 lma 0x8001038
Loading section .fini_array, size 0x4 lma 0x800103c
Loading section .data, size 0xc lma 0x8001040
Start address 0x8000f5c, load size 4172
Transfer rate: 18 KB/sec, 521 bytes/write.
(gdb) monitor reset
(gdb)
```
MCU not respond,Unplug the DAP cable and plug it back in,MCU run,but the original program is still running.

It was only done yesterday, it took a few hours... I can’t use it today...

flit, 2019-04-24:
Hi @PpgpOs, you need to set the target type. You can see the first line in the `pyocd gdbserver` log where it says "Target type is cortex_m", indicating it's using the generic `cortex_m` target type (no flash programming or memory map). Currently, pyocd does not support auto-detection of the target with STLink probes (this will be merged in soon).

Since the STM32F103ZE target is not builtin to pyOCD, you'll need to use a CMSIS-Pack to enable support. With pyOCD 0.19.0 you can run this:

```
pyocd pack -u -i stm32f103ze
```

Then run this to list the supported targets:

```
pyocd list --targets
```

You'll see a target with the name `stm32f103ze` in the list. You can now pass this target name for the `--target` command line argument to `pyocd gdbserver` in order to enable flash programming and the memory map.

(I'm going to be working on much better documentation soon!)

flit, 2019-04-24:
Just noticed you first mentioned the F407… The same process applies for enabling support for this target type.

flit, 2019-04-24:
Fyi, the [target support documentation](https://github.com/mbedmicro/pyOCD/blob/master/docs/target_support.md) covers much of this information and more detail in certain areas.

PpgpOs, 2019-04-25:
@flit 
I probably understand，but...
```
# pyocd pack --find stm32f103ze
usage: pyocd [-h] [-V] [--help-options]  ...
pyocd: error: argument : invalid choice: 'pack' (choose from 'commander', 'cmd', 'erase', 'flash', 'gdbserver', 'gdb', 'json', 'list')
```
The same result if you type pyocd pack -u -i stm32f103ze

PpgpOs, 2019-04-25:
@flit  Hi
`pyocd gdbserver --pack C:/Users/User/Desktop/Keil.STM32F1xx_DFP.2.3.0.pack -t stm32f103ze` is that
Run it successfully!
Thank you!

Why do you say that the `pack` command is a `pyocd pack...`?

flit, 2019-04-25:
Ooops, I'm sorry! The `pack` command hasn't yet been released, it's only on the master branch. I'll be releasing version 0.20.0 quite soon that includes it. My apologies for the confusion!

rgf97, 2019-04-25:
Oh Ok. I was also confused. I thaught i had missed something.
Hi @flit, daily dose 😊: automated_test have 4/9 success on stm32f429zi (and all other boards i tested). All gdb tests, flash_test, and flash_loader_test do not pass. Have i missed something? Btw, debug with pyocd using all these boards works fine. Only tests  that cause problems.

flit, 2019-04-25:
No, you didn't miss anything! There are some known issues with STM32 devices not passing some of the tests: #579, #580, #586. Some fixes will be coming in 0.20.0, but that won't fix everything yet. But I haven't had any issues actually using it for debugging these device.

PpgpOs, 2019-04-26:
@flit 
`0047452:WARNING:freertos:FreeRTOS: pxReadyTasksLists size is unexpected, maybe an unsupported version of FreeRTOS`
This warning will appear when starting gdbserver when using FreeRTOS.
At this point, I can still enter debugging, but once I run `osKernelStart();` (Start scheduler), I will have problems.
Will eventually stop in the `vPortRaiseBASEPRI` function.

```
portmacro.h
...
portFORCE_INLINE static void vPortRaiseBASEPRI( void )
{
uint32_t ulNewBASEPRI;

	__asm volatile //Will eventually stop here,single step debugging did not respond
	(
		"	mov %0, %1												\n"	\
		"	msr basepri, %0											\n" \
		"	isb														\n" \
		"	dsb														\n" \
		:"=r" (ulNewBASEPRI) : "i" ( configMAX_SYSCALL_INTERRUPT_PRIORITY )
	);
}
```

PpgpOs, 2019-04-27:
@flit 
The stack is too large when assigning tasks

flit, 2019-04-27:
@PpgpOs Couple questions: What version of FreeRTOS are you using? Do you have a test application that you could give me to reproduce the error? Thanks!

PpgpOs, 2019-04-28:
@flit 
FreeRTOS v9.0.0. 
CMSIS RTOS v1.02. 
There's still this warning, but it doesn't seem to have a bad effect.
This is the project used https://github.com/PpgpOs/MyProgram.git

closed 2022-03-17 by yfclks