pyocd/pyOCD #625: Attribute or syntax errors while making test on new board added.

rgf97, 2019-04-23:
When i make test on a new board added, i get attribute or syntax error like this:

[image]

What did I did wrong?

flit, 2019-04-23:
You need to set the target type. Either use the `--target` command line argument, or `target_override` user option (best set in a `pyocd.yaml` config file). When running the functional tests you have to use a config file (in the `test/` directory) to set the target type, as well as set the `test_binary` option (set to just the name, not a path, of a file in the `binaries/` directory).

You can check the `Target type is x` log message to verify which target pyOCD is using. If no target type is set it will default to "cortex_m", which doesn't have flash support or a memory map.

rgf97, 2019-04-24:
I had already done that. but i named the .yaml file "test_boards" instead of "pyocd". Rename the file to pyocd solved the problem. Thank you @flit .

flit, 2019-04-24:
Is "test_boards.yaml" still mentioned in the docs? I checked, but could easily have missed it.

closed 2019-04-24 by flit

rgf97, 2019-04-24:
No, it is not mentioned in this docs. I did not use this docs. But i saw the same thing in a past issue solutions. And there, it was indicated to use a test_boards.yaml. 
Honestly, i thought the file name did not matter.

flit, 2019-04-24:
Well, originally it _was_ "test_boards.yaml". But when I added general config file support via "pyocd.yaml", I changed the functional tests to use that so there was only one config file name.