pyocd/pyOCD #632: Failed to install pyocd using pip3 under debian.

leemeans, 2019-04-26:
I try to install pyOCD under my debian stretch with pip upgraded to 
```
pip 19.1 from /home/leemeans/.local/lib/python3.7/site-packages/pip (python 3.7)
```
But it failed with those messages:
```
ERROR: Complete output from command python setup.py egg_info:
    ERROR: warning: no previously-included files found matching '*.nix'
    warning: no previously-included files found matching 'appveyor.yml'
    warning: no previously-included files found matching '.travis.yaml'
    warning: no previously-included files found matching '.pre-commit-config.yaml'
    
    Installed /tmp/easy_install-s8tqu0dp/setuptools_scm_git_archive-1.1/.eggs/setuptools_scm-3.2.0-py3.7.egg
    zip_safe flag not set; analyzing archive contents...
    Traceback (most recent call last):
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/sandbox.py", line 157, in save_modules
        yield saved
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/sandbox.py", line 198, in setup_context
        yield
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/sandbox.py", line 255, in run_setup
        DirectorySandbox(setup_dir).run(runner)
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/sandbox.py", line 285, in run
        return func()
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/sandbox.py", line 253, in runner
        _execfile(setup_script, ns)
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/sandbox.py", line 47, in _execfile
        exec(code, globals, locals)
      File "/tmp/easy_install-s8tqu0dp/setuptools_scm_git_archive-1.1/setup.py", line 39, in <module>
        'setuptools_scm_git_archive',
      File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 209, in run
        os.path.join(archive_root, 'EGG-INFO'), self.zip_safe()
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 245, in zip_safe
        return analyze_egg(self.bdist_dir, self.stubs)
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 355, in analyze_egg
        safe = scan_module(egg_dir, base, name, stubs) and safe
      File "/home/leemeans/.local/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 392, in scan_module
        code = marshal.load(f)
    ValueError: bad marshal data (unknown type code)
```

Then, I tried [this](https://stackoverflow.com/questions/30861493/how-to-fix-python-valueerrorbad-marshal-data), but it just didn't work :-(
Could you guys provide any clue about my situation?

flit, 2019-04-29:
It certainly does seem like a corrupted file related to the `setuptools_scm_git_archive` package.

You might try running `pip install --force-reinstall setuptools_scm_git_archive` before attempting to install pyOCD.

Also, could you run pip with `--verbose` (can be used up to 3 times to incrementally increase log output) and attach the log? If reinstalling setuptools_scm_git_archive fails, the log for that would be helpful.

leemeans, 2019-05-08:
After I tried the `pip install --force-reinstall setuptools_scm_git_archive` and `--verbose` with pip, I found that my setuptools's **version** is lower than 40, this is the problem really is. problem solved! Thanks!

flit, 2019-05-20:
Glad to hear it's solved. 

Fwiw, pyOCD's `setup.py` has a "setup_requires" for setuptools >= 40.0, but it doesn't seem to have any effect.

closed 2019-05-20 by flit