pyocd/pyOCD #634: Error starting gdbserver with CMSIS pack for F401RE

matt-oster, 2019-04-26:
I am trying to start the gdbserver with the STM32F401RE using a CMSIS pack and have encountered a couple of issues. I am running pyocd built in dev from the latest master (0.19.1.dev61+dirty)

The first one was a bit straight forward, it looks like there is a typo in pyocd/target/pack/pack_target.py around line 194 where isinstane should be isinstance.

After editing that file, I ran into the following issue which is well beyond a python novice like me:
```bash
$ pyocd gdbserver -t stm32f401re --pack /Users/matost01/Downloads/CMSIS/STM32F4/Keil.STM32F4xx_DFP.2.13.0.pack 
0000784:INFO:pack_target:Loading CMSIS-Pack: /Users/matost01/Downloads/CMSIS/STM32F4/Keil.STM32F4xx_DFP.2.13.0.pack
0000962:INFO:board:Target type is stm32f401re
0000962:ERROR:__main__:uncaught exception: unbound method _pack_target__init__() must be called with _PackTargetMethods instance as first argument (got Session instance instead)
Traceback (most recent call last):
  File "/Users/matost01/development/projects/mbed/pyOCD/pyocd/__main__.py", line 329, in run
    self._COMMANDS[self._args.cmd](self)
  File "/Users/matost01/development/projects/mbed/pyOCD/pyocd/__main__.py", line 545, in do_gdbserver
    **sessionOptions)
  File "/Users/matost01/development/projects/mbed/pyOCD/pyocd/core/helpers.py", line 196, in session_with_chosen_probe
    session = Session(allProbes[0], options=options, **kwargs)
  File "/Users/matost01/development/projects/mbed/pyOCD/pyocd/core/session.py", line 146, in __init__
    or Board(self, self._options.get('target_override', None))
  File "/Users/matost01/development/projects/mbed/pyOCD/pyocd/board/board.py", line 51, in __init__
    self.target = TARGET[self._target_type](session)
TypeError: unbound method _pack_target__init__() must be called with _PackTargetMethods instance as first argument (got Session instance instead)
```

flit, 2019-04-27:
Thanks @matt-oster! Will get this fixed quickly.

flit, 2019-04-27:
Ah, the second error is specific to Python 2.7 (it doesn't occur in Python 3).

flit, 2019-04-27:
Btw, I highly recommend using Python 3.7 as it's considerably faster than 2.7.

PR #636 (2019-04-27): Several pack targets fixes
- Fixed typo in `PackTargets.populate_targets_from_pack()`.
- Changed pack target loading log messages to be the same for both managed and explicit packs.
- Made methods in `_PackTargetMethods` static. This fixes an error that occurs only with Python 2.7, where it required the type of 'self' to be `_PackTargetMethods`.
- Some other cleanup.

Also included are some automated test improvements to work better with tox.

Closes #634
closed 2019-04-27 by flit