pyocd/pyOCD #638: pyocd pack --install fails

rgf97, 2019-04-29:
Hi, `pyocd pack --install `fails.

(venv) osboxes@osboxes:~/Desktop/pyOCD$ `pyocd --version`
0.20.0
(venv) osboxes@osboxes:~/Desktop/pyOCD$` which pyocd`
/home/osboxes/Desktop/pyOCD/venv/bin/pyocd
(venv) osboxes@osboxes:~/Desktop/pyOCD$ `pyocd pack --install stm32l476rg`
0000181:INFO:__main__:No pack index present, downloading now...
Apr 29 11:10:07.321 ERRO failed to lookup address information: Name or service not known, uri: http://www.keil.com/pack/keil.pidx
Apr 29 11:10:07.322 ERRO failed to lookup address information: Name or service not known, uri: http://www.keil.com/pack/keil.vidx
0000687:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.
(venv) osboxes@osboxes:~/Desktop/pyOCD$ `pyocd pack --install STM32L4xx_DFP`
0000194:INFO:__main__:No pack index present, downloading now...
Apr 29 11:15:16.777 ERRO failed to lookup address information: Name or service not known, uri: http://www.keil.com/pack/keil.pidx
Apr 29 11:15:16.781 ERRO failed to lookup address information: Name or service not known, uri: http://www.keil.com/pack/keil.vidx
0000702:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.
(venv) osboxes@osboxes:~/Desktop/pyOCD$ `pyocd pack --install stm32f429zi`
0000189:INFO:__main__:No pack index present, downloading now...
Apr 29 11:18:29.034 ERRO failed to lookup address information: Name or service not known, uri: http://www.keil.com/pack/keil.pidx
Apr 29 11:18:29.035 ERRO failed to lookup address information: Name or service not known, uri: http://www.keil.com/pack/keil.vidx
0000696:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.
(venv) osboxes@osboxes:~/Desktop/pyOCD$ `pyocd pack --install STM32F4xx_DFP`
0000183:INFO:__main__:No pack index present, downloading now...
Apr 29 11:20:31.595 ERRO failed to lookup address information: Name or service not known, uri: http://www.keil.com/pack/keil.pidx
Apr 29 11:20:31.598 ERRO failed to lookup address information: Name or service not known, uri: http://www.keil.com/pack/keil.vidx
0000689:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.

flit, 2019-04-29:
Weird! This seems like a DNS failure with the "failed to lookup address information" message. Can you load http://www.keil.com/pack/keil.pidx in your browser?

rgf97, 2019-04-29:
@flit It works when i load it in my browser.

flit, 2019-04-29:
Also, could you try running `pack-manager cache pdsc`? That's the separate tool for the [cmsis-pack-manager](https://github.com/armmbed/cmsis-pack-manager/) package that pyOCD is using to manage the pack index and download packs.

flit, 2019-04-29:
Unfortunately we may have to wait for the cmsis-pack-manager author to return from vacation in a couple weeks. The core of cpm is written in Rust, which I unfortunately don't know (yet!), for performance—it's *by far* the fastest CMSIS-Pack manager in existence.

rgf97, 2019-04-29:
@flit I have the same error using `pack-manager cache pdsc -vv`.
Ok.

flit, 2019-04-29:
Alright. Thanks for trying. I'm afraid we'll have to wait for the cpm author to return, this problem is deep in the network code in that project. But thank you very much for reporting it!

In the meantime, could you please describe your network setup? Do you have a proxy? Behind a firewall? etc, etc…

rgf97, 2019-04-29:
Yes, i am behind a proxy, firewall and a VM.

flit, 2019-04-29:
Hmm, ok. That surely sounds like a contributing factor.

flit, 2019-04-29:
Fyi, I created the reference issue directly on the cmsis-pack-manager project.

It may well be that cpm simply doesn't support proxies right now. I'm not sure.

rgf97, 2019-04-29:
OK, Thanks.

flit, 2019-04-29:
Fyi, you can still make use of packs by downloading them manually from https://www.keil.com/dd2/pack/. The [target support docs](https://github.com/mbedmicro/pyOCD/blob/master/docs/target_support.md) has full details.

cederom, 2019-04-29:
Hello @rgf97, looks like something is blocking traffic on the road.. just as @flit mentioned looks like DNS lookup problem more than proxy.. probably a firewall.. what OS do you use inside that VM? Please do the following and report each step:
```
1. ping www.keil.com
2. ping 217.140.99.213
3. ping 8.8.8.8
4A. ifconfig (if on Linux/Unix)
4B. ipconfig /all (if on Windows)
```

If above does not work please review firewall rules or disable firewall for a minute and retry above. This also happens to me with "interactive firewall" that blocks a new application by default for some reason..

rgf97, 2019-04-29:
@flit , Indeed, like in #624. This is the backstop.

rgf97, 2019-04-30:
Yes, this is finally not a bug. This is firewall problem. (my firewall rule 1: do not reach any single service using command prompt). 
Thank you @flit and @cederom

closed 2019-04-30 by rgf97

flit, 2019-04-30:
Thanks @rgf97 and @cederom! Special thanks to @cederom for helping get to the bottom of the issue.

jeromecoutant, 2019-05-03:
Hi
Back to this issue...
I agree that I have some network/PC restrictions due to company policy...

````
$ pyocd pack --install stm32f303
0000348:INFO:__main__:No pack index present, downloading now...
May 03 11:43:05.522 ERRO No such host is known. (os error 11001), uri: http://www.keil.com/pack/keil.vidx
May 03 11:43:05.523 ERRO No such host is known. (os error 11001), uri: http://www.keil.com/pack/keil.pidx
````
OK

````
$ curl -L http://www.keil.com/pack/keil.vidx -O
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  5668  100  5668    0     0  21308      0 --:--:-- --:--:-- --:--:-- 21308
100  9622  100  9622    0     0  17590      0 --:--:-- --:--:-- --:--:-- 17590

$ curl -L http://www.keil.com/pack/keil.pidx -O
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  5672  100  5672    0     0  60340      0 --:--:-- --:--:-- --:--:-- 60340
100 14216  100 14216    0     0  19771      0 --:--:-- --:--:-- --:--:-- 19771
````
So it seems that restrictions I have could be avoid ?

flit, 2019-05-03:
@jeromecoutant Yes, there probably is a way… Could you try appending `--trace trace.txt` to the `curl` command (either one) and attach the resulting `trace.txt` file? That might provide some insight about what curl is doing differently. Though in testing locally it doesn't seem to log anything related to DNS, which seems to be where cmsis-pack-manager is tripping up. Thanks!

jeromecoutant, 2019-05-06:
Seems that curl is calling directly GET HTTP function,
whereas pack manager use first DNS call without any proxy setting.

flit, 2019-05-08:
@jeromecoutant I reopened ARMmbed/cmsis-pack-manager#101 with the information you provided. Let's move the conversation over there.