pyocd/pyOCD #643: Some clarifications on pyocd pack command

rgf97, 2019-05-02:
Hi, what is the correct syntax to support new target after downloading manually its cmsis-pack. 
Is it `pyocd --pack /my/pyOCD/Keil.pack/path/` like target_support docs mentioned?
This do not work.
Or, is this the real command` pyocd gdbserver -- pack /my/pyOCD/Keil.pack/path/`?

flit, 2019-05-03:
You need to add the `--pack /my/pyOCD/Keil.pack/path/` argument to all pyOCD tool invocations, not just one time.

It may be easier for you to create a `pyocd.yaml` config file that sets the `pack` option. This way you don't have to pass `--pack` for every command, assuming the config file is in your working directory (or you pass `--config=path/to/config/file`). ([Configuration docs](https://github.com/mbedmicro/pyOCD/blob/master/docs/configuration.md) have more info.)

Here's an example `pyocd.yaml` config file (copied from my own config file) that adds a bunch of packs:

```yaml
pack:
  - /Users/creed/Downloads/_packs/NXP.MK66F18_DFP.11.0.0.pack
  - /Users/creed/Downloads/_packs/NXP.MK64F12_DFP.11.0.0.pack
  - /Users/creed/Downloads/_packs/NordicSemiconductor.nRF_DeviceFamilyPack.8.17.0.pack
  - /Users/creed/Downloads/_packs/Keil.STM32L0xx_DFP.2.0.0.pack
  - /Users/creed/Downloads/_packs/Keil.STM32L4xx_DFP.2.0.0.pack
  - /Users/creed/Downloads/_packs/Keil.STM32F4xx_DFP.2.13.0.pack
  - /Users/creed/Downloads/_packs/NXP.MIMXRT1052_DFP.11.0.0.pack
  - /Users/creed/Downloads/_packs/Keil.STM32F4xx_DFP.2.13.0.pack
  - /Users/creed/Downloads/_packs/ARM.Musca_A1_BSP.2.0.0.pack
```

rgf97, 2019-05-06:
Thanks @flit, this works.

My `pyocd.yaml`:

```
probes:
  066EFF535550755187241812: # Probe's unique ID.
    target_override:  stm32l476rg
    test_binary:      nucleo_l476rg.bin
  0667FF575550897767151751:
    target_override:  stm32f103rb
    test_binary:      nucleo_f103rb.bin

pack:
  - /home/osboxes/Desktop/pyOCD/Keil.STM32L4xx_DFP.2.2.0.pack
  - /home/osboxes/Desktop/pyOCD/Keil.STM32F1xx_DFP.2.3.0.pack
```

`Auto-detection` too: 
```
(venv) osboxes@osboxes:~/Desktop/pyOCD$ pyocd flash zephyr.bin 
0000270:INFO:session:Using config settings for board 066EFF535550755187241812
0000888:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=415)
0000913:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000927:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0000941:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0000958:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0000971:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0000985:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
[====================] 100%
0009075:INFO:loader:Erased 16384 bytes (8 sectors), programmed 16384 bytes (16 pages), skipped 0 bytes (0 pages) at 2.05 kB/s
```

closed 2019-05-06 by rgf97