pyocd/pyOCD #647: Writing to flash fails with traceback

Fliptron, 2019-05-06:
DAP-Link interface , nRF51822 target
Windows 7 Pro x64

Read and write RAM works:
```
C:\Temp>pyocd cmd -c R32 0x2000000
02000000:  00000000

C:\Temp>pyocd cmd -c R32 0x20000000
20000000:  e00abe00

C:\Temp>pyocd cmd -c W32 0x20000000 0xe00abe01

C:\Temp>pyocd cmd -c R32 0x20000000
20000000:  e00abe01

C:\Temp>pyocd cmd -c W32 0x20000000 0xe00abe00

C:\Temp>pyocd cmd -c R32 0x20000000
20000000:  e00abe00
```

Reading flash works, writing flash fails:

```
C:\Temp>pyocd cmd -c R32 0x10001000
10001000:  ffffffff

C:\Temp>pyocd cmd -c W32 0x10001000 0xA5A55A5A
0000625:ERROR:__main__:uncaught exception: name 'flashPtr' is not defined
Traceback (most recent call last):
  File "c:\uac_ok\python37\lib\site-packages\pyocd\__main__.py", line 329, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\uac_ok\python37\lib\site-packages\pyocd\__main__.py", line 577, in do_commander
    PyOCDCommander(self._args, cmds).run()
  File "c:\uac_ok\python37\lib\site-packages\pyocd\tools\pyocd.py", line 648, in run
    result = self.command_list[cmd](args)
  File "c:\uac_ok\python37\lib\site-packages\pyocd\tools\pyocd.py", line 899, in handle_write32
    return self.do_write(args, 32)
  File "c:\uac_ok\python37\lib\site-packages\pyocd\tools\pyocd.py", line 992, in do_write
    region.flash.program_phrase(addr, data)
  File "c:\uac_ok\python37\lib\site-packages\pyocd\flash\flash.py", line 421, in program_phrase
    if flashPtr % min_len:
NameError: name 'flashPtr' is not defined
```

I am fairly sure the problem is on line 421 of flash,py  .

It should be:

`if address % min_len:`

This problem also occurs in console mode.

flit, 2019-05-08:
Thank you for reporting this bug!

Fliptron, 2019-05-08:
Thanks for handling this. I would have done a PR, but I don't have the skill to do that. I'm currently using pyocd.exe , and was unable to figure out how to run it from sources so I could check that I had figured out the correct solution. Is there an easy way to run form sources? I tried but kept getting errors like

```
D:\temp>C:\Python37\Lib\site-packages\pyocd\tools\pyocd.py
Traceback (most recent call last):
  File "C:\Python37\Lib\site-packages\pyocd\tools\pyocd.py", line 35, in <module>
    from .. import __version__
ValueError: attempted relative import beyond top-level package
```

flit, 2019-05-08:
No worries, the main thing is that you reported the bug, which is a big help.

The easiest way to run from the source is documented in the [developers' guide](https://github.com/mbedmicro/pyOCD/blob/master/docs/developers_guide.md). Basically, create a virtualenv, then run `pip install -r dev-requirements.txt`. This will install pyOCD as "editable" in the virtualenv, which means it uses the local sources instead of copying it into site-packages.

The reason you get the above exception is because pyOCD has to be run via its `pyocd` command line tool (which is created by setuptools when pyOCD is installed). The `pyocd/tools/pyocd.py` script is the deprecated `pyocd-tool` command line tool (plus lots of shared code). The entry point for `pyocd` is `pyocd/__main__.py`. You can also run pyOCD as a module via `python -m pyocd`, which will invoke `__main__.py`.

PR #651 (2019-05-08): Fix for program phrase plus some cleanup
This patch fixes a regression in `Flash.program_phrase()` introduced when the `flashPtr` argument was renamed to `address`.

Also some cleanup:
- `Flash` accesses the region page size attribute directly instead of going through `get_page_info()`, which reduces the need for error checking.
- If a managed CMSIS-Pack is not a valid zip file, a better error is raised that includes the pack's path.

Closes #647
closed 2019-05-08 by flit