pyocd/pyOCD #652: AttributeError: 'module' object has no attribute 'BadZipFile'

maxgerhardt, 2019-05-09:
When trying to execute the recommendation in https://github.com/mbedmicro/pyOCD/issues/623#issuecomment-484945469 in order to get the STM32L476 working, the following error occurss

```
max@XXX:~/Desktop$ pyocd pack -i stm32l476
0000229:INFO:__main__:No pack index present, downloading now...
May 09 17:24:16.494 WARN Failed to open vendor index list read only. Recreating.
May 09 17:24:51.332 WARN No Core found!, Package: CMSIS, Vendor: ARM
May 09 17:24:51.638 ERRO when trying to parse component: child element "description" not found in "component" element, Package: RS14100_DFP, Vendor: Redpine
May 09 17:24:51.641 WARN invalid digit found in string, Package: TLE984x_DFP, Vendor: Infineon
May 09 17:24:51.794 WARN expecting </link> found </head>
May 09 17:24:51.794 WARN expecting </link> found </head>
Downloading packs (press Control-C to cancel):
    Keil::STM32L4xx_DFP::2.2.0
max@XXX:~/Desktop$ pyocd list --targets
0002328:CRITICAL:__main__:uncaught exception: 'module' object has no attribute 'BadZipFile'
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/pyocd/__main__.py", line 330, in run
    self._COMMANDS[self._args.cmd](self)
  File "build/bdist.linux-x86_64/egg/pyocd/__main__.py", line 386, in do_list
    obj = ListGenerator.list_targets()
  File "build/bdist.linux-x86_64/egg/pyocd/tools/lists.py", line 136, in list_targets
    for dev in pack_target.ManagedPacks.get_installed_targets():
  File "build/bdist.linux-x86_64/egg/pyocd/target/pack/pack_target.py", line 62, in get_installed_targets
    pack = CmsisPack(pack_path)
  File "build/bdist.linux-x86_64/egg/pyocd/target/pack/cmsis_pack.py", line 83, in __init__
    except zipfile.BadZipFile as err:
AttributeError: 'module' object has no attribute 'BadZipFile'
```

Seems like some vendor's download link crashed and we only get garbage, leading to `pyocd` self-destructing?

flit, 2019-05-09:
Thanks for the report!

cmsis-pack-manager, which pyOCD uses to manage packs, had a bug (ARMmbed/cmsis-pack-manager#95) where it would incorrectly construct the URL for a pack if the URL didn't have a trailing slash. This is fixed now, and a release will be made shortly that pyOCD will pick up.

The `BadZipFile` error is something I'll have to investigate.

closed 2019-05-09 by flit

maxgerhardt, 2019-05-10:
@flit Are the docs out of sync with the program? Regarding manual downloading, it says

> To use a downloaded pack, just pass the --pack option to the pyocd tool

However `pyocd --pack <path>` is an invalid command. `pyocd pack --help` shows `-f` and `-i` options, but they don't install a pack from a file. (`pyocd pack --install ~/Downloads/Keil.STM32L4xx_DFP.2.2.0.pack -> No matching devices.`). What's the right command?

rgf97, 2019-05-10:
Hi @maxgerhardt, 
There are two different situations and the command is not the same depending of the situation.
If you want to automatically activate a new target: `pyocd pack --install stm32l476`.
If you want to do this manually, there are two possibilities:
  - the simple way: `pyocd gdbserver --pack /your/pyOCD/Keil.pack/path/`
  - the other way and the recommended: create a` pyocd.yaml` file in your `PyOCD` directory and make the configuration like this:
```
probes:
  066EFF535550755187241812: # Probe's unique ID.
    target_override:  stm32l476rg
    test_binary:      nucleo_l476rg.bin

pack:
  - /home/osboxes/Desktop/pyOCD/Keil.STM32L4xx_DFP.2.2.0.pack

```
Do not forget to add a nucleo_l476rg.bin (your_board_name.bin) file in `PyOCD/binaries` folder. (Just copy an existing .bin folder and rename it)

maxgerhardt, 2019-05-10:
@rgf97 Thanks, worked liked a charm 

```
$ pyocd gdbserver --pack Keil.STM32L4xx_DFP.2.2.0.pack 
0000325:INFO:board:Target type is stm32l476rg
0000525:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0000862:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001169:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=415)
0001297:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001349:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0001372:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0001402:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0001428:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0001446:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001477:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0001896:INFO:cortex_m:FPU present: FPv4-SP
0001901:INFO:dwt:4 hardware watchpoints
0001915:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001955:INFO:server:Semihost server started on port 4444
0001955:INFO:gdbserver:GDB server started on port 3333
```

rgf97, 2019-05-10:
@maxgerhardt I made an edit, maybe it can help for the future.

flit, 2019-05-10:
@maxgerhardt I'll update the docs to make it more clear how to use `--pack`. You are not the only person who read it differently than I intended, so obviously I didn't write it very well! 😄

rgf97, 2019-05-10:
@flit Oohh great! Thanks for the precision!

flit, 2019-05-10:
@rgf97 I should also mention that starting with version 0.20.0, you mostly don't need to set `target_override` for STLink probes, as the target type is now auto-detected for STLinkV2-1 probes that provide the USB mass storage drive with the `MBED.HTM` file.

PR #653 (2019-05-09): Worked around misspelled BadZipFile class in Python 2
The `zipfile` module in Python 2 has the `BadZipFile` exception class misspelled as `BadZipfile`.

Closes #652