pyocd/pyOCD #654: 'IntervalTree' object has no attribute 'overlap' while debugging STM32L476RG

maxgerhardt, 2019-05-10:
When starting a `pyocd` server with the STM32L4xx pack and connecting to it via the CLion IDE (which uses `arm-none-eabi-gdb`  version `8.2.50.20181213-git`), the debugger can connect to the target; however, when pressing the "pause" button to halt execution, the following error appears:

```
max@PHYSEC-E550-01:~/Downloads$ pyocd gdbserver --pack Keil.STM32L4xx_DFP.2.2.0.pack 
0000248:INFO:board:Target type is stm32l476rg
0000440:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0000819:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001070:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=415)
0001142:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001211:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0001280:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0001315:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0001358:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0001373:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001387:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0001546:INFO:cortex_m:FPU present: FPv4-SP
0001595:INFO:dwt:4 hardware watchpoints
0001828:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0002143:INFO:server:Semihost server started on port 4444
0002144:INFO:gdbserver:GDB server started on port 3333
0006395:INFO:gdbserver:One client connected!
0006572:ERROR:gdbserver:Unhandled exception in handle_message: 'IntervalTree' object has no attribute 'overlap'
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 545, in handle_message
    reply = handler(msg[msgStart:])
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 903, in get_memory
    mem = self.target_context.read_memory_block8(addr, length)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 523, in read_memory_block8
    return self._memcache.read_memory_block8(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 445, in read_memory_block8
    combined = self._read(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 323, in _read
    cached, uncached = self._get_ranges(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 259, in _get_ranges
    cached = self._cache.overlap(addr, addr + count)
AttributeError: 'IntervalTree' object has no attribute 'overlap'
0006577:ERROR:gdbserver:Unhandled exception in handle_message: 'IntervalTree' object has no attribute 'overlap'
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 545, in handle_message
    reply = handler(msg[msgStart:])
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 903, in get_memory
    mem = self.target_context.read_memory_block8(addr, length)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 523, in read_memory_block8
    return self._memcache.read_memory_block8(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 445, in read_memory_block8
    combined = self._read(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 323, in _read
    cached, uncached = self._get_ranges(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 259, in _get_ranges
    cached = self._cache.overlap(addr, addr + count)
AttributeError: 'IntervalTree' object has no attribute 'overlap'
0006581:ERROR:gdbserver:Unhandled exception in handle_message: 'IntervalTree' object has no attribute 'overlap'
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 545, in handle_message
    reply = handler(msg[msgStart:])
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 903, in get_memory
    mem = self.target_context.read_memory_block8(addr, length)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 523, in read_memory_block8
    return self._memcache.read_memory_block8(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 445, in read_memory_block8
    combined = self._read(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 323, in _read
    cached, uncached = self._get_ranges(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 259, in _get_ranges
    cached = self._cache.overlap(addr, addr + count)
AttributeError: 'IntervalTree' object has no attribute 'overlap'
0006584:ERROR:gdbserver:Unhandled exception in handle_message: 'IntervalTree' object has no attribute 'overlap'
Traceback (most recent call last):
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 545, in handle_message
    reply = handler(msg[msgStart:])
[... 109 lines cut ...]
    return self.v_cont(cmd)
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 787, in v_cont
    if self.is_threading_enabled():
  File "build/bdist.linux-x86_64/egg/pyocd/gdbserver/gdbserver.py", line 1317, in is_threading_enabled
    return (self.thread_provider is not None) and self.thread_provider.is_enabled \
  File "build/bdist.linux-x86_64/egg/pyocd/rtos/rtx5.py", line 412, in is_enabled
    return self.get_kernel_state() != 0 and not self._target.in_thread_mode_on_main_stack()
  File "build/bdist.linux-x86_64/egg/pyocd/rtos/rtx5.py", line 450, in get_kernel_state
    return self._target_context.read8(self._os_rtx_info + RTX5ThreadProvider.KERNEL_STATE_OFFSET)
  File "build/bdist.linux-x86_64/egg/pyocd/core/memory_interface.py", line 64, in read8
    return self.read_memory(addr, 8, now)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 514, in read_memory
    return self._memcache.read_memory(addr, transfer_size, now)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 420, in read_memory
    data = self.read_memory_block8(addr, 1)[0]
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 445, in read_memory_block8
    combined = self._read(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 323, in _read
    cached, uncached = self._get_ranges(addr, size)
  File "build/bdist.linux-x86_64/egg/pyocd/debug/cache.py", line 259, in _get_ranges
    cached = self._cache.overlap(addr, addr + count)
AttributeError: 'IntervalTree' object has no attribute 'overlap'
```

The used hardware is a Nucleo L476RG (onboard ST-Link v2-1). The firmware of the ST-Link was recently updated to the most recent version provided by ST. The firmware uses mbed-os (RTX5 RTOS)

maxgerhardt, 2019-05-10:
Update: When I treat instead use `-t cortex_m` it works.

[image]

maxgerhardt, 2019-05-10:
This is again a Python2-specific error: I uninstalled the Python2 version of `pyocd` and re-installed with `python3` and the error is gone ._. 

```
$ pyocd gdbserver --pack Keil.STM32L4xx_DFP.2.2.0.pack 
0000645:INFO:board:Target type is stm32l476rg
0000776:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0000901:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0000945:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=415)
0000948:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000951:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0000954:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0000956:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0000958:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0000960:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0000962:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0000970:INFO:cortex_m:FPU present: FPv4-SP
0000973:INFO:dwt:4 hardware watchpoints
0000976:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000988:INFO:server:Semihost server started on port 4444
0000989:INFO:gdbserver:GDB server started on port 3333
0007825:INFO:gdbserver:One client connected!
0008389:INFO:gdbserver:Attempting to load Argon
0008389:INFO:gdbserver:Attempting to load FreeRTOS
0008389:INFO:gdbserver:Attempting to load Zephyr
0008389:INFO:gdbserver:Attempting to load RTX5
0008390:INFO:gdbserver:RTX5 loaded successfully
```

flit, 2019-05-12:
@maxgerhardt Thanks for the bug report! And especially thanks for figuring out that it's specific to Python 2. I'll try to reproduce it and update this issue.

flit, 2019-05-14:
@maxgerhardt So far I'm unable to reproduce it with gdb.

I've also tried more directly testing by attempting to call the `overlap()` in Python 2:

```
Python 2.7.15 (v2.7.15:ca079a3ea3, Apr 29 2018, 17:49:27) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import intervaltree
>>> t=intervaltree.IntervalTree()
>>> t.overlap(0,0)
set([])
```

That was with intervaltree 3.0.2, which is the version that pyOCD's requirements in `setup.py` specifies.

The only thing I can think of is that somehow you had an old version of intervaltree. Were you running pyOCD from the sources (i.e., an "editable" installation)? Pulling changes via git will not update dependency requirements unless you reinstall pyOCD ( by running `pip install -r dev-requirements.txt`).

flit, 2019-05-14:
I think the reason it worked for cortex_m is that the call to `overlap()` was done from the memory cache. cortex_m has no memory regions defined, so pyOCd doesn't attempt to cache memory (because it doesn't know what's safe to cache).

maxgerhardt, 2019-05-14:
I'll check up on the installed `intervaltree` package version and let you know. I've installed it by cloning the repo and executing `python setup.py install`, so I might really not have the required version.

maxgerhardt, 2019-05-21:
The installed version was 

```
max@X:~/$ pip list | grep inter
intervaltree                       2.1.0      
```
so that was very likely my problem. Closing.

closed 2019-05-21 by maxgerhardt

flit, 2019-05-21:
Thanks for checking and reporting back! You just have to be careful when you pull changes for a source-based installation, since requirements aren't automatically updated. I've been caught by that more than once when switching between different virtualenvs and test VMs.