pyocd/pyOCD #661: STLink Error (9): Get IDCODE Error

samvrit, 2019-05-17:
I am using a NUCLEO-L476RG board, and I have removed the CN2 jumpers, which would use the ST-Link instead of the Nucleo. When I run a flashing command with the following command line:
`pyocd flash mbed_blinky.bin --target stm32l476rg`
I get the following error:
`0001855:ERROR:__main__:uncaught exception: STLink error (9): Get IDCODE error
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\__main__.py", line 329, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\python27\lib\site-packages\pyocd\__main__.py", line 448, in do_flash
    **convert_session_options(self._args.options))
  File "c:\python27\lib\site-packages\pyocd\core\helpers.py", line 199, in session_with_chosen_probe
    session.open()
  File "c:\python27\lib\site-packages\pyocd\core\session.py", line 300, in open
    self._board.init()
  File "c:\python27\lib\site-packages\pyocd\board\board.py", line 70, in init
    self.target.init()
  File "c:\python27\lib\site-packages\pyocd\core\coresight_target.py", line 152, in init
    seq.invoke()
  File "c:\python27\lib\site-packages\pyocd\utility\sequencer.py", line 189, in invoke
    resultSequence = call()
  File "c:\python27\lib\site-packages\pyocd\coresight\dap.py", line 100, in init
    self.link.connect()
  File "c:\python27\lib\site-packages\pyocd\probe\stlink_probe.py", line 114, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "c:\python27\lib\site-packages\pyocd\probe\stlink\stlink.py", line 215, in enter_debug
    self._check_status(response)
  File "c:\python27\lib\site-packages\pyocd\probe\stlink\stlink.py", line 253, in _check_status
    raise exceptions.ProbeError(error_message)
ProbeError: STLink error (9): Get IDCODE error`

I have updated the ST-Link firmware to the latest one (V2J34M25), yet I get this issue. Anyone with similar issues and have solved it?

flit, 2019-05-20:
To make sure I understand, you're using a standalone STLinkV2 connected to the SWD header on the Nucleo-L476RG board?

Unfortunately, I don't have a standalone STLinkV2, but I do have an STLinkV3. I'll try to finish up #447 so I can try to replicate this issue as best as possible.

cc @schstm

flit, 2019-05-20:
I'm able to reproduce this with an STLinkV3 and a Nucleo-L073RZ. At this point, I don't know what could be causing it.

samvrit, 2019-05-20:
I am using the embedded ST-Link circuit on the Nucleo, but I have unplugged the CN2 jumper, which on the datasheet, indicates that it should use the ST-Link interface and not the Nucleo interface.

When I plug a standalone ST-Link connector directly to the headers (PA 13 and PA14 for SWDIO and SWDCLK), it works/flashes properly.

flit, 2019-05-20:
Sorry, I don't quite understand what you're trying to achieve. CN2 seems to control whether the external SWD header connects to the on-board STLink MCU (STM32F103CB) rather than the STM32L476 target.

Anyway, this definitely seems to be a board configuration issue rather than a problem with pyOCD.

schstm, 2019-05-21:
Hello,
which datasheet are you talking about ? In the UM1724 (rev13) associated to this board it is stated:
Both CN2 jumpers ON: ST-LINK/V2-1 functions enabled for on board programming
(default)
Both CN2 jumpers OFF: ST-LINK/V2-1 functions enabled for external CN4 connector
(SWD supported)
Which means if you remove the CN2 jumpers you must connect an external target to CN4, instead of using the STM32L476 embedded on the Nucleo board.

samvrit, 2019-05-22:
Hi,
I am getting the same error even when I use a standalone debug probe.

schstm, 2019-05-22:
Hello,
Does it work with the embedded ST-Link/V2-1 and CN2 jumpers ON ?
The NUCLEO-L476RG is designed for 2 use cases:
1. Embedded ST-Link/V2-1 + embedded STM32L476 target (CN2 jumpers ON)
2. Embedded ST-Link/V2-1 + external target through CN4 (CN2 jumpers OFF).
I don't understand what you are doing with a standalone debug probe and a NUCLEO-L476RG (where do you connect, did you cut the ST-Link part from STM32L476 part, ... ?)

samvrit, 2019-05-22:
Hello,

I think I had misunderstood the function of the NUCLEO when the CN2 jumpers are off (I assumed it is to program the same embedded target with different configurations). It works well with the embedded ST-Link to program the embedded target when the CN2 jumpers are on.

Thank you for the clarification.

flit, 2019-05-22:
@schstm Thanks very much for your help!

Closing this issue.

thegecko, 2019-11-28:
We have also hit this problem using an external STLink v2 probe on a Nucleo F429ZI (headers off, SWD connector).

A valid use case here is to support the external probe for custom hardware scenarios.

@flit You mention you were able to reproduce this error, did you get to the bottom of what was causing it?

Alternatively, do you know of a target which works correctly with STLink probes and PyOCD?

schstm, 2019-11-28:
On the Nucleo F429ZI, as already said (in other words) for the Nucleo-L472RG, the SWD connector (CN6 on nucleo F429ZI) is an output of the embedded ST-Link/V2-1, not an input for the target MCU of the board. If really wanting to connect to the target MCU through an external probe, it should be possible to use CN4.2 (SWCLK) and CN4.4 (SWDIO) instead, with the CN4 headers OFF, but be aware that the board was not designed for such a use case and that the embedded ST-Link/V2-1 might interfere with the external probe on other signals (T_NRST, PWR_ENn, ...). Please have a look at the board schematics for more details: https://www.st.com/resource/en/schematic_pack/nucleo_144pins_sch.zip
I do not recommend to use the board in this way, unless you are comfortable with the given schematics (and ready to solder/unsolder some bridges if required)

thegecko, 2019-11-28:
Thanks for your help @schstm I'll take a look.

Are you aware of any Mbed enabled ST boards which could be used with an external STLink probe or are they all setup in this way?

schstm, 2019-11-29:
The recent boards embedding the STLINK-V3 version (like NUCLEO-H745ZI-Q, G431RB, G474RE, ...) provide an ARM10 input connector to the target on which you can plug an external probe for debugging purpose. Note however that the mbed functionality is rather linked to the board itself (embedded ST-Link+target micro), and not to the target micro only;  when using an external probe through ARM10 you will loose the Virtual Com Port and mass storage interfaces, which are exposed only through the embedded STLINK-V3 USB connector. I still don't understand your final use case (mbed through external probe ?). Some features might be provided by the STLINK-V3SET, such as the VCP, but it won't be mbed-enabled.

thegecko, 2019-11-29:
Perfect, thanks