pyocd/pyOCD #665: AssertionError: Flash pages (1024 bytes) do not fit evenly into sector (128 bytes)

wnts, 2019-05-20:
I'm using the B-L072Z-LRWAN evaluation board from ST, and I'm trying to flash a program into the MCU memory using the gdb "load" command, but I receive the following error:
```
0000392:INFO:board:Target type is stm32l072cz
0000467:INFO:dap:DP IDR = 0x0bc11477 (v1 MINDP rev0)
0000496:INFO:ap:AP#0 IDR = 0x04770031 (AHB-AP var3 rev0)
0000514:INFO:rom_table:AP#0 ROM table #0 @ 0xf0000000 (designer=020 part=447)
0000528:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c0>
0000528:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0000542:INFO:rom_table:  [0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0000551:INFO:rom_table:  [1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0000560:INFO:rom_table:  [2]<e0002000:BPU class=14 designer=43b part=00b>
0000565:INFO:cortex_m:CPU core #0 is Cortex-M0+ r0p1
0000574:INFO:dwt:2 hardware watchpoints
0000584:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0000604:INFO:server:Semihost server started on port 4444
0000605:INFO:gdbserver:GDB server started on port 3333
0001201:INFO:gdbserver:One client connected!
0008731:ERROR:gdbserver:Unhandled exception in handle_message: Flash pages (1024 bytes) do not fit evenly into sector (128 bytes)
Traceback (most recent call last):
  File ".local/lib/python3.5/site-packages/pyocd/gdbserver/gdbserver.py", line 539, in handle_message
    reply = handler(msg[msgStart:])
  File ".local/lib/python3.5/site-packages/pyocd/gdbserver/gdbserver.py", line 760, in v_command
    return self.flash_op(data)
  File ".local/lib/python3.5/site-packages/pyocd/gdbserver/gdbserver.py", line 876, in flash_op
    self.flash_loader.commit()
  File ".local/lib/python3.5/site-packages/pyocd/flash/loader.py", line 505, in commit
    keep_unwritten=self._keep_unwritten)
  File ".local/lib/python3.5/site-packages/pyocd/flash/flash_builder.py", line 384, in program
    self._build_sectors_and_pages(keep_unwritten)
  File ".local/lib/python3.5/site-packages/pyocd/flash/flash_builder.py", line 245, in _build_sectors_and_pages
    current_sector.add_page(current_page)
  File ".local/lib/python3.5/site-packages/pyocd/flash/flash_builder.py", line 78, in add_page
    "into sector (%d bytes)" % (page.size, self.size)
AssertionError: Flash pages (1024 bytes) do not fit evenly into sector (128 bytes)
```
I downloaded the CMSIS pack for the stm32l072cz target.

flit, 2019-05-20:
Unfortunately, the flash algorithm in the Keil.STM32L0xx_DFP CMSIS-Pack is broken, because it has a page size that is larger than the sector size. I've already reported this to the Keil team and they plan to issue a fixed CMSIS-Pack (not sure what timeframe, though).

Even so, I'm going to try to implement some workarounds for common mistakes in packs, like this one.

zougloub, 2023-10-19:
I tried pyOCD on an stm32h7a3zitxq, pack Keil.STM32H7xx_DFP   3.1.1, and get a similar error:
```
Error: Flash pages (32768 bytes) do not fit evenly into sector (8192 bytes) 
```