pyocd/pyOCD #673: using pyocd.yaml file in python api

harrrson, 2019-05-24:
In docs files, i've seen, how to use '''pyocd.yaml''' file in pyocd API. But i cannot find, how to use this file in python API. Or, when it is not implemented yet, how can i change target in code? This is important for me, because my probe doesn't auto-detect type of my uC.

harrrson, 2019-05-24:
I've fixed it faster than i thought. I dont know why, but, when API works good with `probes` class, python API didn't read it, or unique ID read by pyAPI didn't match one in class. I had fixed it with taking `target_override` as global.

closed 2019-05-24 by harrrson

flit, 2019-05-24:
Hi @harrrson, glad to hear it worked out for you. If you have any other questions about the Python API, I'll be happy to help.

For a little more info… these options can be set when creating a session (via `ConnectHelper.session_with_chosen_probe()` or another method) to explicitly control files/directories that pyOCD uses:

- `project_dir`: the working directory where pyOCD looks for the config file and user script, 
- `config`: the config file relative or absolute path, default "pyocd.yaml"
- `user_script`: the user script relative or absolute path, default "pyocd_user.py"

If `config` or `user_script` are not absolute paths, they are relative to the project dir. 

The first two have to be set when the session is created, since they determine where the rest of the user options are loaded from.