pyocd/pyOCD #680: Target cy8c6xxa: pyocd-gdbserver: does not work with latest version 0.21.0

farrenv, 2019-05-31:
pyocd gdbserver does not work for the target cy8c6xxa for the cypress Cy8PROTO-062-4343W target with release 0.21.0.  It does work with version 0.16.1 when the target was introduced.
version 0.21.0
```
$ pyocd gdbserver
0000985:INFO:board:Target type is cy8c6xxa
0001021:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0001037:INFO:ap:AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
0001049:INFO:ap:AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0001063:INFO:ap:AP#2 IDR = 0x24770011 (AHB-AP var1 rev2)
0001078:INFO:rom_table:AP#0 ROM table #0 @ 0xf1000000 (designer=034 part=102)
0001090:INFO:rom_table:AP#1 ROM table #0 @ 0xf0000000 (designer=034 part=102)
0001099:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c0>
0001100:INFO:rom_table:  AP#1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0001109:INFO:rom_table:  [0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0001115:INFO:rom_table:  [1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0001121:INFO:rom_table:  [2]<e0002000:BPU class=14 designer=43b part=00b>
0001126:INFO:rom_table:[1]<f0002000:CTI class=9 designer=43b part=9a6 devtype=14 archid=1a14 devid=0:0:1040800>
0001133:INFO:rom_table:[2]<f0003000:MTB-M0+ class=9 designer=43b part=932 devtype=31 archid=0a31 devid=0:0:0>
0001144:INFO:rom_table:AP#2 ROM table #0 @ 0xe00ff000 (designer=034 part=102)
0001152:INFO:rom_table:[0]<e0080000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=0:0:40800>
0001158:INFO:rom_table:[3]<e008e000:TPIU-M3 class=9 designer=43b part=923 devtype=11 archid=0000 devid=0:0:ca1>
0001163:INFO:rom_table:[4]<e007f000:ROM class=1 designer=034 part=102>
0001164:INFO:rom_table:  AP#2 ROM table #1 @ 0xe007f000 (designer=034 part=102)
0001173:INFO:rom_table:  [0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001179:INFO:rom_table:  [1]<e0001000:DWT class=14 designer=43b part=002>
0001185:INFO:rom_table:  [2]<e0002000:FPB class=14 designer=43b part=003>
0001191:INFO:rom_table:  [3]<e0000000:ITM class=14 designer=43b part=001>
0001197:INFO:rom_table:  [4]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=0:0:40800>
0001202:INFO:rom_table:  [5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001206:CRITICAL:__main__:uncaught exception: 'CY8C6xxA' object has no attribute 'options'
Traceback (most recent call last):
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\__main__.py", line 566, in do_gdbserver
    with session:
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\coresight_target.py", line 158, in init
    seq.invoke()
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\utility\sequencer.py", line 201, in invoke
    resultSequence = call()
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\target\builtin\target_CY8C6xxA.py", line 412, in create_cy8c6xx7_core
    core0 = CortexM_CY8C6xxA(self, self.aps[1], self.memory_map, 0)
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\coresight\cortex_m.py", line 409, in __init__
    Target.__init__(self, session, memoryMap)
  File "c:\users\farrenv\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\target.py", line 98, in __init__
    self.halt_on_connect = session.options.get('halt_on_connect')
AttributeError: 'CY8C6xxA' object has no attribute 'options'
```

version 0.16.1

```
$ pyocd gdbserver
0000295:INFO:board:Target type is cy8c6xxa
0000447:INFO:dap:DP IDR = 0x6ba02477
0000514:INFO:dap:AP#0 IDR = 0x84770001
0000543:INFO:dap:AP#1 IDR = 0x84770001
0000578:INFO:dap:AP#2 IDR = 0x24770011
0000603:INFO:rom_table:AP#0 ROM table #0 @ 0xf1000000 (designer=034 part=102)
0000654:INFO:rom_table:AP#1 ROM table #0 @ 0xf0000000 (designer=034 part=102)
0000684:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c0>
0000684:INFO:rom_table:  AP#1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0000721:INFO:rom_table:  [0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0000743:INFO:rom_table:  [1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0000765:INFO:rom_table:  [2]<e0002000:BPU class=14 designer=43b part=00b>
0000781:INFO:rom_table:[1]<f0002000:CTI class=9 designer=43b part=9a6 devtype=14 archid=1a14 devid=0:0:1040800>
0000810:INFO:rom_table:[2]<f0003000:MTB-M0+ class=9 designer=43b part=932 devtype=31 archid=0a31 devid=0:0:0>
0000848:INFO:rom_table:AP#2 ROM table #0 @ 0xe00ff000 (designer=034 part=102)
0000878:INFO:rom_table:[0]<e0080000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=0:0:40800>
0000894:INFO:rom_table:[1]<e008e000:TPIU-M3 class=9 designer=43b part=923 devtype=11 archid=0000 devid=0:0:ca1>
0000922:INFO:rom_table:[2]<e007f000:ROM class=1 designer=034 part=102>
0000922:INFO:rom_table:  AP#2 ROM table #1 @ 0xe007f000 (designer=034 part=102)
0000965:INFO:rom_table:  [0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000981:INFO:rom_table:  [1]<e0001000:DWT class=14 designer=43b part=002>
0001010:INFO:rom_table:  [2]<e0002000:FPB class=14 designer=43b part=003>
0001026:INFO:rom_table:  [3]<e0000000:ITM class=14 designer=43b part=001>
0001049:INFO:rom_table:  [4]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=0:0:40800>
0001065:INFO:rom_table:  [5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001082:INFO:cortex_m:CPU core is Cortex-M0+ r0p1
0001104:INFO:cortex_m:CPU core is Cortex-M4 r0p1
0001135:INFO:cortex_m:FPU present: FPv4-SP
0001151:INFO:dwt:2 hardware watchpoints
0001160:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0001184:INFO:dwt:4 hardware watchpoints
0001192:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001241:INFO:semihost:Telnet server started on port 4444
0001242:INFO:gdbserver:GDB server started on port 3333
0001268:INFO:semihost:Telnet server started on port 4445
0001270:INFO:gdbserver:GDB server started on port 3334
```

40Grit, 2019-06-02:
@flit @maclobdell

c1728p9, 2019-07-04:
I get the same failure with `CY8C6xx7`. Any idea what is happening @flit?
```
0001022:ERROR:gdb_server:uncaught exception: 'CY8C6xx7' object has no attribute 'options'
Traceback (most recent call last):
  File "c:\python2.7.10_64\lib\site-packages\pyocd\tools\gdb_server.py", line 287, in run
    with session:
  File "c:\python2.7.10_64\lib\site-packages\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "c:\python2.7.10_64\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "c:\python2.7.10_64\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "c:\python2.7.10_64\lib\site-packages\pyocd\core\coresight_target.py", line 158, in init
    seq.invoke()
  File "c:\python2.7.10_64\lib\site-packages\pyocd\utility\sequencer.py", line 201, in invoke
    resultSequence = call()
  File "c:\python2.7.10_64\lib\site-packages\pyocd\target\builtin\target_CY8C6xx7.py", line 221, in create_cy8c6xx7_core
    core0 = CortexM_CY8C6xx7(self, self.aps[1], self.memory_map, 0)
  File "c:\python2.7.10_64\lib\site-packages\pyocd\coresight\cortex_m.py", line 409, in __init__
    Target.__init__(self, session, memoryMap)
  File "c:\python2.7.10_64\lib\site-packages\pyocd\core\target.py", line 98, in __init__
    self.halt_on_connect = session.options.get('halt_on_connect')
AttributeError: 'CY8C6xx7' object has no attribute 'options'
```

flit, 2019-07-09:
This is fixed on master.

farrenv, 2019-07-09:
Feel free to close when that release is made

flit, 2019-09-04:
Fixed in v0.22.0.

closed 2019-09-04 by flit