pyocd/pyOCD #684: OSError with CMSIS-DAP v2 on Windows

AndriiLishchynskyi, 2019-06-04:
Hi,

I am observing the following error with CMSIS-DAP v2 (WinUSB) devices either running DAPLink ver. 0254 or Cypress KitProg3:

> C:\Users\anjo>pyocd list
> Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x0000017F52B96E18>
> Traceback (most recent call last):
>   File "c:\users\anjo\appdata\local\programs\python\python37\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
>     self.finalize()
>   File "c:\users\anjo\appdata\local\programs\python\python37\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
>     self._finalizer()
>   File "c:\users\anjo\appdata\local\programs\python\python37\lib\weakref.py", line 552, in __call__
>     return info.func(*info.args, **(info.kwargs or {}))
>   File "c:\users\anjo\appdata\local\programs\python\python37\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
>     obj._do_finalize_object()
>   File "c:\users\anjo\appdata\local\programs\python\python37\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
>     self._finalize_object()
>   File "c:\users\anjo\appdata\local\programs\python\python37\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
>     _lib.libusb_unref_device(self.devid)
> OSError: exception: access violation writing 0x0000000000000024_

The problem appears when running any command in pyOCD, however, commands itself do work. Looks like this is known issue with pyusb/libusb:
https://github.com/pyusb/pyusb/issues/203 
https://github.com/pyusb/pyusb/pull/227
https://github.com/libusb/libusb/issues/412

I can reproduce this on Windows only with Python3 + libusb 1.0.22. In order not to confuse user with this exception, I propose to update README with an recommendation to use libusb 1.0.21 under Windows until problem is fixed.

OS: Win10 Pro x64
Python: v3.7.3 x64
libusb: MinGW64 v1.0.22

eamars, 2019-08-07:
Can reproduce with Python 3.7 + libusb 1.0.22 + pyocd-0.21

OS: Window 10 Pro 1809 (Build 17763.557)

flit, 2019-08-17:
Yeah, unfortunately there seems to be a bug in pyusb in how it handles releasing objects in the finalizer.

Good suggestion to recommend using libusb 1.0.21 instead. Were there any fixes in 1.0.22 that could cause problems by reverting to the earlier version?

mcuee, 2019-12-07:
Have you tried latest pyusb and latest libusb?

I can no longer reproduce the issue in Windows 10, git pyusb and git libusb.
https://github.com/pyusb/pyusb/issues/203

mcuee, 2019-12-07:
https://github.com/pyusb/pyusb/pull/227
pyusb pull request #227 has been merged.

flit, 2019-12-07:
@mcuee I haven't yet tried it, but I did notice just yesterday that pyusb/pyusb#227 has been merged. (Thanks for merging the fix!) It would be really helpful to have a new pyusb release.

mcuee, 2019-12-11:
Thanks. Yes now we have a new maintainer for pyusb and we will work towards a new pyusb release soon.

flit, 2019-12-11:
@mcuee Thanks very much for taking over maintainership of pyusb!

AndriiLishchynskyi, 2019-12-11:
@mcuee, with using latest pyusb from git, the problem seems to be fixed

mcuee, 2019-12-12:
@flit Thanks. I am not really the maintainer myself, rather I am helping out on the supporting and testing side of pyusb, similar to my role in libusb, libusb-win32 and libusbk.

Jonas Malaco is the current developer/maintainer of pyusb.

flit, 2019-12-21:
@mcuee Well, either way, your help with these critical projects is greatly appreciated. 😄