pyocd/pyOCD #686: HIDError: Report size has to be 1025 elements (bytes)

shank923, 2019-06-05:
Hi, when I ran "pyocd gdbserver" command, I get this error.

> 0000253:INFO:board:Target type is cortex_m
0000253:WARNING:board:Generic 'cortex_m' target type is selected; is this intentional? You will be able to debug but not program flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
0000269:CRITICAL:__main__:uncaught exception: Report size has to be 1025 elements (bytes)
Traceback (most recent call last):
  File "build\bdist.win32\egg\pyocd\__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "build\bdist.win32\egg\pyocd\__main__.py", line 566, in do_gdbserver
    with session:
  File "build\bdist.win32\egg\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "build\bdist.win32\egg\pyocd\core\session.py", line 363, in open
    self._probe.open()
  File "build\bdist.win32\egg\pyocd\probe\cmsis_dap_probe.py", line 141, in open
    self._link.open()
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 552, in open
    self._packet_count = self._protocol.dap_info(self.ID.MAX_PACKET_COUNT)
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\cmsis_dap_core.py", line 126, in dap_info
    self.interface.write(cmd)
  File "build\bdist.win32\egg\pyocd\probe\pydapaccess\interface\pywinusb_backend.py", line 144, in write
    self.report.send([0] + data)
  File "build\bdist.win-amd64\egg\pywinusb\hid\core.py", line 1475, in send
    % self.__raw_report_size)
HIDError: Report size has to be 1025 elements (bytes)

shank923, 2019-06-05:
I'm using LPC-LINK 2 board with LPC4370 on-board MCU

[image]

flit, 2019-07-22:
This is a known issue with LPC-Link2 and pywinusb. You can try the hidapi USB backend instead, which should work. Just run 'pip install hidapi'. pywinusb is still the default for Windows mostly for historical reasons, but it changing this to hidapi might make sense.