pyocd/pyOCD #687: Can not program KW36 Dflash

jsplyy, 2019-06-15:
I have a binary which size is larger than 256kB, so need to program both Pflash and Dflash.
The KW36 memory map is as below image.
[image]
I try to use pyocd flash command to program my binary with below command:
` pyocd flash -b 0251000041114e45002d3007bcf900209e51000097969900 -a 0x00000000 "D:\mcu\wireless_uart_frdmkw.bin"`
but it report an error as below:
[image]

So, I check the flash algo file at pyocd\target\builtin\target_MKW36Z512xxx4.py
and find something is not aligned with MCU user manual, marked as red:
[image]

I want to fix this issue, so I try to generate flash algo for alisa flash and D flash from FLM files, and modify this target_MKW36Z512xxx4.py script as bolow:
[image]
I assign different flash algo for Pflahs / alisa Dflash and Dflash from 3 FLMs in Kinetis_KWxx_DFP 1.9.
I try to program this 3 part flash with a small image seperately with different target address, like;
`pyocd flash -b 0251000041114e45002d3007bcf900209e51000097969900 -a 0x10000000 "hello.bin`
`pyocd flash -b 0251000041114e45002d3007bcf900209e51000097969900 -a 0x00004000 "hello.bin`
`pyocd flash -b 0251000041114e45002d3007bcf900209e51000097969900 -a 0x00000000 "hello.bin`
They are all OK in flashing binary progress as below:
[image]

But when I try to program my big binary(larger than 256kB), it still report an error after a while as below:
[image]

jsplyy, 2019-06-15:
@flit Do you have any idea about this issue? Does pyocd support the feature to change flash algo dynamically according to different programming address?

flit, 2019-08-17:
Really sorry for the slow reply! Yes, pyOCD can have a different flash algo for each flash region. Everything you are describing should work.

There is obviously some bug when programming an image that crosses flash region boundaries. I *thought* I had fixed that a while back. What version of pyOCD are you using?

Could you create a pull request with your changes? Or attach a patch file to this issue. Also please attach the test image binary that you are using.