pyocd/pyOCD #693: flashing LPC55S69 can fail if board not in a good state

galak, 2019-06-24:
I've somehow 'bricked' a board and than try and re-flash, I get:

[galak@localhost build]$ pyocd flash zephyr/zephyr.bin 
Secure
0001090:CRITICAL:__main__:uncaught exception: 
Traceback (most recent call last):
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 475, in do_flash
    file_format=self._args.format)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/loader.py", line 160, in program
    self._loader.commit()
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/loader.py", line 504, in commit
    keep_unwritten=self._keep_unwritten)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/flash_builder.py", line 449, in program
    sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/flash_builder.py", line 625, in _compute_sector_erase_pages_and_weight
    self._analyze_pages_with_partial_read()
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/flash_builder.py", line 561, in _analyze_pages_with_partial_read
    self._enable_read_access()
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/flash_builder.py", line 221, in _enable_read_access
    self.flash.init(self.flash.Operation.VERIFY)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/flash.py", line 235, in init
    self.target.reset_and_halt(Target.ResetType.SW)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/core/coresight_target.py", line 333, in reset_and_halt
    return self.selected_core.reset_and_halt(reset_type)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/target/builtin/target_LPC55S69JBD100.py", line 295, in reset_and_halt
    xpsr = self.read_core_register('xpsr')
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/coresight/cortex_m.py", line 987, in read_core_register
    regValue = self.read_core_register_raw(regIndex)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/coresight/cortex_m.py", line 1001, in read_core_register_raw
    vals = self.read_core_registers_raw([reg])
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/coresight/cortex_m.py", line 1061, in read_core_registers_raw
    assert dhcsr_val & CortexM.S_REGRDY
AssertionError

galak, 2019-06-27:
Was also seeing this when flashing:

```
[galak@localhost zephyr]$ pyocd flash zephyr.bin 
Secure
[====================] 100%
Secure
0006128:CRITICAL:__main__:uncaught exception: 
Traceback (most recent call last):
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 475, in do_flash
    file_format=self._args.format)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/loader.py", line 160, in program
    self._loader.commit()
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/loader.py", line 504, in commit
    keep_unwritten=self._keep_unwritten)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/flash/flash_builder.py", line 472, in program
    self.flash.target.reset_and_halt()
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/core/coresight_target.py", line 333, in reset_and_halt
    return self.selected_core.reset_and_halt(reset_type)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/target/builtin/target_LPC55S69JBD100.py", line 295, in reset_and_halt
    xpsr = self.read_core_register('xpsr')
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/coresight/cortex_m.py", line 987, in read_core_register
    regValue = self.read_core_register_raw(regIndex)
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/coresight/cortex_m.py", line 1001, in read_core_register_raw
    vals = self.read_core_registers_raw([reg])
  File "/home/galak/.local/lib/python3.7/site-packages/pyocd/coresight/cortex_m.py", line 1061, in read_core_registers_raw
    assert dhcsr_val & CortexM.S_REGRDY
AssertionError
```

galak, 2019-06-27:
Here's the image I'm trying to flash that causes the failure:

[zephyr.bin.gz](https://github.com/mbedmicro/pyOCD/files/3336133/zephyr.bin.gz)

flit, 2019-08-17:
I'm able to reproduce this.

galak, 2019-09-17:
> I'm able to reproduce this.

Now the question is can you fix it? :)