pyocd/pyOCD #696: message "Waiting for a debug probe to be connected..."

pawelm87, 2019-07-08:
Hi all,

I trying debuging mbed program on nucleo board, simply blinki example. But when I started debug with pyOCD I got messages :

> 1: (131) LaunchOptions<LocalLaunchOptions xmlns='http://schemas.microsoft.com/vstudio/MDDDebuggerOptions/2014'
> 1: (139) LaunchOptions  ExePath='C:\Users\...\Desktop\workspace\debug-mbed-os\debug_vscode_gcc_arm_nucleo_l152re\debug-pyocd\BUILD\NUCLEO_L152RE\GCC_ARM\debug-pyocd.elf'
> 1: (139) LaunchOptions  WorkingDirectory='C:\Users\...\Desktop\workspace\debug-mbed-os\debug_vscode_gcc_arm_nucleo_l152re\debug-pyocd'
> 1: (139) LaunchOptions  ExeArguments='2&gt;CON 1&gt;CON &lt;CON'
> 1: (139) LaunchOptions  MIMode='gdb'
> 1: (139) LaunchOptions  MIDebuggerPath='C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin\arm-none-eabi-gdb.exe'
> 1: (139) LaunchOptions  WaitDynamicLibLoad='false'
> 1: (139) LaunchOptions  DebugServer='C:\Users\...\AppData\Local\Mbed Studio\mbed-studio-tools\python\Scripts\pyocd-gdbserver.exe'
> 1: (139) LaunchOptions  DebugServerArgs=''
> 1: (139) LaunchOptions  ServerStarted='GDB\ server\ started'
> 1: (139) LaunchOptions  FilterStderr='true'
> 1: (139) LaunchOptions  ServerLaunchTimeout='20000'
> 1: (139) LaunchOptions>
> 1: (140) LaunchOptions    <SetupCommands>
> 1: (140) LaunchOptions        <Command IgnoreFailures='false' Description=''>-environment-cd C:\Users\...\Desktop\workspace\debug-mbed-os\debug_vscode_gcc_arm_nucleo_l152re\debug-pyocd\BUILD\NUCLEO_L152RE\GCC_ARM\</Command>
> 1: (140) LaunchOptions        <Command IgnoreFailures='false' Description='connect to target'>-target-select remote localhost:3333</Command>
> 1: (140) LaunchOptions        <Command IgnoreFailures='false' Description='load file'>-file-exec-and-symbols C:\Users\...\Desktop\workspace\debug-mbed-os\debug_vscode_gcc_arm_nucleo_l152re\debug-pyocd/BUILD/NUCLEO_L152RE/GCC_ARM/debug-pyocd.elf</Command>
> 1: (140) LaunchOptions        <Command IgnoreFailures='false' Description=''>-interpreter-exec console "monitor endian little"</Command>
> 1: (140) LaunchOptions        <Command IgnoreFailures='false' Description=''>-interpreter-exec console "monitor reset"</Command>
> 1: (140) LaunchOptions        <Command IgnoreFailures='false' Description=''>-interpreter-exec console "monitor halt"</Command>
> 1: (140) LaunchOptions        <Command IgnoreFailures='false' Description=''>-interpreter-exec console "monitor arm semihosting enable"</Command>
> 1: (140) LaunchOptions        <Command IgnoreFailures='false' Description='flash target'>-target-download</Command>
> 1: (140) LaunchOptions    </SetupCommands>
> 1: (140) LaunchOptions</LocalLaunchOptions>
> 1: (285) Starting: "C:\Users\...\AppData\Local\Mbed Studio\mbed-studio-tools\python\Scripts\pyocd-gdbserver.exe" 
> 1: (5049) ->Waiting for a debug probe to be connected...
> 1: (20315) <-logout
> 1: (20326) Send Event AD7MessageEvent

When I ran command "pyocd-gdbserver" in termianal I got :

> 0001955:WARNING:gdb_server:pyocd-gdbserver is deprecated; please use the new combined pyocd tool.
> 0002027:WARNING:common:STLink and CMSIS-DAPv2 probes are not supported because no libusb library was found.
> Waiting for a debug probe to be connected...

I installed usblib:

> pip install -U libusb
> pyocd --version
> 0.21.0
> pyocd-gdbserver --version
> 0.21.0

Someone can more explain proces instal "usblib", where exactly  I need paste DLL files and which files DLL copy from folder usblib.
"Windows: download libusb from libusb.info and place the DLL in your Python installation folder next to python.exe. Make sure to use the same 32- or 64-bit architecture as your Python installation."

maratnek, 2019-10-03:
Were you resolving this question? I have some problems.

pawelm87, 2019-10-07:
No. :(

flit, 2019-10-08:
Sorry I'm just now getting to this issue!!

Please see the instructions for installing libusb in the [readme](https://github.com/mbedmicro/pyOCD/blob/master/README.md). Did you download the library from the indicated site and place the DLL next to python.exe, as stated in the instructions?

flit, 2019-10-08:
@maratnek What problems are you seeing?

maratnek, 2019-10-28:
Sorry. Yes it was under Windows. I had problem with the lib libusb.
python 2.7

flit, 2019-10-28:
@pawelm87 Is this issue resolved for you?

I'll try to add some more explicit install instructions for libusb on Windows.

maratnek, 2019-10-30:
Tnx!

pawelm87, 2019-10-30:
Hi I too have problem with libusb, I had tryed to copy lib to many places :(

I will try soon reinstall python may it will helps

lukzeg, 2020-01-24:
I'm noticing the same problems as people above on Windows 10.

----------------
The board which I'm using is:
 Nucleo-F072RB
Firmware: V2.J34.M25
----------------

Things done :

1. To fix the problem with libusb I installed libusbk.

2. Right now it does not matter, which ver of python I use[32 bit or 64 bit] , I receive:
```
PS C:\Python\Python38-32\Scripts> .\pyocd.exe list
No available debug probes are connected
```

Env setup:
Driver used by device: 
ST-Link Debug v2.1.0.0 
released :08/06/2017  

I does not notice any problems with discovering this device from openocd or under STM32 ST-Link Utility.

There is no problem to debug board through them via using arm-none-eabi-gdb.exe.

tanhengqi, 2023-04-25:
请问作者问题怎么解决的呢

tanhengqi, 2023-04-25:
[image]

tanhengqi, 2023-04-27:
thank you ,我已经在segger官网下载了驱动在ubuntu安装好了，然后可以把hex烧进板子了
[image]
在ubuntu使用JLinkRTTViewer需要在ubuntu安装一个：

tanhengqi, 2023-04-27:
[image]

tanhengqi, 2023-04-27:
我前面已经解决了，然后看到了你的回答，还是感谢你的回答

flit, 2023-04-27:
Glad it's working now!

flit, 2023-04-27:
Closing this issue. The original question related to installing the libusb driver. This should not be a problem any more, since pyocd now uses a package that directly includes the libusb dll/shared lib.

closed 2023-04-27 by flit

fghoussen, 2024-04-01:
Same issue / message "Waiting for a debug probe to be connected..." here using arduino board.

Any clue is welcome!...

```
>> git clone https://github.com/pyocd/pyOCD/
>> cd pyOCD/udev/
>> sudo cp 50-cmsis-dap.rules /etc/udev/rules.d
>> sudo udevadm control --reload
>> sudo udevadm trigger
>> sudo dmesg | tail
[434698.867330] usb 3-2: USB disconnect, device number 19
[434708.615661] synth uevent: /devices/virtual/input/input40: failed to send uevent
[434708.615669] input input40: uevent: failed to send synthetic uevent: -12
[434715.703174] usb 3-2: new full-speed USB device number 20 using xhci_hcd
[434715.853574] usb 3-2: New USB device found, idVendor=2a03, idProduct=0043, bcdDevice= 0.01
[434715.853596] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[434715.853603] usb 3-2: Product: Arduino Uno
[434715.853608] usb 3-2: Manufacturer: Arduino Srl            
[434715.853612] usb 3-2: SerialNumber: 9543135333835110B050
[434715.856454] cdc_acm 3-2:1.0: ttyACM0: USB ACM device

>> west build -p always -b arduino_uno_r4_minima blinky
>> west flash
Waiting for a debug probe to be connected...
^C

>> lsusb -v
Bus 003 Device 020: ID 2a03:0043 dog hunter AG Arduino Uno Rev3
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x2a03 dog hunter AG
  idProduct          0x0043 Arduino Uno Rev3
  bcdDevice            0.01
  iManufacturer           1 Arduino Srl            
  iProduct                2 Arduino Uno
  iSerial               220 9543135333835110B050
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003e
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               10.01
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

```

TheGuy920, 2024-06-06:
Same problem as  @fghoussen  but on windows 11 😭