pyocd/pyOCD #698: Hardware reset and halt on connect

michieldwitte, 2019-07-11:
Is there a way to connect to a device in sleep mode?
Can I issue a hardware reset, and halt the processor before trying to connect?

Currently I'm getting this error:
```
  File "C:\Python37\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "C:\Python37\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "C:\Python37\lib\site-packages\pyocd\core\coresight_target.py", line 158, in init
    seq.invoke()
  File "C:\Python37\lib\site-packages\pyocd\utility\sequencer.py", line 201, in invoke
    resultSequence = call()
  File "C:\Python37\lib\site-packages\pyocd\coresight\dap.py", line 85, in init
    self.link.connect()
  File "C:\Python37\lib\site-packages\pyocd\probe\stlink_probe.py", line 114, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "C:\Python37\lib\site-packages\pyocd\probe\stlink\stlink.py", line 256, in enter_debug
    self._check_status(response)
  File "C:\Python37\lib\site-packages\pyocd\probe\stlink\stlink.py", line 294, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (11): Debug power error
```

I've had this issue before using OpenOCD
Changing the following line in the device config file resolved it.
```
from: reset_config srst_nogate
to:   reset_config trst_and_srst srst_nogate connect_assert_srst
```

When first connecting and disconnecting the stlink-utility and then running the script there are no issues.

michieldwitte, 2019-07-20:
I've been able to create a quick workaround for this issue. (https://github.com/michieldwitte/pyOCD/commit/2ad02e2c4fd0e390edc971dd081b175d5a3942d6)

flit, 2019-07-27:
Addressing this through some connect options is on the todo list.