pyocd/pyOCD #707: Memory error when reading on Python 2.7 32 bit

ashy-cypress, 2019-07-31:
When I read from big portion of data from external memory of cypress PSoC6 kits I get memory error, issue is observed only on  Python 2.7 32 bit on Windows.
Log:
(venv) D:\CyP\pyOCD\pyocd_3_7\pyocd>pyocd cmd -c savemem 0x18000000 0x3000000 D:\1.bin
1161952:ERROR:_main_:uncaught exception:
Traceback (most recent call last):
File "d:\cyp\pyocd\pyocd_3_7\pyocd\pyocd_main_.py", line 300, in run
self._commands[self._args.cmd]()
File "d:\cyp\pyocd\pyocd_3_7\pyocd\pyocd_main_.py", line 491, in do_commander
PyOCDCommander(self._args, cmds).run()
File "d:\cyp\pyocd\pyocd_3_7\pyocd\pyocd\tools\pyocd.py", line 647, in run
result = self.command_list[cmd](args)
File "d:\cyp\pyocd\pyocd_3_7\pyocd\pyocd\tools\pyocd.py", line 915, in handle_savemem
data = bytearray(self.target.aps[self.selected_ap].read_memory_block8(addr, count))
File "d:\cyp\pyocd\pyocd_3_7\pyocd\pyocd\core\memory_interface.py", line 89, in read_memory_block8
res += conversion.u32le_list_to_byte_list(mem)
File "d:\cyp\pyocd\pyocd_3_7\pyocd\pyocd\utility\conversion.py", line 45, in u32le_list_to_byte_list
res.append((x >> 16) & 0xff)
MemoryError

flit, 2019-08-17:
That's Python literally running out of memory. See [MemoryError](https://docs.python.org/3.7/library/exceptions.html#MemoryError) in the Python docs.

Try using 64-bit Python and see if it works.

0xc0170, 2019-08-19:
@ashy-cypress Please let us know if that fixes it.

@flit if it does, is this something to be in the docs or captured in the code (our conversion if it catches this MemoryError, prints more meaningful msg or not possible?

ashy-cypress, 2019-08-19:
@0xc0170 Yes, no problem with 64-bit Python, error ocurs only with Python 32.

flit, 2019-08-19:
Sure, we could catch `MemoryError` exceptions and report a more helpful message. Probably the only place it makes sense to do so is in the `pyocd cmd` REPL. There aren't many other situations where the user can directly cause such large allocations.

I did think about handling very large allocations in the code. It would require either custom code per command (i.e., the `savemem` command) and/or a complex buffer management scheme. Given that 64-bit systems are the norm now, it's just not worth the effort. There are a lot of higher priority issues. It *might* be worth extending `savemem` and `loadmem` to handle large sizes without attempting a general fix.

Something else that will help a lot is moving PyOCD from representing data as lists of integers to using `bytearray` or `array.array`. These types are far more efficient and performance for large blocks of data. This change is tracked by #566.