pyocd/pyOCD #708: smart flash algorithm leaves unprogrammed areas on stm32f091

rmu75, 2019-07-31:
Programming a STM32F091RC with pyocd and the "smart flash" algorithm (sector erase) can result in non-programmed areas on an stm32f091. Datasheet of the device says flash page size is 2048bytes, but pyocd assumes a 1024 byte page size. 1024bytes next to programmed page remain erased. This makes for a strange and frustrating debugging experience because the erased page "jumps around" with each invocation.

```
0005159:INFO:board:Target type is stm32f091rc
0005391:INFO:dap:DP IDR = 0x0bb11477 (v1 MINDP rev0)
0005525:INFO:ap:AP#0 IDR = 0x04770021 (AHB-AP var2 rev0)
0005688:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=440)
0005741:INFO:rom_table:[0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0005770:INFO:rom_table:[1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0005800:INFO:rom_table:[2]<e0002000:BPU class=14 designer=43b part=00b>
0005853:INFO:cortex_m:CPU core #0 is Cortex-M0 r0p0
0005963:INFO:dwt:2 hardware watchpoints
0006045:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0006062:INFO:server:Semihost server started on port 4444
0006063:INFO:gdbserver:GDB server started on port 3333
0013595:INFO:gdbserver:One client connected!
0014897:INFO:gdbserver:Semihosting enabled
0015062:INFO:gdbserver:Attempting to load Argon
0015109:INFO:gdbserver:Attempting to load FreeRTOS
0015157:INFO:gdbserver:Attempting to load Zephyr
0015210:INFO:gdbserver:Attempting to load RTX5
[====================] 100%
0024689:INFO:loader:Erased 1024 bytes (1 sector), programmed 1024 bytes (1 page), skipped 148480 bytes (145 pages) at 30.80 kB/s
0025923:WARNING:gdbserver:Invalid mon command 'b'reg''
0074545:INFO:gdbserver:Client disconnected!
0074659:INFO:server:Semihost server stopped
```

flit, 2019-08-17:
The STM32F091RC is not built-in to PyOCD, so you must be using a CMSIS Pack. I checked the STM32F0xx_256.FLM flash algo in Keil.STM32F0xx_DFP.2.0.0.pack and it specifies a sector size of 1024 bytes. If this is incorrect, please contact Keil to report the issue. PyOCD naturally depends on the CMSIS Packs, provided by other vendors, to be correct.

rmu75, 2019-08-19:
I was under the impression that both mbed and keil are part of ARM. I tried posting in the arm community Keil forum. If there is a more appropriate place to report this pls advise.

flit, 2019-08-19:
Indeed they are. But Arm is a quite large company. (Same with any large multinational corporation.)

I'm happy for issues with CMSIS Packs to be reported here because that allows others to see them. But ultimately, the best thing to do is create a [Keil support request](http://www2.keil.com/support/silver?P=&V=&S=). It's better than me doing it because it shows there is a direct customer need.

rmu75, 2019-08-20:
I opened a support case. Will report back here for future reference. Closing this issue for now. Thanks.

closed 2019-08-20 by rmu75

jeromecoutant, 2019-08-20:
ST_INTERNAL_REF 70900

jeromecoutant, 2020-05-07:
Hi
Keil.STM32F0xx_DFP.2.1.0.pack is now available and should correct the issue