pyocd/pyOCD #712: Not all devices show progress during flash

thegecko, 2019-08-13:
OS: MacOS 10.12
pyOCD: 0.21.0
Devices: LPC1768

Usually during a flash operation, progress is shown on stdout in the format:

```
[===                        ] 20%
```

For some target devices, this progress is not updated and simply shows no progress until completed:

```
[                           ]
```

flit, 2019-08-17:
That's almost certainly due to the initial analysis scan to determine unmodified pages. Progress is not updated during analysis. So if the flash is entirely unmodified, you'll get the second case you describe.

You can verify this by enabling info logs during flash programming. You'll see a log message like, "Erased x bytes (y), programmed n bytes (m), skipped u bytes (v) at q kB/s".

I'll try to think of some way to improve this. (I did make one attempt earlier but it didn't work out.)

flit, 2019-10-12:
What size of application are you testing with? The flash pages on this device are quite large: 4 kB pages for the first 64 kB, then 32 kB pages for the remaining 448 kB. Progress is only updated as each page is erased and programmed (there's no way around that). So depending on the download size, it may appear as it there is no progress until it's complete, or progress is updated at a very coarse granularity.

thegecko, 2019-10-13:
This is using mbed-os-example-blinky and only observed on this target.
As this target is particularly slow at flashing, the lack of progress is quiet apparent.

thegecko, 2019-10-13:
This is using mbed-os-example-blinky and only observed on this target.
As this target is particularly slow at flashing, the lack of progress is very apparent.

flit, 2019-10-15:
I just ran a quick test and don't see any problem here.

Specifically, I first erased the device, then ran `pyocd flash binaries/l1_lpc1768.bin` to program the LPC1768 test binary included with pyOCD. Progress was shown as expected.

Also tested with mbed-os-example-blinky and progress was shown as expected.

thegecko, 2019-10-17:
I can also confirm this is working again. Very strange

closed 2019-10-17 by thegecko

flit, 2019-10-17:
The only suggestion I have is that it could be related to the combination of firmware size and data, previous flash contents, and flash programming options. If you encounter it again, capturing all of those is important for reproduction.