pyocd/pyOCD #714: Intermittent 'DP Wait' error on St boards causes bad state

thegecko, 2019-08-14:
OS: MacOS 10.14.5
Mbed OS: 5.13.1
Compiler: GCC or AC6.11
pyOCD: 0.21.0
Targets: ST boards (seen on F411, F413, F042K6, F475IOT)

When deploying or debugging ST targets, occasionally a `DP Wait` error is thrown. This puts the board into a state which still allows updates via mass storage device but any pyOCD interaction continues to show this error even after a hard reset.

With reference to https://github.com/mbedmicro/pyOCD/issues/623, the environment isn't in a VM and all firmware is up to date.

It would be good to understand how to recover from this state.

flit, 2019-08-17:
Are you using the most recent STLink firmware?

An infinite wait response normally means a stuck transaction. But pyOCD has code to issue an abort, which should recover from this state (and does, in the few cases I've seen).

thegecko, 2019-08-21:
Have updated to the latest ST firmware and can confirm this problem still exists.

flit, 2019-10-01:
Update: I'm now able to reproduce this issue. It requires power cycling the board after programming firmware. The `connect_mode=under-reset` option on my [feature/connect_mode](https://github.com/flit/pyOCD/tree/feature/connect_mode) branch allows you to connect successfully. But there are still questions about what's causing the problem that need to be resolved.

thegecko, 2019-10-01:
Thanks @flit

jeromecoutant, 2019-10-02:
Hi
When you connect pyOCD, target is already on I think ?

With latest mbed-os version, target could now is sleep or deep sleep.
So I am wondering how pyOCD can sync with target to attach it during a "wake up" ?

@schstm

LMESTM, 2019-10-10:
Note: mbed-os offers a DEBUG profile that would prevent deep-sleep to be entered and can be used for easy debug.

stephwongwt, 2019-11-19:
> Update: I'm now able to reproduce this issue. It requires power cycling the board after programming firmware. The `connect_mode=under-reset` option on my [...]

I'm having the same issue but I cannot find your linked branch anymore.

flit, 2020-11-16:
Closing this issue.

There were two fixes implemented long ago:
- PyOCD was failing to abort a stuck transfer.
- Implemented the `connect_mode` option, now accessible via `--connect/-M` on the command line.

If this is still a problem for anyone, please create a new issue.

jeromecoutant, 2021-10-20:
> Closing this issue.

still opened :-)

flit, 2021-10-20:
Oops, thanks @jeromecoutant :)

closed 2021-10-20 by flit