pyocd/pyOCD #715: LPC55S69 flash freezes

juhhov, 2019-08-15:
The freeze occurs very often. Flash never continues when at this state.
Version in use: ddf36148a959fe27598c6d244b1039cd7b888293

```
$ pyocd flash -vvv -u 023600001985ea0a00000000000000000000000097969905 ~/mbed_cliapp_LPC55S69.hex 
0000286:DEBUG:session:Project directory: /home/raas/pyOCD
0000288:INFO:board:Target type is lpc55s69
0000297:DEBUG:dap_access_cmsis_dap:CMSIS-DAP probe 023600001985ea0a00000000000000000000000097969905 firmware version: 1.10
0000305:DEBUG:sequencer:Running task load_svd
0000305:DEBUG:sequencer:Running task create_flash
0000306:DEBUG:sequencer:Running task dp_init
0000344:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0000344:DEBUG:sequencer:Running task power_up
0000346:DEBUG:sequencer:Running task find_aps
0000359:DEBUG:sequencer:Running task create_aps
0000360:DEBUG:sequencer:Running task create_ap.0
0000365:INFO:ap:AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
0000369:DEBUG:ap:AP#0 default HPROT=3 HNONSEC=1
0000371:DEBUG:ap:AP#0 implemented HPROT=f HNONSEC=1
0000380:DEBUG:sequencer:Running task create_ap.1
0000384:INFO:ap:AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0000386:DEBUG:ap:AP#1 default HPROT=3 HNONSEC=1
0000388:DEBUG:ap:AP#1 implemented HPROT=f HNONSEC=1
0000393:DEBUG:sequencer:Running task create_ap.2
0000399:INFO:ap:AP#2 IDR = 0x002a0000 (proprietary)
0000399:DEBUG:sequencer:Running task init_ap_roms
0000399:DEBUG:sequencer:Running task init_ap.0
0000432:INFO:rom_table:AP#0 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0000440:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0000440:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000450:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000481:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000508:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000562:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000600:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=0:0:ca0>
0000605:DEBUG:sequencer:Running task set_ap1_nonsec
0000610:DEBUG:sequencer:Running task init_ap.1
0000630:INFO:rom_table:AP#1 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0000637:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0000637:INFO:rom_table:  AP#1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000644:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000649:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000654:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000659:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000664:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=0:0:ca0>
0000664:DEBUG:sequencer:Running task create_cores
0000670:INFO:cortex_m_v8m:CPU core #0 is Cortex-M33 r0p3 (security ext present)
0000678:INFO:cortex_m:FPU present: FPv5-SP
0000685:INFO:cortex_m_v8m:CPU core #1 is Cortex-M33 r0p3
0000690:DEBUG:sequencer:Running task enable_traceclk
0000694:DEBUG:sequencer:Running task create_components
0000694:DEBUG:coresight_target:Creating DWT component
0000698:INFO:dwt:4 hardware watchpoints
0000699:DEBUG:coresight_target:Creating BPU component
0000701:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0000701:DEBUG:fpb:fpb has been disabled
0000702:DEBUG:coresight_target:Creating ITM component
0000707:DEBUG:coresight_target:Creating TPIU-M33 component
0000709:DEBUG:coresight_target:Creating DWT component
0000713:INFO:dwt:4 hardware watchpoints
0000713:DEBUG:coresight_target:Creating BPU component
0000715:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0000715:DEBUG:fpb:fpb has been disabled
0000716:DEBUG:coresight_target:Creating ITM component
0000721:DEBUG:coresight_target:Creating TPIU-M33 component
0000723:DEBUG:sequencer:Running task check_for_cores
0000723:DEBUG:sequencer:Running task notify
0001111:DEBUG:flash_builder:Analyze time: 0.000981
0001111:DEBUG:flash_builder:Using double buffer sector erase program
[                    ]   0%0001209:DEBUG:manager:added=[] removed=[]
0001209:DEBUG:manager:bps after flush={}
0001209:DEBUG:manager:added=[] removed=[]
0001209:DEBUG:manager:bps after flush={}
```

RomanSaveljev, 2019-08-27:
Hi, @flit did you have time to look into this issue?

flit, 2019-08-29:
My apologies, I have not. But I've been using the LPC55S69 lately and for some reason have not come across this issue.

What is the state of the device prior to flashing?
Does it still fail like this if you run 'pyocd erase --chip' before this test?

juhhov, 2019-09-04:
This usually reproduces when another flash is started right after previous has finished.

JanneKiiskila, 2019-12-13:
@flit - any chance of getting some runtime for this?

JanneKiiskila, 2020-04-28:
@mmahadevan108 @maclobdell  - can we get some runtime for this? This is severely hampering the usage of this board in the CI/automated testing setup.