pyocd/pyOCD #717: Error when disconnecting debug from LPC1768

thegecko, 2019-08-20:
OS: MacOS 10.14.5
Mbed OS: 5.13.1
Target: LPC1768

When disconnecting a debug session from the LPC1768, an error is thrown and the debug session does not disconnect. This has not been observed on other targets.

```
0017315:ERROR:board:link exception during target disconnect:
Traceback (most recent call last):
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/board/board.py", line 96, in uninit
    self.target.disconnect(resume)
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 244, in disconnect
    core.disconnect(resume)
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 505, in disconnect
    self.resume()
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 958, in resume
    self.flush()
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/core/target.py", line 142, in flush
    self.session.probe.flush()
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 237, in flush
    self._link.flush()
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 637, in flush
    self._read_packet()
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 860, in _read_packet
    raw_data = self._interface.read()
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/probe/pydapaccess/interface/hidapi_backend.py", line 113, in read
    return self.device.read(self.packet_size)
```

flit, 2019-10-12:
What is the traceback for the F466ZE target? It will be different (assuming you're using STLink for it).

thegecko, 2019-10-17:
@flit, the target we saw this problem with was actually `NUCLEAO_F446ZE` (not 466 - apologies, my typo).
I've been unable to get this to crash with PyOCD 0.21.0, so have updated the ticket to reflect this is an LPC168 issue only.

flit, 2019-10-17:
Thanks for the update!

flit, 2019-10-21:
I've not been able to reliably reproduce this issue. I did see an exception once or twice on disconnect, but not exactly the same.

Can you provide a command sequence to reproduce this from the command line, starting from blank flash? Thanks.