pyocd/pyOCD #719: Issue to connect dual core MCU

jeromecoutant, 2019-08-27:
Hi

Trying to connect a STM32H747I-DISCO board...
- dual core M7 + M4
- STLink V3

````
$ pyocd gdbserver
0002245:INFO:board:Target type is stm32h747xihx
0002282:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0002299:ERROR:dap:Exception reading AP#0 IDR: STLink error (5): No device connected
0002302:ERROR:dap:Exception reading AP#1 IDR: STLink error (5): No device connected
0002308:ERROR:dap:Exception reading AP#2 IDR: STLink error (5): No device connected
0002312:ERROR:dap:Exception reading AP#3 IDR: STLink error (5): No device connected
0002319:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "build\bdist.win32\egg\pyocd\__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "build\bdist.win32\egg\pyocd\__main__.py", line 571, in do_gdbserver
    with session:
  File "build\bdist.win32\egg\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "build\bdist.win32\egg\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "build\bdist.win32\egg\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "build\bdist.win32\egg\pyocd\core\coresight_target.py", line 162, in init
    seq.invoke()
  File "build\bdist.win32\egg\pyocd\utility\sequencer.py", line 201, in invoke
    resultSequence = call()
  File "build\bdist.win32\egg\pyocd\core\coresight_target.py", line 242, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
DebugError: No cores were discovered!
````

0xc0170, 2019-09-18:
One thing that just came to me, this error is being reported multiple times in the last 2 weeks (no cores were discovered): #729 and #723

MarceloSalazar, 2019-10-30:
Note this also impacts the H743 (single-core) platform.

jeromecoutant, 2019-10-31:
Hi

Test with latest version (73710a27b3fdfc730c981ec2c3b7ae15cf09dc92 commit):

- NUCLEO_H743ZI2: OK
- DISCO_H747I: OK with a M7 application

Can you try on your side to confirm ?

jeromecoutant, 2019-10-31:
@erwango

flit, 2019-10-31:
Huh. I'll test it out.

MarceloSalazar, 2019-11-13:
@flit do you have any comments?
Can we close this ticket, as @jeromecoutant commented that he's able to connect to the platforms?
There is a separate discussion on debugging dual-core MCUs, but IMO that's outside the scope of this ticket.

flit, 2019-11-13:
I have not had a chance to test it yet.

MarceloSalazar, 2019-11-27:
Closing as agreed. Will create new one about dual-core support.

closed 2019-11-27 by MarceloSalazar