pyocd/pyOCD #722: Option to disable overriding of security

embed, 2019-09-06:
When flashing a Kinetis device, pyOCD overrides the Flash Configuration Field with safe defaults, in order to avoid breaking/bricking the device. This is implemented as the method `override_security_bits`.

This behaviour should be optional (and enabled by default, to be safe) and switchable via a command line option, so developers can implement their custom security settings.

flit, 2019-09-06:
Good idea, thanks. This would be handled by a user option, so from the command line it would look something like `-Oflash.override_security_bits=no`.

embed, 2019-11-14:
I can implement this but I am not familiar with the source. Can someone point me where the command line options are listed and parsed?

flit, 2019-11-15:
The best way to implement this is to add a user option as described in the comment above.

1. First add the option definition:
    - Add the definition to `pyocd/core/options.py`.
    - Document the option in `docs/configuration.md`.

    This documents the option, sets its type, and sets the default value. The option can now be set on the command line with the `-O` argument.

2. Next, handle the option in the `Flash` class methods by skipping the call to `self.override_security_bits(...)` when enabled. To get the option value, use the `self.target.session.options.get('flash.override_security_bits')`.

Unfortunately, there's not really a good way to automatically test it right now, as target-specific tests aren't supported (yet). So no additions to the functional tests are required.