pyocd/pyOCD #723: "No cores were discovered" with nrf52840

44670, 2019-09-07:
I have tried stlink and a cmsis-dap dongle, the results are same.

pyocd commander -v --target nrf52840
0000585:INFO:board:Target type is nrf52840
0000610:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0000628:INFO:ap:AP#0 IDR = 0x23000000 (proprietary)
0000631:INFO:ap:AP#1 IDR = 0x02880000 (proprietary)
Exception while initing board: No cores were discovered!
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\tools\pyocd.py", line 704, in connect
    self.session.open(init_board=not self.args.no_init)
  File "c:\python27\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "c:\python27\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "c:\python27\lib\site-packages\pyocd\core\coresight_target.py", line 162, in init
    seq.invoke()
  File "c:\python27\lib\site-packages\pyocd\utility\sequencer.py", line 201, in invoke
    resultSequence = call()
  File "c:\python27\lib\site-packages\pyocd\core\coresight_target.py", line 242, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
DebugError: No cores were discovered!

44670, 2019-09-07:
pyocd -V
0.22.0

44670, 2019-09-07:
Just found that the nrf52840 chip is locked.  Issuing a ERASEALL task by writing to AP #1 could solve this problem.

https://sourceforge.net/p/openocd/mailman/message/36600551/

brathis, 2022-10-05:
Hi,

I'm having an issue that appears to be very closely related. Target is an **nRF52840** also,  The issue occurs while creating a new debug probe session while the chip has the `APPROTECT` bit set. pyOCD 0.34.1, Python 3.10.

```
Traceback (most recent call last):
  File "/home/xxx/python_workspace/zzz/example.py", line 266, in <module>
    example()
  File "/home/xxx/python_workspace/zzz/example.py", line 70, in example
    with ptu.getDebugProbeSession(target="nrf52840") as debugProbeSession:
  File "/home/xxx/.local/share/virtualenvs/yyy/lib/python3.10/site-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
  File "/home/xxx/.local/share/virtualenvs/yyy/lib/python3.10/site-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/home/xxx/.local/share/virtualenvs/yyy/lib/python3.10/site-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/home/xxx/.local/share/virtualenvs/yyy/lib/python3.10/site-packages/pyocd/core/soc_target.py", line 147, in init
    seq.invoke()
  File "/home/xxx/.local/share/virtualenvs/yyy/lib/python3.10/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/xxx/.local/share/virtualenvs/yyy/lib/python3.10/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/xxx/.local/share/virtualenvs/yyy/lib/python3.10/site-packages/pyocd/target/family/target_nRF52.py", line 106, in check_flash_security
    self.dp.create_1_ap(AHB_AP_NUM)
AttributeError: 'DebugPort' object has no attribute 'create_1_ap'
```

The offending code is in `pyocd/target/family/target_nRF52.py` in `NRF52.check_flash_security()`. There, it checks if the target is locked and if the `auto_unlock` option is set. If it is, it performs a mass erase, which succeeds. Then, it calls `self.dp.create_1_ap(AHB_AP_NUM)`, which is a method that does not exist. If I remove this line, I then get the "No cores discovered" error above. It seems that this line was introduced to patch the core discovery issue, but at some point the `create_1_ap()` method was moved.