pyocd/pyOCD #725: LPC1768 is not discovered on Windows 10

arekzaluski, 2019-09-09:
OS: Windows 10 64bit
Target: LPC1768 (https://os.mbed.com/platforms/mbed-LPC1768/)
pyOCD version: 0.22.0 (issue exists also on older versions)
firmware: rev 141212 (https://os.mbed.com/handbook/Firmware-LPC1768-LPC11U24)

Target can't be discovered by `pyocd gdbserver` or `pyocd flash`.
mbed-ls discovers target correctly:

 platform_name | platform_name_unique | mount_point | serial_port | target_id                                                        | daplink_version |
|---------------|----------------------|-------------|-------------|------------------------------------------------------------------|-----------------|
| LPC1768       | LPC1768[0]           | F:          | unknown     | 101000000000000000000002F7F22A7723b2b78374ccbc18aac7c0bf7986522f | unknown         |

flit, 2019-10-12:
It looks like this bug is caused by the mbed-LPC1768 device's firmware, probably the device descriptors. Under Windows 10, only the USB mass storage interface is enabled. You can see an artifact of this in the mbed-ls output, where it shows "unknown" for the serial port.

A workaround is to manually install the Mbed Composite Driver using Device Manager. You might be able to install the Mbed Serial Driver using the installer with the device connected instead of doing it manually via Device Manager. I haven't tried this.

1. Install the [Mbed Serial Driver](https://os.mbed.com/handbook/Windows-serial-configuration) without the device connected. Drivers are placed under `C:\Program Files\mbed\drivers\`.
2. Open Device Manager and disclose "Universal Serial Bus controllers".
3. Connect Mbed LPC1768. You'll see a new "USB Mass Storage Device" appear in Device Manager under the USB controllers.
4. In Device Manager, get properties on the new USB Mass Storage Device.
5. Select the Details tab, select the "Hardware Ids" property, and verify that the string contains "VID_0D28&PID_0204". This is just to make sure it's the correct device. If not, find the right device before continuing.
6. Select the Driver tab, click Update Driver.
7. Click "Browse my computer for driver software".
8. On the next screen, click the Browse button or enter "C:\mbed\drivers\" in the path field. Then click Next.

I think there was another screen where you had to click and confirm to use the Mbed Composite Driver (the only one in the list it shows). But I don't seem to be able to revert back to the default driver, so I'm unable to reproduce that step.

After switching drivers, you'll see several new devices in Device Manager:
- "mbed Composite Device" under "Universal Serial Bus controllers"
- "HID-compliant vendor-defined device" under "Human Interface Devices"
- "mbed Serial Port (COMx)" under "Ports (COM & LPT)".

After this, "pyocd list" shows the device and is able to connect to it.

It looks like Windows is only activating the first USB interface provided by the device, which is MSC. This might caused because the device class is set to 2 (Communication) rather than Composite or Miscellaneous–Interface Association.

closed 2019-10-12 by flit

thegecko, 2019-10-30:
The Mbed Serial Driver is well out of date and is unstable on Windows 10, so we don't recommend using it. Unfortunately this means it's not a valid workaround for this issue.
Understand this isn't a PyOCD issue, however does it make sense to keep this issue open until the firmware or driver is fixed?