pyocd/pyOCD #727: CMSIS Pack for STM32F7/H7 broken?

Funky185540, 2019-09-12:
Hi there,

I'm not sure if this is the right place to post this (guess so from reading the [docs](https://github.com/mbedmicro/pyOCD/blob/master/docs/target_support.md) ) but I suspect that currently the CMSIS packs for the STM32F7 and STM32H7 aren't functional.

Here's the output of `pyocd pack -u`:
```shell
> pyocd pack -u
0000419:INFO:__main__:Updating pack index...
Sep 12 09:18:27.751 ERRO download of "https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc" failed: unexpected end of file
Sep 12 09:18:27.771 ERRO download of "https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc" failed: unexpected end of file
Sep 12 09:18:27.771 ERRO download of "https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc" failed: unexpected end of file
Sep 12 09:18:28.351 WARN expecting </META> found </HEAD>
Sep 12 09:18:29.506 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.516 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.516 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.516 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.516 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.516 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.516 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.516 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.526 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.526 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.526 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:29.526 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
Sep 12 09:18:31.485 WARN No Core found!, Package: CMSIS, Vendor: ARM
Sep 12 09:18:31.595 WARN invalid digit found in string, Package: TLE984x_DFP, Vendor: Infineon
```

Thus, I am not able to use pyocd for these targets. I have tried downloading the pack files manually from the site (which, as such, worked) and supplying them to pyocd through the `--pack` option, but that didn't work either. Here is the output:
```shell
> pyocd gdb --pack C:/Users/ahartmann/Downloads/PyOCD_PACKS/Keil.STM32F7xx_DFP.2.12.0.pack
  #   Probe                           Unique ID
----------------------------------------------------------------
  0   NUCLEO-F303RE [stm32f303re]     066FFF504955657867091142
  1   NUCLEO-F767ZI [stm32f767zi]     0676FF504955857567171922
  2   NUCLEO-H743ZI [stm32h743zitx]   066CFF545257717867101851
  3   NUCLEO-L073RZ [stm32l073rz]     0674FF333036434B43055429

Enter the number of the debug probe or 'q' to quit> 1
0004324:INFO:board:Target type is stm32f767zi
0004474:INFO:dap:DP IDR = 0x5ba02477 (v2 rev5)
0004634:INFO:ap:AP#0 IDR = 0x74770001 (AHB-AP var0 rev7)
0004634:WARNING:rom_table:Invalid coresight component, cidr=0x0
0004634:WARNING:rom_table:Warning: ROM table @ 0xe00fd000 has unexpected CIDR component class (0x0)
0004634:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "c:\users\ahartmann\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\users\ahartmann\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\__main__.py", line 571, in do_gdbserver
    with session:
  File "c:\users\ahartmann\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "c:\users\ahartmann\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "c:\users\ahartmann\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "c:\users\ahartmann\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\coresight_target.py", line 162, in init
    seq.invoke()
  File "c:\users\ahartmann\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\utility\sequencer.py", line 201, in invoke
    resultSequence = call()
  File "c:\users\ahartmann\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\coresight_target.py", line 242, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!
```

For reference I tried a STM NUCLEO-F303RE that I had lying around. That shows up in my local pack index and is correctly recognized by PyOCD. No issues here.   
I also tried using an older version of the STM32F7 pack files (Version 2.11.0 instead of the current 2.12.0), but to no extent. The result is the same.

Can anybody reproduce this issue?  
Or tell me how to reach the pack vendor (ST, I guess?) so I can report to them?

I'm using the Python 3 verion of PyOCD, PyOCD Version is 0.22.0.  
Host OS is Windows 10 for anybody who might find it to be of interest.

whj4674672, 2020-06-18:
I met the same problem as you. Have you solved it？

Funky185540, 2020-06-25:
> I met the same problem as you. Have you solved it？

Not really. I just went with downloading the CMSIS pack manually and erasing the STMs flash memory before flashing and/or using PyOCD. It's more of a workaround than a true "solution", but once you automate it you'll forget about it really.
I haven't looked into the issue since December because I'm working on other projects now. So unfortunately I'm not able to tell you any more about it...

alexc-cazacu, 2022-04-12:
Hi, I have the same issue, how did you supply the CMSIS packs manually?

flit, 2022-04-12:
Please see the [target support](https://pyocd.io/docs/target_support.html) documentation for how to do this. In short, download packs from [the pack list](https://www.keil.com/dd2/pack/), then use `--pack` to use with pyocd.

Fyi, the pack download problem described by this issue is actually fixed. But there is a problem where pyocd/cmsis-pack-manager is too aggressive in parallelizing the index download. This triggers the servers' DoS protection and the requests are blocked. Only Keil and NXP servers seem to have this response. Documented in pyocd/cmsis-pack-manager#190.

alexc-cazacu, 2022-04-13:
I'm trying to build and run code through MBed Studio, and when I try to build any program it compiles successfully, but when I try to run it I get the 'No cores were discovered' error. I don't really know if I can update the pyocd that is built in Mbed Studio unfortunately.

flit, 2022-04-13:
The STM32 F7xx and H7xx (and a few other recent families, I think) require DBGMCU registers to be written to enable clocks in order to be fully discoverable by debuggers. This is handled in CMSIS Packs via "debug sequences". Support for debug sequences is in the works for pyocd (at a painfully slow pace, though almost done). Until that features is merged, you'll have to configure DBGMCU yourself in a [pyocd user script](https://pyocd.io/docs/user_scripts.html). It's just like any other debugger's init scripts, but in Python. See the Debug chapter of your STM32 device's RM, specifically the DBGMCU.CR register, for what settings to apply to enable all clocks.

I can't really help for the specifics of Mbed Studio integration since I'm not familiar with it.