pyocd/pyOCD #728: IntervalTree error when debugging nRF-52840 using elf built with GCC

thegecko, 2019-09-16:
PyOCD: 0.21.0
Mbed OS: 5.13.1
OS: MacOS 10.14.5
Compiler: GCC (works fine on AC6)

When debugging [blinky](https://github.com/ARMmbed/mbed-os-example-blinky), the following error occurs:

```
> ./pyocd gdbserver --target nrf52840 --elf "/Users/thegecko/Mbed Programs/mbed-os-example-blinky/BUILD/NRF52840_DK/GCC_ARM/mbed-os-example-blinky.elf"
```

```
0002303:CRITICAL:__main__:IntervalTree: Null Interval objects not allowed in IntervalTree: Interval(85192, 85192, )
Traceback (most recent call last):
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/__main__.py", line 569, in do_gdbserver
    session.board.target.elf = self._args.elf
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 91, in elf
    self.cores[0].set_target_context(FlashReaderContext(self.cores[0].get_target_context(), self._elf))
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/debug/elf/flash_reader.py", line 31, in __init__
    self._build_regions()
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/debug/elf/flash_reader.py", line 39, in _build_regions
    self._tree.addi(start, start + length, sect)
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/intervaltree/intervaltree.py", line 343, in addi
    return self.add(Interval(begin, end, data))
  File "/Users/thegecko/mbed-studio-tools/python/lib/python2.7/site-packages/intervaltree/intervaltree.py", line 326, in add
    " {0}".format(interval)
ValueError: IntervalTree: Null Interval objects not allowed in IntervalTree: Interval(85192, 85192, )
```

When the elf file isn't specified, debugging works fine.

thegecko, 2019-09-18:
Perhaps a guard could be added to ignore elf loading errors so that debugging can continue albeit without the elf context?

flit, 2019-09-18:
The guard is a good idea. Of course, it should also not try to insert null intervals into the interval tree! 😅