pyocd/pyOCD #739: How to debug 'TransferError' for a new port?

oclyke, 2019-10-09:
I am adding an [Ambiq Apollo3 Blue](https://www.ambiqmicro.com/mcu/) target to pyOCD but have run into some trouble that I could really use expert advice on. 

The question, once the symptoms are explained, will be this:

*Where should I begin to debug this problem? Does it seem to be a SW or HW problem? Could it be related to the flash algorithm that I am using? Is the target (Cortex-M4 on Ambiq Apollo3) being properly supported (can Coresight find all the required ports)?*

## Steps to Reproduce:
* build an [artmbed](https://github.com/sparkfun/artmbed) development board (design *very* closely copied from [micro:bit schematic](https://github.com/bbcmicrobit/hardware/blob/master/SCH_BBC-Microbit_V1.3B.pdf)) (you can just use the [docs](https://github.com/sparkfun/artmbed/tree/master/docs) as a reference rather than actually building one - schematic and top, bottom, and combined layout PDFs provided)
* use the Apollo3.FLM file from the [Ambiq Device Family Pack](http://s3.ambiqmicro.com/pack/AmbiqMicro.Apollo_DFP.1.1.0.pack) along with [FlashAlgo](https://github.com/mbedmicro/FlashAlgo)'s ```generate_blobs.py``` to generate flash algorithm files
``` bash
python generate_blobs.py Apollo3.FLM 
Flash Device:
  name=Ambiq Micro Apollo 3 Blue
  version=0x101
  type=1
  start=0x0
  size=0x100000
  page_size=0x2000
  value_empty=0xff
  prog_timeout_ms=500
  erase_timeout_ms=100
  sectors:
    start=0x0, size=0x2000
```
* use the [ambiq-apollo3](https://github.com/sparkfun/DAPLink/tree/ambiq-apollo3) branch of my fork of DAPLink and uVision to build the projects ```kl26z_bl``` and ```kl26z_artmbed_if``` (flash bootloader firmware onto interface mcu with external debug probe, then drag + drop interface firmware onto the ```MAINTENANCE``` drive)
* use the [ambiq-apollo3](https://github.com/sparkfun/pyOCD/tree/ambiq-apollo3) branch of my pyOCD fork to run ```test/basic_test.py```

## Error
``` bash
python3 test/basic_test.py
INFO:pyocd.board.board:Target type is ama3b1kk
INFO:pyocd.coresight.dap:DP IDR = 0x2ba01477 (v1 rev2)
INFO:pyocd.coresight.ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
INFO:pyocd.coresight.rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=43b part=4c4)
INFO:pyocd.coresight.rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
INFO:pyocd.coresight.rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
INFO:pyocd.coresight.rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
INFO:pyocd.coresight.rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
INFO:pyocd.coresight.rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca0>
INFO:pyocd.coresight.cortex_m:CPU core #0 is Cortex-M4 r0p1
INFO:pyocd.coresight.cortex_m:FPU present: FPv4-SP
INFO:pyocd.coresight.dwt:4 hardware watchpoints
INFO:pyocd.coresight.fpb:6 hardware breakpoints, 4 literal comparators
binary file: C:\Users\owen.lyke\Projects\pyOCD\binaries\l1_artmbed.bin

------ GET Unique ID ------
Unique ID: A127780152564e45004b400f0000002c0000000097969901

------ TEST READ / WRITE CORE REGISTER ------
initial pc: 0x11180
now pc: 0x3D82
initial pc value rewritten: 0x11180
MSP = 0x10000fe8; PSP = 0x00000000
CONTROL = 0x04; FAULTMASK = 0x00; BASEPRI = 0x00; PRIMASK = 0x00
New PRIMASK = 0x00
Restored PRIMASK = 0x00
S0 = 0 (0x00000000)
New S0 = 3.14159 (0x40490fdb)
Restored S0 = 0 (0x00000000)

------ TEST HALT / RESUME ------
resume
halt
HALT: pc: 0x11B6C

------ TEST STEP ------
reset and halt
ERROR:pyocd.board.board:link exception during target disconnect:
Traceback (most recent call last):
  File "C:\Users\owen.lyke\Projects\pyOCD\pyocd\probe\cmsis_dap_probe.py", line 285, in read_ap_result_callback
[... 133 lines cut ...]
    self.daplink.flush()
  File "C:\Users\owen.lyke\Projects\pyOCD\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 638, in flush
    self._read_packet()
  File "C:\Users\owen.lyke\Projects\pyOCD\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 863, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "C:\Users\owen.lyke\Projects\pyOCD\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 420, in decode_data
    data = self._decode_transfer_block_data(data)
  File "C:\Users\owen.lyke\Projects\pyOCD\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 389, in _decode_transfer_block_data
    raise DAPAccessIntf.TransferError()
pyocd.probe.pydapaccess.dap_access_api.TransferError

The above exception was the direct cause of the following exception:
# ...
# More stack traces follow here...
# ...

  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferErro
```

* Tests all pass for the micro:bit
* SWD programming with SEGGER J-Link works on the same lines that are connected to the KL26Z interface MCU (when disconnected from KL26 to prevent contention)
* I tried desoldering the NRF5182 target from a micro:bit and replacing it with the [Apollo3 module](https://www.sparkfun.com/artemis) (SWCLK, SWDIO, RESET, 3V3, GND) -- but then I found the same error (but this time the KL26Z was running the miro:bit interface firmware, presumably with the FlashAlgo for the NRF)

Thanks!

flit, 2019-10-22:
Hi! Very glad to see someone wanting to support the Ambiq devices in pyOCD! It'll take a little to dig through your copius notes (thanks for the detail).

flit, 2019-11-02:
Ok, finally able to get back to this, sorry for the delay. Thanks again for all the detail.

#### Initial questions

1. What firmware image are you testing with? (Given the data, this probably doesn't have an impact but might be helpful.)
2. Does the Apollo3 have a boot ROM?
3. Does the boot ROM disable debug access while it executes, before jumping to user code in flash?

#### What's working

- SWD is working, no problems there.
- CoreSight probing is successful, pyOCD is able to detect all components.
- Unlikely to be flash algo related. Code seems to be running early in `basic_test.py`, and the flash algo hasn't been invoked yet in the test.

#### Some diagnosis

- The initial fault happens in `CortexM.reset_and_halt()` just after the target has been reset. The code in that method is waiting for the target to come out of reset and enter the halted state.
  - The standard way to perform reset and halt is to enable reset vector catch (`DEMCR.VC_CORERESET`) prior to reset.
  - If the device has a boot ROM, this can interfere with reset and halt. Especially if the ROM temporarily disables debug access while initialising sensitive resources. Some devices come out of reset with debug access disabled, and it is only enabled after the ROM runs to a certain point. 
  - There is no standard way to deal with boot ROMs. Reset handling logic is the thing that most frequently requires customisation in debugger support (aside from the obvious device info and flash algo).
- The exception is a `TransferError`, which is only raised in two relatively rare cases:
  1. "No ACK", where the device didn't respond at all to the SWD request. This the most likely case.
  2. An SWD protocol error happened. Unlikely except at higher SWD frequencies than DAPLink can support.

You can verify that the fault was "No ACK" by running the test with the current HEAD of master (fc84c82 as of this writing). I recently merged a change to include a description in `TransferError` exceptions indicating which of the two cases listed above occurred.

Often the [debug sequences](https://arm-software.github.io/CMSIS_5/Pack/html/debug_description.html#pdsc_SequenceNameEnum_pg) in [CMSIS-Packs](https://www.keil.com/dd2/pack/) will provide guidance. I checked the Apollo DFP but it didn't have anything, most likely because it is produced by Ambiq themselves and a lot of vendors don't include the debug sequences (only Keil are really good at that).

Please contact me at chris.reed at arm.com for further discussion… Let's get this working! I'm very interested in supporting both Ambiq and Sparkfun.

oclyke, 2019-11-12:
Thanks for your analysis(!) - I will get back to it soon but right now my priorities have shifted to porting mbed software.

rowanham, 2023-07-20:
I hate to dig up a thread this old, but I am having a possibly related problem.
I have had very good experiences working with `pyOCD` in the past. (I have been writing embedded software since 1994, so finding an open source tool for embedded systems that was so powerful really blew me away.)
Anyway, recently I have been working with an ARM Cortex M4F based SoC and using the AmbiqSuite SDK, and my host machine is an Ubuntu 20.04 box. I have two debug probes - an Atmel ICE that I have used for years and an Olimex tiny-H USB device. I have not yet been able to get either to work with gdb (which is my immediate goal).
What else? The SoC is on a SparkFun MicroMod board, mounted on the poorly named "main board double". I soldered in an SWD header to connect to the probe. (I am an old man and my soldering skills are very out of date, so that is certainly an issue to question.) This is not a work issue - this is a hobby project to keep myself busy while I am between jobs. I am trying to make an oscilloscope (for future embedded work) since the board has an 8 channel 24 (-ish) bit ADC.
OK, that was too many words. I am looking for advice on getting gdb running. I have more experience with the Atmel ICE, so that is what I have been focussing on. But every pyocd build I have tried (and all the different client sides...) have failed to connect with an "No Ack" message. (It seems to be on reset.) I stopped trying the Olimex probe since I got failures that I understood even less, but I can go back to that if someone thinks that would be a more productive path.
I know this is a very stale thread, but if anyone responds I can post logs, screen captures,  dmesg or lsusb captures - whatever can get me closer to my 'scope. At this rate I may get a job before that happens...

flit, 2023-07-21:
@rowanham 

First, I assume you are using the Artemis module with an Apollo3.

The "No ACK" errors indicate that the target is not responding at all to requests over SWD. There is a description of possible causes and some possible solutions [in the FAQ](https://pyocd.io/docs/faq.html#getting-no-ack-errors-when-attempting-to-connect).

It looks like SparkFun have a [fork of pyocd](https://github.com/sparkfun/pyOCD) with Apollo3 support. This has not been upstreamed yet. I recommend using that version, otherwise you will probably have a lot of problems. Although that fork seems to be several years out of date compared to the current pyocd version, it's still a better bet given how difficult the Apollo3 boot process is. Any further issues specific to the Apollo3 should probably be directed to SparkFun support.

At some point, I'd like to upstream the Apollo3 support (now that I know it exists!), but it's relatively low priority for me. Someone else could do it, however.

flit, 2023-07-21:
Closing this issue. Apollo3 support is currently present in SparkFun's fork of pyocd, but has not been mainstreamed. See https://github.com/sparkfun/pyOCD.

closed 2023-07-21 by flit

rowanham, 2023-07-21:
I see this issue is closed. I'll just leaver a couple of comments:
1) Yes, I am using Apollo3. I tried the SparkFun pyOCD fork before, and I will try it again since I know more about the system now.
2) I have no root cause, but I can reliably work around the "No ACK" issue. If I disconnect and reconnect the USB cable before launching pyOCD, the error does not occur. (This also fixes issues I have seen with the artemis_loader GUI and with the "bootload" targets in the Ambiq SDK makefiles.)
3) By changing reset_type to sw_emulated, I can get arm-none-eabi-gdb to connect to pyocd gdbserver without error. I am not sure yet whether the debugger is actually working, but at least it connects without error.

flit, 2023-07-31:
@rowanham Thanks for the additional notes! Sorry the Apollo devices aren't supported upstream. It will take someone with time and some devotion to those devices to port the SparkFun fork to the upstream code.

If the board is powered by USB, then it's likely that reconnecting USB forcibly wakes the device.

The reason `sw_emulated` reset works is almost certainly because it does not cause the device to boot from the internal boot ROM. Devices that have a boot ROM that does security tasks often completely disable debug from the point when reset is asserted until the ROM enables debug (after it has checked the security state). This causes lots of issues for debuggers, and always requires custom device-specific logic to handle correctly. It also makes it impossible to stop at the first instruction of user code without additional device-specific logic.

rowanham, 2023-08-01:
My whole purpose in going down this path is to turn the device into an oscilloscope. ('m tired of being jealous of my friends who have these keyfob-sized 'scopes.) I got all the pieces working without the debugger (esxcept for the host UART writes to the target), but my code was a mess with all the fixes, so I've spent the last week working on a complete rewrite. Right now it is looking like getting the rewrite working could use the debugger...

Anyway, thanks for the detail on the `sw_emulated` reset - that makes a lot of sense and is good to have in my back pocket. I will likely use it this week!