pyocd/pyOCD #743: "No cores were discovered!" with nucleo_g071rb

erwango, 2019-10-16:
When trying to flash nucleo_g071rb using pyocd, following error happens:
```
0001231:WARNING:rom_table:Invalid coresight component, cidr=0x90909090
0001232:WARNING:rom_table:Warning: ROM table @ 0xf0000000 has unexpected CIDR component class (0x0)
0001235:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "$HOME/.local/lib/python3.5/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "/$HOME/.local/lib/python3.5/site-packages/pyocd/__main__.py", line 468, in do_flash
    with session:
  File "$HOME/.local/lib/python3.5/site-packages/pyocd/core/session.py", line 287, in __enter__
    self.open()
  File "$HOME/.local/lib/python3.5/site-packages/pyocd/core/session.py", line 367, in open
    self._board.init()
  File "$HOME/.local/lib/python3.5/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "$HOME/.local/lib/python3.5/site-packages/pyocd/core/coresight_target.py", line 158, in init
    seq.invoke()
  File "$HOME/.local/lib/python3.5/site-packages/pyocd/utility/sequencer.py", line 201, in invoke
    resultSequence = call()
  File "$HOME/.local/lib/python3.5/site-packages/pyocd/core/coresight_target.py", line 238, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discove
```

jeromecoutant, 2019-10-16:
@schstm

flit, 2019-10-21:
Unfortunately I don't have an STM32G0 to test with. When I get a chance, I'll review the reference manual for this device to see if anything stands out as to why the ROM table accesses are returning invalid values.

erwango, 2019-10-22:
@flit, If this can help, one hint is that read cidr value is not always the same.
I've seen cidr=0x2b2b2b2b2b or seomthing like that.
This looks strange if 'we're looking to an hardcoded value.

lbarthelemy, 2019-10-22:
I encounter the same issue.
Board: nucleo_g071rb
Pyocd command:
`pyocd flash --pack Keil.STM32G0xx_DFP.1.2.0.pack -e sector -a 0x8000000 -t stm32g071rbtx zephyr.hex`

Error log:
`0000709:WARNING:rom_table:Invalid` coresight component, cidr=0x90909090
0000709:WARNING:rom_table:Warning: ROM table @ 0xf0000000 has unexpected CIDR component class (0x0)
0000711:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS\[self._args.cmd\](self)
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/__main__.py", line 468, in do_flash
    with session:
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/session.py", line 287, in __enter__
    self.open()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/session.py", line 367, in open
    self._board.init()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 158, in init
    seq.invoke()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/utility/sequencer.py", line 201, in invoke
    resultSequence = call()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 238, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!
`

Additional remarks:
- this issue persist if I unplug and replug the nucleo_g071rb
- the issue disappear if I flash any app using STM32CubeIDE after each pyocd flashing procedure.

flit, 2019-10-22:
The comment about the problem disappearing if you have valid code in flash is very Interesting.

I'm not sure what the different cidr values means. Except, yes, it's obviously not hardcoded.

flit, 2019-10-22:
Looks like the `RCC_APBENR1.DBGEN` register might need to be set to enable clocks to debug resources. (Thanks to the debug sequences in the STM32G0 CMSIS-Pack and therefore the Keil team!)

If STM32CubeIDE is setting `DBGEN` to flash the app, it will probably remain set until the next warm or cold reset (depending on the design), explaining why the issue disappears when an app is flashed this way.

Please try creating a `pyocd_user.py` user script containing this:

```py
def will_init_target(target, init_seq):
    # add init task to set RCC_APBENR1.DBGEN before reading ROM tables
    init_seq.insert_after('create_aps',
        ('set_dbgen', lambda : target.write32(0x4002103c, 0x08000000))
        )
    return init_seq
```

(Note that I have _not_ tested this since I don't have the STM32G0 hardware.)

For more about user scripts, the documentation is here: https://github.com/mbedmicro/pyOCD/blob/master/docs/user_scripts.md

erwango, 2019-10-23:
@flit , any required pyocd version to use scripts ? I'm using 0.21.0
and facing following error :
```
$ pyocd flash -e sector -a 0x8000000 -t stm32g071rb /local/mcu/zephyr/zephyr-project/samples/hello_world/build/nucleo_g071rb/zephyr/zephyr.hex --script=/local/mcu/zephyr/zephyr-project/boards/arm/nucleo_g071rb/pyocd_user.py
0000924:CRITICAL:__main__:uncaught exception: will_init_target() missing 1 required positional argument: 'init_seq'
Traceback (most recent call last):
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/__main__.py", line 468, in do_flash
    with session:
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/session.py", line 287, in __enter__
    self.open()
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/session.py", line 367, in open
    self._board.init()
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/coresight_target.py", line 157, in init
    self.call_delegate('will_init_target', target=self, init_sequence=seq)
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/target.py", line 120, in call_delegate
    return getattr(self._delegate, method_name)(*args, **kwargs)
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/session.py", line 412, in __call__
    self._fn(**args)
TypeError: will_init_target() missing 1 required positional argument: 'init_seq'
```

lbarthelemy, 2019-10-23:
@flit Thank you for the help.
I've tried the user script you provided, unfortunately it won't run as it, I've renamed `init_seq` into `init_sequence`
`def will_init_target(target, init_sequence):
    # add init task to set RCC_APBENR1.DBGEN before reading ROM tables
    init_sequence.insert_after('create_aps',
        ('set_dbgen', lambda : target.write32(0x4002103c, 0x08000000))
        )
    return init_sequence`
But I'm still stuck with this error:
`0000548:CRITICAL:__main__:uncaught exception: 'NoneType' object has no attribute 'write_memory'
Traceback (most recent call last):
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/__main__.py", line 468, in do_flash
    with session:
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/session.py", line 287, in __enter__
    self.open()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/session.py", line 367, in open
    self._board.init()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 158, in init
    seq.invoke()
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/utility/sequencer.py", line 201, in invoke
    resultSequence = call()
  File "<string>", line 4, in <lambda>
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/memory_interface.py", line 44, in write32
    self.write_memory(addr, value, 32)
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 268, in write_memory
    return self.selected_core.write_memory(addr, value, transfer_size)
AttributeError: 'NoneType' object has no attribute 'write_memory'
`

flit, 2019-10-23:
Sorry! Among other things, I forgot that `CoreSightTarget.write32()` goes through the selected core object, which doesn't exist yet at that stage of the init sequence… (Lesson: don't write code during a meeting 😉 )

Please try this updated version. I was able to test this enough to make sure the register write happens at the right time.

```py
def will_init_target(target, init_sequence):
    def set_dbgen():
        LOG.info("setting RCC_APBENR1.DBGEN")
        aps[0].write32(0x4002103c, 0x08000000)
        dp.flush()
            
    # add init task to set RCC_APBENR1.DBGEN before reading ROM tables
    init_sequence.insert_after('create_aps',
        ('set_dbgen', set_dbgen)
        )
    return init_sequence
```

erwango, 2019-10-24:
Same issue, though it seems change applied correctly.
```
$ pyocd flash -e sector -a 0x8000000 -t stm32g071rb /local/mcu/zephyr/zephyr-project/samples/hello_world/build/nucleo_g071rb/zephyr/zephyr.hex --script=/local/mcu/zephyr/zephyr-project/boards/arm/nucleo_g071rb/pyocd_user.py --verbose
0000924:INFO:board:Target type is stm32g071rb
0001037:INFO:dap:DP IDR = 0x0bc11477 (v1 MINDP rev0)
0001133:INFO:ap:AP#0 IDR = 0x04770031 (AHB-AP var3 rev0)
0001140:INFO:<string>:setting RCC_APBENR1.DBGEN
0001155:WARNING:rom_table:Invalid coresight component, cidr=0xb2b2b2b2
0001156:WARNING:rom_table:Warning: ROM table @ 0xf0000000 has unexpected CIDR component class (0x0)
0001158:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/__main__.py", line 468, in do_flash
    with session:
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/session.py", line 287, in __enter__
    self.open()
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/session.py", line 367, in open
    self._board.init()
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/coresight_target.py", line 158, in init
    seq.invoke()
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/utility/sequencer.py", line 201, in invoke
    resultSequence = call()
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/pyocd/core/coresight_target.py", line 238, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!

```

jeromecoutant, 2019-10-31:
Test with 73710a2 commit version:

````
$ pyocd gdbserver
0000753:WARNING:mbed_board:Board ID 0729 is not recognized, using generic cortex_m target.
0001271:INFO:board:Target type is cortex_m
0001271:WARNING:board:Generic 'cortex_m' target type is selected; is this intentional? You will be able to debug but not program flash. To set the target type use the '--target' argument or 'target_override' option. Use 'pyocd list --targets' to see available targets types.
0001290:DEBUG:dap:read_dp:000001 (addr=0x00000000) -> 0x0bc11477
0001290:INFO:dap:DP IDR = 0x0bc11477 (v1 MINDP rev0)
0001290:DEBUG:dap:write_dp:000002 (addr=0x00000000) = 0x0000001e
0001292:DEBUG:dap:write_dp:000003 (addr=0x00000008) = 0x00000000
0001292:DEBUG:dap:write_dp:000004 (addr=0x00000004) = 0x50000000
0001293:DEBUG:dap:read_dp:000005 (addr=0x00000004) -> 0xf0000040
0001293:DEBUG:dap:write_dp:000006 (addr=0x00000004) = 0x50000f00
0001294:DEBUG:dap:write_dp:000007 (addr=0x00000008) = 0x00000000
0001299:DEBUG:dap:read_ap:000008 (addr=0x000000fc) -> 0x04770031
0001302:DEBUG:dap:read_ap:000009 (addr=0x010000fc) -> 0x00000000
0001305:DEBUG:dap:read_ap:000010 (addr=0x000000fc) -> 0x04770031
0001312:DEBUG:dap:read_ap:000011 (addr=0x000000f8) -> 0xf0000003
0001312:INFO:ap:AP#0 IDR = 0x04770031 (AHB-AP var3 rev0)
0001313:DEBUG:dap:read_ap:000012 (addr=0x00000000) -> 0x03000052
0001313:DEBUG:dap:write_ap:000013 (addr=0x00000000) = 0x4f000052
0001317:DEBUG:dap:read_ap:000014 (addr=0x00000000) -> 0x0f000052
0001317:DEBUG:dap:write_ap:000015 (addr=0x00000000) = 0x03000052
0001319:DEBUG:dap:read_ap:000016 (addr=0x00000000) -> 0x03000052
0001320:DEBUG:dap:write_ap:000017 (addr=0x00000000) = 0x23000052
0001322:DEBUG:dap:read_ap:000018 (addr=0x00000000) -> 0x03000052
0001338:INFO:rom_table:AP#0 ROM table #0 @ 0xf0000000 (designer=020 part=460)
0001352:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c0>
0001353:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0001369:INFO:rom_table:  [0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0001378:INFO:rom_table:  [1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0001388:INFO:rom_table:  [2]<e0002000:BPU class=14 designer=43b part=00b>
0001392:INFO:cortex_m:CPU core #0 is Cortex-M0+ r0p1
0001401:INFO:dwt:2 hardware watchpoints
0001412:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0001451:INFO:server:Semihost server started on port 4444
0001453:INFO:gdbserver:GDB server started on port 3333
````

erwango, 2019-10-31:
@jeromecoutant what about `pyocd flash`?

flit, 2019-10-31:
@erwango @lbarthelemy  I don't know why writing `RCC_APBENR1.DBGEN` wouldn't fix the issue… That's the only nonstandard modification in the G0 CMSIS-Pack. Without a board it's hard to debug! 😦 

@jeromecoutant Can you try with the `stm32g071rb` target type selected (via `-t`)? I don't know why using the generic `cortex_m` target type would be different; the CoreSight component discovery process is the same for all target types. Did you have code in flash?

lbarthelemy, 2019-11-04:
> @erwango @lbarthelemy I don't know why writing `RCC_APBENR1.DBGEN` wouldn't fix the issue… That's the only nonstandard modification in the G0 CMSIS-Pack. Without a board it's hard to debug! frowning

@flit I understand, thank you for the support any way!

jeromecoutant, 2019-11-04:
Hi

pyocd gdbserver -t stm32g071rbtx

is OK (tested with Keil.STM32G0xx_DFP.1.2.0.pack file)

lbarthelemy, 2019-11-15:
Hi,
@jeromecoutant 
Thank you the additional testing.
I've updated my pyocd to [73710a2 ](https://github.com/mbedmicro/pyOCD/commit/73710a27b3fdfc730c981ec2c3b7ae15cf09dc92)
(using `python3 setup.py install --user`, and now I have `pyocd --version 0.23.0a2.dev23` )
And running `pyocd gdbserver --pack "Keil.STM32G0xx_DFP.1.2.0.pack"  -t stm32g071rbtx
` produces the same error:

> 0000603:INFO:board:Target type is stm32g071rbtx
0000669:INFO:dap:DP IDR = 0x0bc11477 (v1 MINDP rev0)
0000706:INFO:ap:AP#0 IDR = 0x04770031 (AHB-AP var3 rev0)
0000711:WARNING:rom_table:Invalid coresight component, cidr=0x90909090
0000711:WARNING:rom_table:Warning: ROM table @ 0xf0000000 has unexpected CIDR component class (0x0)
0000712:CRITICAL:__main__:No cores were discovered!
...
  File "/home/lbarthelemy/.local/lib/python3.6/site-packages/pyocd-0.23.0a2.dev23-py3.6.egg/pyocd/core/coresight_target.py", line 298, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!

Could you please try the following sequence:
- `pyocd flash --pack Keil.STM32G0xx_DFP.1.2.0.pack -e sector -a 0x8000000 -t stm32g071rbtx zephyr.hex`
- `pyocd gdbserver --pack "Keil.STM32G0xx_DFP.1.2.0.pack"  -t stm32g071rbtx`

(I need to flash using pyocd one time, before "no cores" error show up for subsequent pyocd calls.)

Thank you by advance

thegecko, 2019-12-10:
Looks similar to #729

thegecko, 2019-12-10:
I see the same issue with the L073RZ(#729) and the F767ZI:

PyOCD: 0.23.0
MacOS: 10.12.6
Target: F767ZI
Latest FW: V2J34M25
Pack: [Keil.STM32F7xx_DFP.2.12.0.pack](https://developer.arm.com/embedded/cmsis/cmsis-packs/devices/STMicroelectronics/STM32F767ZITx)

```bash
$ ./pyocd flash --frequency 1800000 -O connect_mode=under-reset --uid 0672FF574953867567051035 --target STM32F767ZI --pack ~/Desktop/Keil.STM32F7xx_DFP.2.12.0.pack --erase=chip /Users/user/mbed-os-example-blinky.bin

0003040:ERROR:ap:Transfer error while reading AP#0 ROM table: SWD/JTAG Transfer Fault @ 0xe000efbc-0xe000f043
0003109:CRITICAL:__main__:No cores were discovered!
Traceback (most recent call last):
  File "pyocd/__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "pyocd/__main__.py", line 473, in do_flash
    with session:
  File "pyocd/core/session.py", line 287, in __enter__
    self.open()
  File "pyocd/core/session.py", line 367, in open
    self._board.init()
  File "pyocd/board/board.py", line 83, in init
    self.target.init()
  File "pyocd/core/coresight_target.py", line 164, in init
    seq.invoke()
  File "pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "pyocd/core/coresight_target.py", line 298, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
DebugError: No cores were discovered!
```

erwango, 2020-01-07:
@lbarthelemy, @flit, I manage to flash using pyocd by holding the board reset during flash (but not too long otherwise I get another error). This is using pyocd v0.21.0.
This unlock some use cases, of course this is non suitable for CI hw testing

erwango, 2020-01-07:
@lbarthelemy, I'm now able to flash from command line without manualinteraction with the board:

```
$ pyocd --version
0.24.1
$ pyocd flash -e sector -a 0x8000000 -t 'stm32g071rb' -O reset_type=hw -O connect_mode='under-reset' zephyr.hex
[====================] 100%
0003079:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 12288 bytes (12 pages) at 6.90 kB/s
```

lbarthelemy, 2020-01-08:
@erwango Thank you for the update!!
It works even with the pyocd dev version I updated a while ago:

```
$ pyocd --version
0.23.0a2.dev23
$ pyocd flash --pack "/media/data/docs/tmp/unity/mcu/oses/zephyr/Firmwares/CMSIS_Device_Family_Pack_DFP/Keil.STM32G0xx_DFP.1.2.0.pack" -e sector -a 0x8000000 -t 'stm32g071rbtx' -O reset_type=hw -O connect_mode='under-reset' zephyr.hex
[====================] 100%
0002147:INFO:loader:Erased 26624 bytes (13 sectors), programmed 26624 bytes (26 pages), skipped 0 bytes (0 pages) at 18.89 kB/s
```
It still need to use `-t 'stm32g071rbtx'`, instead of `-t 'stm32g071rb'` which is unrecognized on my setup: I may have missed something while manually workarounding  https://github.com/ARMmbed/cmsis-pack-manager/issues/121 issue.
Anyway, that works, thanks again.

erwango, 2020-01-09:
@flit , I'm closing this point. I think solution could be proposed for other similar issues.

closed 2020-01-09 by erwango

thegecko, 2020-01-09:
I still think there is an underlying issue in PyOCD which needs to be investigated here. At the least a guard against the error which can guide the user?

kiram9, 2020-03-04:
I found a workable solution by looking at openocd which works correctly. and also looking at 
Openocd notes the following reset options, and declares that STM32 must be connected while under reset, and also sets APBENR1, DBGMCU_CR and DBGMCU_APB1_FZ1 to enable debug in low power modes. By default zephyr puts the core in a low power sleep mode, which prevents you from connecting over the debugger, so you have to manually set the DBGMCU registers correctly. 

From the openocd target config 
```
# Reset configuration
# use hardware reset, connect under reset
# connect_assert_srst needed if low power mode application running (WFI...)
reset_config srst_only srst_nogate connect_assert_srst
set CONNECT_UNDER_RESET 1 
set CORE_RESET 0
```

This is my configuration that allows me to debug, however it does reset the core on startup which causes the application to restart. I am not sure how to change this in pyocd. 

I stepped release_reset through the different phases until things worked. It seems to need to be released before some additional configuration happens. 

The following is my pyocd_user.py
```
def will_init_target(target, init_sequence):
    def print_registers():
        registers = [
            ["SRAM_0        ", 0x20000000],
            ["FLASH_0       ", 0x08000000],
            ["RCC_APBENR1  ", 0x40021000 + 0x3c],
            ["DBGMCU_CR     ", 0x40015800 + 0x04],
            ["DBGMCU_APB1_FZ1", 0x40015800 + 0x08],
            ["DBGMCU_APB1_FZ2", 0x40015800 + 0x0c],

        ]
        for r in registers:
            LOG.info("{} {:08x}".format(r[0], aps[0].read32(r[1])))
    def print_device_id(): 
        REG_ID_BASE = 0x1FFF7590
        id_low = aps[0].read32(REG_ID_BASE)
        id_mid = aps[0].read32(REG_ID_BASE + 0x04)
        id_high = aps[0].read32(REG_ID_BASE + 0x08)
        LOG.info("device_id 0x{:x}{:x}{:x}".format(id_high,id_mid,id_low))

    def hold_reset():
        dp.probe.assert_reset(1)

    def release_reset():
        dp.probe.assert_reset(0)

    def set_dbgen():
        reg_enbits = [
            # Set the DBGMCU CLOCK = enabled
            ["RCC_APBENR1   ", 0x40021000 + 0x3c, 0x08000000],
            # Enable debug during low power modes STANDBY and STOP
            ["DBGMCU_CR     ", 0x40015800 + 0x04, 0x00000006],
            # Stop WDT during halt
            ["DBGMCU_APB1_FZ1", 0x40015800 + 0x08, 0x00001800],
            # Stop other peripherals during halt
            ["DBGMCU_APB1_FZ2", 0x40015800 + 0x08, 0x00000000],

        ]
        for r in reg_enbits:
            reg = aps[0].read32(r[1])
            reg = reg | r[2]
            aps[0].write32(r[1], reg)
            #dp.flush()
            LOG.info("Set {} {:08x}".format(r[0], reg))
        #Force register values
        reg_wrbits = [
            #["CR     ", 0x40015800 + 0x04, 0x00000006],
            #["FRZ1   ", 0x40015800 + 0x08, 0x00000000],
            #["FRZ2   ", 0x40015800 + 0x0C, 0x00000000],
        ]
        for r in reg_wrbits:
            aps[0].write32(r[1], r[2])
            LOG.info("Set {} {:08x}".format(r[0], aps[0].read32(r[1])))

        dp.flush()

    init_sequence.insert_after('halt_on_connect',
                               ('device_uid', print_device_id),
                               ('print_registers', print_registers),
                               ('set_dbgen', set_dbgen)
                               )
    init_sequence.insert_before('dp_init', ('hold_reset', hold_reset))
    init_sequence.insert_after("create_cores", ('release_reset', release_reset))

    return init_sequence

```