pyocd/pyOCD #745: TransferError when debugging Mbed-LPC1768

flit, 2019-10-21:
Reproducible using Mbed Studio and mbed-os-example-blinky on Mbed-LPC1768 board. Started with a device with blank flash.

This is not caused by the device entering sleep mode. I had completely disabled calls to `hal_sleep()` or `hal_deepsleep()` in `sleep_manager_sleep_auto()` before running this test.

```
Selected port 50000 for debugging
0000332:INFO:board:Target type is lpc1768
0000429:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0000498:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0000592:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=000 part=000)
0000631:INFO:rom_table:[0]
0000652:INFO:rom_table:[1]
0000691:INFO:rom_table:[2]
0000818:INFO:rom_table:[3]
0000870:INFO:rom_table:[4]
0000934:INFO:rom_table:[5]
0000975:INFO:cortex_m:CPU core #0 is Cortex-M3 r2p0
0001103:INFO:dwt:4 hardware watchpoints
0001109:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001135:INFO:server:Semihost server started on port 4444
0001136:INFO:gdbserver:GDB server started on port 50000
Reading symbols from /Users/creed/Mbed Programs/mbed-os-example-blinky/BUILD/LPC1768/ARMC6/mbed-os-example-blinky.elf...
done.
0001203:INFO:gdbserver:One client connected!
0x1fff0ba4 in ?? ()
0001263:INFO:gdbserver:Attempting to load Argon
0001265:INFO:gdbserver:Attempting to load RTX5
0001266:INFO:gdbserver:RTX5 loaded successfully
Attached to debugger on port 50000
0006736:INFO:loader:Erased chip, programmed 37888 bytes (37 pages), skipped 3072 bytes (3 pages) at 7.41 kB/s
Image loaded: /Users/creed/Mbed Programs/mbed-os-example-blinky/BUILD/LPC1768/ARMC6/mbed-os-example-blinky.elf
Note: automatically using hardware breakpoints for read-only addresses.
0006938:ERROR:gdbserver:Unhandled exception in handle_message: 
Traceback (most recent call last):
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 549, in handle_message
    reply = handler(msg[msgStart:])
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 776, in v_command
    return self.v_cont(cmd)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 827, in v_cont
    return self.resume(None)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 700, in resume
    self.target.resume()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 958, in resume
    self.flush()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/core/target.py", line 142, in flush
    self.session.probe.flush()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 239, in flush
    six.raise_from(self._convert_exception(exc), exc)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferError
0006947:ERROR:gdbserver:Unhandled exception in handle_message: 
Traceback (most recent call last):
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 547, in handle_message
    reply = handler()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 982, in get_registers
    return self.create_rsp_packet(self.target_facade.get_register_context())
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/gdbserver/context_facade.py", line 69, in get_register_context
    vals = self._context.read_core_registers_raw(reg_num_list)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/debug/cache.py", line 531, in read_core_registers_raw
    return self._regcache.read_core_registers_raw(reg_list)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/debug/cache.py", line 133, in read_core_registers_raw
    self._check_cache()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/debug/cache.py", line 110, in _check_cache
    if self._core.is_running():
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 943, in is_running
    return self.get_state() == Target.TARGET_RUNNING
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 920, in get_state
    dhcsr = self.read_memory(CortexM.DHCSR)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 605, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 564, in _read_memory
    result = read_mem_cb()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 546, in read_mem_cb
    res = result_cb()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 276, in read_ap_cb
    result = result_cb()
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 307, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/Library/Application Support/Mbed Studio/mbed-studio-tools/python/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferError
```

flit, 2019-10-21:
cc @thegecko

alejandroaps, 2019-12-16:
I have the same problem, any solution?

djj198, 2020-01-10:
I have this error as well

fepz, 2020-06-09:
@alejandroaps @djj198 No exception when using pyOCD 0.10. It seems to be a problem when using pyOCD version 0.11 or later.