pyocd/pyOCD #750: CRITICAL:__main__:Unable to open device: open failed

medjutim, 2019-11-03:
Hello,

I have a problem starting pyocd gdbserver.

sudo pyocd gdbserver
0000309:INFO:board:Target type is k64f
0000309:CRITICAL:__main__:Unable to open device: open failed
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/interface/hidapi_backend.py", line 49, in open
    self.device.open_path(self.device_info['path'])
  File "hid.pyx", line 72, in hid.device.open_path
OSError: open failed

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 141, in open
    self._link.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 545, in open
    self._interface.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/interface/hidapi_backend.py", line 51, in open
    raise six.raise_from(DAPAccessIntf.DeviceError("Unable to open device: " + str(exc)), exc)
  File "<string>", line 3, in raise_from
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.DeviceError: Unable to open device: open failed

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/__main__.py", line 571, in do_gdbserver
    with session:
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/session.py", line 287, in __enter__
    self.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/session.py", line 363, in open
    self._probe.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 153, in open
    six.raise_from(self._convert_exception(exc), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.ProbeError: Unable to open device: open failed

flit, 2019-11-03:
Please provide more information. What probe are you using? What OS and version? I can make guesses, but I'd rather not.

medjutim, 2019-11-03:
Thank you for the quick response. I am on macOS mojave and using segger j-link on frdm-k64f.

flit, 2019-11-03:
pyOCD doesn't support J-Link probes; it supports CMSIS-DAP and STLink (see the [readme](https://github.com/mbedmicro/pyOCD/blob/master/README.md) for details). It shouldn't even try to connect, or show an available probe, if you only have a J-Link connected.

It's unclear what it is attempting to connect to… the log shows it using the hidapi backend, which only supports USB HID interfaces, but J-Link only provides a vendor-specific bulk interfaces, so hidapi should not see it at all.

Btw, you should _never_ run pyOCD via sudo on a Mac, nor should you ever need to.

Could you please provide the output of these two commands:
1. `pyocd list -vv`
2. `pyocd gdb -vv`

This might help shed some light on what it's trying to connect to.

medjutim, 2019-11-03:
pyocd list -vv
0000567:DEBUG:session:Project directory: /Users/murdocgal
  #   Probe              Unique ID                                         
---------------------------------------------------------------------------
  0   FRDM-K64F [k64f]   024002261e101e0d000000000000000000000000e3c9e3b5  

 pyocd gdb -vv
0000551:DEBUG:session:Project directory: /Users/murdocgal
0000557:INFO:board:Target type is k64f
0000562:DEBUG:dap_access_cmsis_dap:CMSIS-DAP probe 024002261e101e0d000000000000000000000000e3c9e3b5 firmware version: 1.0
0000570:DEBUG:sequencer:Running task load_svd
0000572:DEBUG:sequencer:Running task create_flash
0000572:DEBUG:sequencer:Running task dp_init
0000612:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0000612:DEBUG:sequencer:Running task power_up
0000614:DEBUG:sequencer:Running task find_aps
0000620:DEBUG:sequencer:Running task create_aps
0000620:DEBUG:sequencer:Running task create_ap.0
0000624:INFO:ap:AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0000626:DEBUG:ap:AP#0 default HPROT=3 HNONSEC=0
0000628:DEBUG:ap:AP#0 implemented HPROT=3 HNONSEC=0
0000630:DEBUG:sequencer:Running task create_ap.1
0000638:INFO:ap:AP#1 IDR = 0x001c0000 (proprietary)
0000638:DEBUG:sequencer:Running task check_mdm_ap_idr
0000638:DEBUG:target_kinetis:MDM-AP version 0
0000639:DEBUG:sequencer:Running task check_flash_security
0000667:INFO:target_kinetis:K64F not in secure state
0000667:DEBUG:sequencer:Running task init_ap_roms
0000667:DEBUG:sequencer:Running task init_ap.0
0000680:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=43b part=4c4)
0000711:INFO:rom_table:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000716:INFO:rom_table:[1]<e0001000:DWT class=14 designer=43b part=002>
0000725:INFO:rom_table:[2]<e0002000:FPB class=14 designer=43b part=003>
0000731:INFO:rom_table:[3]<e0000000:ITM class=14 designer=43b part=001>
0000738:INFO:rom_table:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
0000745:INFO:rom_table:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0000751:INFO:rom_table:[6]<e0042000:ETB class=9 designer=43b part=907 devtype=21 archid=0000 devid=0:0:0>
0000757:INFO:rom_table:[7]<e0043000:CSTF class=9 designer=43b part=908 devtype=12 archid=0000 devid=0:0:28>
0000757:DEBUG:sequencer:Running task create_cores
0000757:DEBUG:coresight_target:Creating SCS-M4 component
0000762:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0000771:INFO:cortex_m:FPU present: FPv4-SP
0000816:DEBUG:sequencer:Running task halt_on_connect
0000824:DEBUG:sequencer:Running task create_components
0000824:DEBUG:coresight_target:Creating DWT component
0000831:INFO:dwt:4 hardware watchpoints
0000831:DEBUG:coresight_target:Creating FPB component
0000838:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000838:DEBUG:fpb:fpb has been disabled
0000846:DEBUG:coresight_target:Creating ITM component
0000874:DEBUG:coresight_target:Creating TPIU-M4 component
0000882:DEBUG:sequencer:Running task check_for_cores
0000882:DEBUG:sequencer:Running task notify
0000954:INFO:server:Semihost server started on port 4444
0000966:INFO:gdbserver:GDB server started on port 3333

flit, 2019-11-04:
`pyocd gdb` is working now. What changed? Maybe that you weren't running as root?

Also, you don't have the J-Link firmware installed on your FRDM-K64F, you have DAPLink installed. (Evident by the fact that it shows up at all with `pyocd list`, and the unique ID format.)

flit, 2019-11-16:
I don't see any issues here? The gdbserver is running succesfully.

Can this issue be closed?

flit, 2019-11-16:
@medjutim Can this issue be closed now?

closed 2025-08-20 by TeoMahnic