pyocd/pyOCD #755: ValueError: Attempted relative import in non-package

medjutim, 2019-11-07:
Hello,

When I try to debug pyocd in Pycharm I get this message. Can somebody tell me what is the problem? 

Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm 2019.2.4\helpers\pydev\pydevd.py", line 1415, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:/Users/Software/Downloads/pyOCD-master/pyocd/__main__.py", line 30, in <module>
    from . import __version__
ValueError: Attempted relative import in non-packa

flit, 2019-11-17:
This happens because Python is directly executing `__main__.py` (from within the `pyocd` package), so it doesn't see pyocd as a package.

I'm not familiar with PyCharm, but I can tell you how to debug pyocd using VSCode. From the log, it looks like PyCharm uses pydevd as its backend just like VSCode, so similar options should be available.

1. Create a virtualenv as described in the pyocd developer docs.
2. Set the python executable to the one in the venv.
3. Tell VSCode to debug the module `pyocd.__main__`, as opposed to the script `.../pyocd/__main__.py`. (As stated, I haven't a clue how you'd configure this in PyCharm, sorry.)

For instance, here is a VSCode launch config for `pyocd cmd`:

```json
        {
            "name": "pyocd cmd",
            "type": "python",
            "request": "launch",
            "module": "pyocd.__main__",
            "console": "integratedTerminal",
            "stopOnEntry": true,
            "pythonPath": "/Users/creed/projects/pyOCD-flit/venv/mac37/bin/python3.7",
            "showReturnValue": true,
            "args": ["cmd", "-v", "-u9900", "-tk20d50m"]
        },
```

Once you get it working, please report back to this issue with the details. I'd like to add some docs for both VSCode and PyCharm.

anthrax-0, 2019-11-18:
As a PyCharm user, I found that this helps: https://youtrack.jetbrains.com/issue/PY-2784#focus=streamItem-27-3064762.0-0
Just set module name to 'pyocd' and it works for me.

TeoMahnic, 2026-01-07:
Closing as completed, based on the comments above.