pyocd/pyOCD #758: Add support for PSoC 4 series

raveslave, 2019-11-07:
@Cypress-OpenOCD @ihos-cypress
Saw the psoc 6 contribution. Would it be possible to also get PSoC 4xxx series added.

Thanks

40Grit, 2019-11-07:
I believe the decision is on cypress.
I'm currently trying to find a way to convince them to add Mbed OS support for PSOC4XXX as well.

jsiverskog, 2019-11-07:
+1 on this!

ifyall, 2019-11-08:
@raveslave , Is this a request for programming support only, or are you also asking for targets in Mbed OS for the PSoC 4 kits?

Thank you in advance,

Ian

raveslave, 2019-11-08:
programming only

cederom, 2019-11-08:
+1 :-) i know folks creating lots of nice products on PSoC4 that could shift to MBED if the target had production grade implementation :-)

loverdeg-ep, 2019-11-26:
@and anything running mbed is one step closer to running Pelion device client. Even if it is an M0+.

raveslave, 2020-10-06:
anyone from cypress/infineon here willing to do it?

raveslave, 2020-10-06:
cc: @Cypress-OpenOCD

40Grit, 2020-10-07:
Looking back at some of these parts, they are seriously RAM/FLASH limited.

Last I bothered cypress/infineon about this they were not sure if it was worth the effort.

Mbed had made decent progress in terms of resource consumption though.

Id still be an advocate for this.

40Grit, 2020-10-07:
Then again this topic was for pyocd support not mbed support so resource consumption is not relevant.

raveslave, 2024-02-28:
hola cypress -> infineon, any chance?