pyocd/pyOCD #760: SWD/JTAG Transfer Fault

Quincy-Zh, 2019-11-14:
> Probe: LPC-LINK2 CMSIS-DAP
> PyOCD Version: 0.22.0
> Target: LPC55s69

When start gdb-server, pyocd reported error. I killed process and start again, it would be OK.
I can reproduce the problem by steps: start -> Error occurred -> Restart -> OK -> Restart -> Error occurred -> Restart -> OK ...

I use other CMSIS-DAP probe, the problem occurred too.

Log:
```
C:\Python27\Scripts\pyocd gdb -t lpc55s69 -f 40000000 --persist
0000437:WARNING:common:STLink and CMSIS-DAPv2 probes are not supported because no libusb library was found.
0004373:INFO:board:Target type is lpc55s69
0004404:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0004419:INFO:ap:AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
0004451:INFO:ap:AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0004467:INFO:ap:AP#2 IDR = 0x002a0000 (proprietary)
0004467:INFO:rom_table:AP#0 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0004483:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0004483:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0004483:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0004483:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0004483:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0004513:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0004529:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=0:0:ca0>
0004529:INFO:rom_table:AP#1 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0004544:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0004544:INFO:rom_table:  AP#1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0004544:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0004544:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0004560:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0004560:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0004560:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=0:0:ca0>
0004576:INFO:cortex_m_v8m:CPU core #0 is Cortex-M33 r0p3 (security ext present)
0004576:INFO:cortex_m:FPU present: FPv5-SP
0004608:INFO:cortex_m_v8m:CPU core #1 is Cortex-M33 r0p3
0004622:INFO:dwt:4 hardware watchpoints
0004638:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0004654:INFO:dwt:4 hardware watchpoints
0004654:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0004701:INFO:server:Semihost server started on port 4444
0004701:INFO:gdbserver:GDB server started on port 3333
0004717:INFO:server:Semihost server started on port 4445
0004717:INFO:gdbserver:GDB server started on port 3334
0013952:INFO:gdbserver:One client connected!
[                    ]   0%0014220:ERROR:gdbserver:Unhandled exception in handle_message: SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 549, in handle_message
    reply = handler(msg[msgStart:])
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 768, in v_command
    return self.flash_op(data)
  File "c:\python27\lib\site-packages\pyocd\gdbserver\gdbserver.py", line 884, in flash_op
    self.flash_loader.commit()
  File "c:\python27\lib\site-packages\pyocd\flash\loader.py", line 508, in commit
    keep_unwritten=self._keep_unwritten)
  File "c:\python27\lib\site-packages\pyocd\flash\flash_builder.py", line 466, in program
    flash_operation = self._sector_erase_program_double_buffer(progress_cb)
  File "c:\python27\lib\site-packages\pyocd\flash\flash_builder.py", line 859, in _sector_erase_program_double_buffer
    self.flash.init(self.flash.Operation.ERASE)
  File "c:\python27\lib\site-packages\pyocd\flash\flash.py", line 235, in init
    self.target.reset_and_halt(Target.ResetType.SW)
  File "c:\python27\lib\site-packages\pyocd\core\coresight_target.py", line 333, in reset_and_halt
    return self.selected_core.reset_and_halt(reset_type)
  File "c:\python27\lib\site-packages\pyocd\target\builtin\target_LPC55S69JBD100.py", line 257, in reset_and_halt
    if (self.read32(base + FLASH_INT_STATUS) & 0x00000004) != 0:
  File "c:\python27\lib\site-packages\pyocd\core\memory_interface.py", line 56, in read32
    return self.read_memory(addr, 32, now)
  File "c:\python27\lib\site-packages\pyocd\coresight\cortex_m.py", line 614, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 564, in _read_memory
    result = read_mem_cb()
  File "c:\python27\lib\site-packages\pyocd\coresight\ap.py", line 546, in read_mem_cb
    res = result_cb()
  File "c:\python27\lib\site-packages\pyocd\coresight\dap.py", line 276, in read_ap_cb
    result = result_cb()
  File "c:\python27\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 288, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "c:\python27\lib\site-packages\six.py", line 737, in raise_from
    raise value
TransferFaultError: SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
0014246:INFO:gdbserver:Client detached
0014247:INFO:gdbserver:Client disconnected!

```

flit, 2019-11-17:
Thanks for the bug report! There are some known issues with programming LPC55S69 flash, see #693 and #715. Your log adds some valuable information that should help fix the bugs.

thegecko, 2019-12-17:
I've just come across this issue when simply flashing the LPC55S69 using PyOCD

PyOCD: 0.23.0
Host: Mac OS 10.14.5
Target: LPC55S69
Pack: https://mcuxpresso.nxp.com/cmsis_pack/repo/NXP.LPC55S69_DFP.12.0.1.pack

It seems to flip-flop between a successful flash and then one which fails. It looks like the failed flash puts the board in a halted state which then allows the board to be flashed properly. It then resets and the cycle starts again.

Tested with native PyOCD support and by specifying the latest cmsis-pack and see the same problem:

```bash
Robs-iMac:bin thegecko$ ./pyocd flash --uid 023600001985bd0100000000000000000000000097969905 --target lpc55s69 --erase=chip --pack ~/Desktop/NXP.LPC55S69_DFP.12.0.1.pack /Users/User/mbed-os-example-blinky/BUILD/LPC55S69_NS/ARMC6/mbed-os-example-blinky.hex
[                    ]   0%0002378:CRITICAL:__main__:SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
Traceback (most recent call last):
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/__main__.py", line 480, in do_flash
    file_format=self._args.format)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/loader.py", line 160, in program
    self._loader.commit()
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/loader.py", line 508, in commit
    keep_unwritten=self._keep_unwritten)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/flash_builder.py", line 460, in program
    flash_operation = self._chip_erase_program_double_buffer(progress_cb)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/flash_builder.py", line 701, in _chip_erase_program_double_buffer
    self.flash.init(self.flash.Operation.ERASE)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/flash.py", line 235, in init
    self.target.reset_and_halt(Target.ResetType.SW)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 389, in reset_and_halt
    return self.selected_core.reset_and_halt(reset_type)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/target/builtin/target_LPC55S69JBD100.py", line 257, in reset_and_halt
    if (self.read32(base + FLASH_INT_STATUS) & 0x00000004) != 0:
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/core/memory_interface.py", line 56, in read32
    return self.read_memory(addr, 32, now)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/coresight/cortex_m.py", line 617, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 564, in _read_memory
    result = read_mem_cb()
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/coresight/ap.py", line 546, in read_mem_cb
    res = result_cb()
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 276, in read_ap_cb
    result = result_cb()
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 288, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/Users/User/python/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferFaultError: SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
Robs-iMac:bin thegecko$ ./pyocd flash --uid 023600001985bd0100000000000000000000000097969905 --target lpc55s69 --erase=chip --pack ~/Desktop/NXP.LPC55S69_DFP.12.0.1.pack /Users/User/mbed-os-example-blinky/BUILD/LPC55S69_NS/ARMC6/mbed-os-example-blinky.hex
[====================] 100%
0015417:INFO:loader:Erased chip, programmed 173056 bytes (338 pages), skipped 32256 bytes (63 pages) at 15.21 kB/s
Robs-iMac:bin thegecko$ ./pyocd flash --uid 023600001985bd0100000000000000000000000097969905 --target lpc55s69 --erase=chip --pack ~/Desktop/NXP.LPC55S69_DFP.12.0.1.pack /Users/User/mbed-os-example-blinky/BUILD/LPC55S69_NS/ARMC6/mbed-os-example-blinky.hex
[                    ]   0%0002331:CRITICAL:__main__:SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
Traceback (most recent call last):
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/__main__.py", line 480, in do_flash
    file_format=self._args.format)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/loader.py", line 160, in program
    self._loader.commit()
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/loader.py", line 508, in commit
    keep_unwritten=self._keep_unwritten)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/flash_builder.py", line 460, in program
    flash_operation = self._chip_erase_program_double_buffer(progress_cb)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/flash_builder.py", line 701, in _chip_erase_program_double_buffer
    self.flash.init(self.flash.Operation.ERASE)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/flash/flash.py", line 235, in init
    self.target.reset_and_halt(Target.ResetType.SW)
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/core/coresight_target.py", line 389, in reset_and_halt
    return self.selected_core.reset_and_halt(reset_type)
[... cut ...]
    res = result_cb()
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/coresight/dap.py", line 276, in read_ap_cb
    result = result_cb()
  File "/Users/User/python/lib/python2.7/site-packages/pyocd/probe/cmsis_dap_probe.py", line 288, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "/Users/User/python/lib/python2.7/site-packages/six.py", line 737, in raise_from
    raise value
TransferFaultError: SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3

```

korjaa, 2020-01-03:
I'm having similar issues. Sometimes it succeeds, sometimes it fails. If it fails, it fails nearly immediately and never gets to the "download" phase.
```
$ mbed_flash_pyocd_auto -t lpc55s69
[                    ]   0%0000841:CRITICAL:__main__:SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
Traceback (most recent call last):
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 294, in read_ap_result_callback
    return result()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 810, in read_reg_cb
    res = transfer.get_result()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 141, in get_result
    self.daplink.flush()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 665, in flush
    self._read_packet()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 894, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 449, in decode_data
    data = self._decode_transfer_data(data)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 356, in _decode_transfer_data
    self._check_response(data[2])
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 335, in _check_response
    raise DAPAccessIntf.TransferFaultError()
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferFaultError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/__main__.py", line 344, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/__main__.py", line 481, in do_flash
    file_format=self._args.format)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/flash/file_programmer.py", line 157, in program
    self._loader.commit()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/flash/loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/flash/builder.py", line 466, in program
    flash_operation = self._sector_erase_program_double_buffer(progress_cb)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/flash/builder.py", line 859, in _sector_erase_program_double_buffer
    self.flash.init(self.flash.Operation.ERASE)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/flash/flash.py", line 235, in init
    self.target.reset_and_halt(Target.ResetType.SW)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/core/coresight_target.py", line 387, in reset_and_halt
    return self.selected_core.reset_and_halt(reset_type)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/target/builtin/target_LPC55S69JBD100.py", line 257, in reset_and_halt
    if (self.read32(base + FLASH_INT_STATUS) & 0x00000004) != 0:
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/core/memory_interface.py", line 56, in read32
    return self.read_memory(addr, 32, now)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/coresight/cortex_m.py", line 638, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 142, in _locking
    return func(self, *args, **kwargs)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 583, in _read_memory
    result = read_mem_cb()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/coresight/ap.py", line 565, in read_mem_cb
    res = result_cb()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 412, in read_ap_cb
    result = result_cb()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 297, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferFaultError: SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
```

korjaa, 2020-01-03:
I just encountered similar error message when I ran gdbserver with pyOCD and then issued `monitor reset halt` from gdb.

```
$ pyocd gdbserver -t lpc55s69
0000226:INFO:board:Target type is lpc55s69
0000288:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0000308:INFO:ap:AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
0000329:INFO:ap:AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0000342:INFO:ap:AP#2 IDR = 0x002a0000 (proprietary)
0000375:INFO:rom_table:AP#0 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0000381:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0000381:INFO:rom_table:  AP#0 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000388:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000407:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000439:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000451:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000463:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=0:0:ca0>
0000514:INFO:rom_table:AP#1 ROM table #0 @ 0xe00fe000 (designer=015 part=000)
0000527:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0000527:INFO:rom_table:  AP#1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000531:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000533:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000535:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000537:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000539:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=0:0:ca0>
0000544:INFO:cortex_m_v8m:CPU core #0 is Cortex-M33 r0p3 (security ext present)
0000552:INFO:cortex_m:FPU present: FPv5-SP
0000557:INFO:cortex_m_v8m:CPU core #1 is Cortex-M33 r0p3
0000569:INFO:dwt:4 hardware watchpoints
0000571:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0000581:INFO:dwt:4 hardware watchpoints
0000583:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0000598:INFO:server:Semihost server started on port 4444
0000598:INFO:gdbserver:GDB server started on port 3333
0000605:INFO:server:Semihost server started on port 4445
0000605:INFO:gdbserver:GDB server started on port 3334
0009359:INFO:gdbserver:One client connected!
0009510:INFO:gdbserver:Attempting to load Argon
0009511:INFO:gdbserver:Attempting to load FreeRTOS
0009512:INFO:gdbserver:Attempting to load Zephyr
0009513:INFO:gdbserver:Attempting to load RTX5
0009514:INFO:gdbserver:RTX5 loaded successfully
0012243:ERROR:gdbserver:Unhandled exception in handle_message: SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
Traceback (most recent call last):
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 294, in read_ap_result_callback
    return result()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 810, in read_reg_cb
    res = transfer.get_result()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 141, in get_result
    self.daplink.flush()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 665, in flush
    self._read_packet()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 894, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 449, in decode_data
    data = self._decode_transfer_data(data)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 356, in _decode_transfer_data
    self._check_response(data[2])
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 335, in _check_response
    raise DAPAccessIntf.TransferFaultError()
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferFaultError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/gdbserver/gdbserver.py", line 379, in handle_message
    reply = handler(msg[msgStart:])
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/gdbserver/gdbserver.py", line 887, in handle_query
    return self.handle_remote_command(cmd)
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/gdbserver/gdbserver.py", line 1006, in handle_remote_command
    self.target.reset_and_halt()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/target/builtin/target_LPC55S69JBD100.py", line 257, in reset_and_halt
    if (self.read32(base + FLASH_INT_STATUS) & 0x00000004) != 0:
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/core/memory_interface.py", line 56, in read32
[... cut ...]
    res = result_cb()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/coresight/dap.py", line 412, in read_ap_cb
    result = result_cb()
  File "/home/jaakor01/workspace/pyOCD/venv3/lib/python3.6/site-packages/pyocd/probe/cmsis_dap_probe.py", line 297, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferFaultError: SWD/JTAG Transfer Fault @ 0x40034fe0-0x40034fe3
```

JanneKiiskila, 2020-04-28:
@mmahadevan108  @maclobdell  - this one, too. They all might be interlinked.