pyocd/pyOCD #768: Need support for dual-core targets

MarceloSalazar, 2019-11-27:
After investigating how to connect to individual targets (see https://github.com/mbedmicro/pyOCD/issues/719), realized that need support in pyOCD to connect to dual-core targets and handle the debug operations correctly (halt and resume both cores, read/write registers, etc)

flit, 2019-11-27:
pyOCD does support multicore targets. See the [multicore_debug.md](https://github.com/mbedmicro/pyOCD/blob/master/docs/multicore_debug.md) documentation.

closed 2019-11-27 by flit

flit, 2019-11-28:
If you encounter specific issues with multicore debug, of course we'll fix them.

Unfortunately, gdb simply doesn't support heterogenous multicore in a single instance. (It does support SMP in a single instance just fine.) Early on I experimented with reporting each core to gdb as separate threads, then tried separate processes. There issues with both options that couldn't be solved, so I went with the same model that everyone else uses (i.e., NXP MCUXpresso IDE)—separate gdb instances for each core.