pyocd/pyOCD #769: ROMTable: default address is set to None in ap.py

0xc0170, 2019-11-29:
The ap.py in coresight implementation contains this:

```
    @_locked
    def init_rom_table(self):
        try:
            if self.has_rom_table:
                self.rom_table = ROMTable(self)
                self.rom_table.init()
```

if there is rom table, it initializes ROMTable but without address. Address is set to None by default,  this causes a traceback as later in the `init()` function, there is arithmetic done with address. Why default address is not set?

From what I read, Cortex M7 has this at `0xe00ff000` so this core should provide this address. Not certain about others, would need to check.

Once I set address for a target (in my case M7), gdbserver started working.

0xc0170, 2019-11-29:
`In typical Cortex-M3 or Cortex-M4 systems, the ROM table is located in address 0xE00FF000`. 

Would it make sense instead of `addr=None`, use this default value ? Or how can we fix this one?

cc @flit

flit, 2019-11-29:
A default address is not set because there is not such a thing. The ROM table base address is either specified by the AP or there is no ROM table.

The question is, why does it think there is a ROM table but no base address?

Try out my [feature/aps](https://github.com/flit/pyOCD/tree/feature/aps) branch. It has some fixes that may be relevant.

0xc0170, 2019-12-02:
> Try out my feature/aps branch. It has some fixes that may be relevant.

No change to this issue at least to get ROMTable properly.

> The question is, why does it think there is a ROM table but fails(?) to extract the base address?

ROM table address is in our case `0` that is not valid for that target.
. 
OpenOCD provides `dbgbase` to have this customized for each core. I saw a configuration for a board I am using `-dbgbase 0xE00FE000`.  This is the address where the ROM table is. I'll find out why `self.read_reg(AP_BASE)` does not return this one.

How address is then passed to ROMTable?

pyOCD reads it: `self.rom_addr = self.read_reg(AP_BASE)` and validates this read (has it rom table member). However, I don't see the member `rom_addr` used anywhere in `ap.py` or `rom_table.py`.

I would expect to be passed here (ap.py file):

```
    @_locked
    def init_rom_table(self):
        try:
            if self.has_rom_table:
                self.rom_table = ROMTable(self)
                self.rom_table.init()
```

If `has_rom_table` is valid, why ROMTable is not getting the address ? I can't find the reference anywhere.

0xc0170, 2019-12-02:
I'll close this as ROMTable code looks like it should be getting the address , will reopen with more findings if any

flit, 2019-12-02:
Got it

flit, 2019-12-03:
Fixed in flit/pyOCD@4401798.