pyocd/pyOCD #770: PyOCD seems to fail with Python3.8

kristofmulier, 2019-11-29:
**1. My system**
I'm working on a 64-bit Windows 10 desktop. My Python installation is the latest Python3.8.

**2. PyOCD installation**
I followed the instructions to install **PyOCD** meticulously. I cloned the source from Git and issued the command:
```shell
$ python setup.py install
```
in the cloned PyOCD folder. The installation procedure started, and things looked okay. Next, I downloaded the right **libusb** dll and put it next to `python.exe`.

**3. First run**
I tried to run **PyOCD** with the following command in my Windows console:
```shell
$ pyocd gdbserver
```
Unfortunately, I get the following error:

```sh
Traceback (most recent call last):
  File "C:\Python38\Scripts\pyocd-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "C:\Python38\lib\site-packages\pkg_resources\__init__.py", line 3251, in <module>
    def _initialize_master_working_set():
  File "C:\Python38\lib\site-packages\pkg_resources\__init__.py", line 3234, in _call_aside
    f(*args, **kwargs)
  File "C:\Python38\lib\site-packages\pkg_resources\__init__.py", line 3263, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\Python38\lib\site-packages\pkg_resources\__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "C:\Python38\lib\site-packages\pkg_resources\__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\Python38\lib\site-packages\pkg_resources\__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'cffi>=1.6.0' distribution was not found and is required by milksnake
```

**4. Install `cffi`**
I run the following command:
```sh
pip install cffi
```
with the following result:
```sh
[...]

Successfully installed cffi-1.13.2 pycparser-2.19
```

Now I can successfully run the command:
```sh
pyocd gdbserver --help
```

**5. Conclusion**
It's an easy-to-solve bug. One should simply issue the command `pip install cffi`. Nevertheless, getting an error message on the first run of PyOCD as a newcomer is a bit intimidating - because the user expects it to "just work". I think it's good to solve this problem :-)

flit, 2019-11-29:
Interesting, thanks for the report! Very strange. Looks like cffi is a requirement from pyocd–>cmsis-pack-manager–>milksnake–>cffi. So I don't understand why it wouldn't be installed… the dependencies are all there. I wonder if it's Windows-only? I've been using Python 3.8 on macOS for over a month without issues.

I'll try a fresh virtualenv on Windows and report back.

kristofmulier, 2019-11-29:
Thank you @flit

0xc0170, 2019-12-02:
I've just installed on Friday latest master with py 3.7 to fresh new virtualenv, no problem on windows 10 64bit.

kristofmulier, 2019-12-02:
Hi @0xc0170 ,
Thanks for your reply. I did my installation on a native Windows 10 machine (no virtual machine) and I had **Python 3.8** running (not **Python 3.7**). I don't know if that could explain anything.

kristofmulier, 2019-12-02:
Hi @0xc0170 and @flit ,
I know this is not the place to ask for this kind of help... but I've got no other choice. All questions I ask on StackOverflow related to either **PyOCD** or **OpenOCD** get flagged as "off-topic" and closed down. Take for example this one:
https://stackoverflow.com/questions/59118658/pyocd-doesnt-find-nu-link2-me-probe-on-my-numaker-board

I checked out the people who voted to close the question. Four out of five are Javascript and PHP experts. They don't know what embedded software is about.

Can you please vote to "reopen" this question? For me it's a matter of principles: **PyOCD** questions are not "general computing hardware" questions. There are tons of other questions on StackOverflow related to embedded software. Those embedded software questions are accepted. So questions about **PyOCD** should get accepted as well.

Again my apologies I'm posting this here on GitHub - not really the proper place. But I don't know another way to reach out to you for help.

cederom, 2019-12-02:
Hey @kristofmulier this GitHub issues system is probably the best place to work on pyOCD and ask questions, not really the StackOverflow as there are people that does not even understand the question and will report/block your issue..

You got my vote, plus redirection from OpenOCD to pyOCD platform as a solution. You are welcome to ask pyOCD questions here, preferably post your own fixes :-)

kristofmulier, 2019-12-02:
Thank you @cederom  :-)

TeoMahnic, 2025-08-20:
Closing as completed. Installation instructions have been updated in recent releases.

closed 2025-08-20 by TeoMahnic