pyocd/pyOCD #772: Python37 64-bits and STM32WB55 :STLink error (9): Get IDCODE error

pawelm87, 2019-12-02:
Hi all,

I have got problem during debug nucleo with STM32WB55 chip. In console after run command I got error:
```
PS C:\Users\pawel> C:\Users\pawel\AppData\Local\Programs\Python\Python37\Scripts\pyocd.exe gdbserver
0001639:INFO:board:Target type is stm32wb55rgvx
0002270:CRITICAL:__main__:STLink error (9): Get IDCODE error
Traceback (most recent call last):
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\__main__.py", line 571, in do_gdbserver
    with session:
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\core\coresight_target.py", line 164, in init
    seq.invoke()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\coresight\dap.py", line 85, in init
    self.link.connect()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\probe\stlink_probe.py", line 118, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\probe\stlink\stlink.py", line 256, in enter_debug
    self._check_status(response)
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\probe\stlink\stlink.py", line 294, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (9): Get IDCODE error

PS C:\Users\pawel> C:\Users\pawel\AppData\Local\Programs\Python\Python37\Scripts\pyocd-gdbserver.exe
0000379:WARNING:gdb_server:pyocd-gdbserver is deprecated; please use the new combined pyocd tool.
Waiting for a debug probe to be connected...
0015925:INFO:board:Target type is stm32wb55rgvx
0016545:ERROR:gdb_server:uncaught exception: STLink error (9): Get IDCODE error
Traceback (most recent call last):
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\tools\gdb_server.py", line 287, in run
    with session:
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\core\session.py", line 287, in __enter__
    self.open()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\core\session.py", line 367, in open
    self._board.init()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\board\board.py", line 83, in init
    self.target.init()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\core\coresight_target.py", line 164, in init
    seq.invoke()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\coresight\dap.py", line 85, in init
    self.link.connect()
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\probe\stlink_probe.py", line 118, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\probe\stlink\stlink.py", line 256, in enter_debug
    self._check_status(response)
  File "c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages\pyocd\probe\stlink\stlink.py", line 294, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (9): Get IDCODE error
```

I used STLink in nucleo, I upgraded it to the newest version of firmware:
[image]

Similar error occurred at Visual Studio Code when I start debug mbed program:
[image]

Version of dependencies:
```
PS C:\Users\pawel> pip3 install -U pyocd
Requirement already up-to-date: pyocd in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (0.23.0)
Requirement already satisfied, skipping upgrade: pyelftools in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from pyocd) (0.25)
Requirement already satisfied, skipping upgrade: pyusb<2.0,>=1.0.0b2 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from pyocd) (1.0.2)
Requirement already satisfied, skipping upgrade: intelhex<3.0,>=2.0 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from pyocd) (2.2.1)
Requirement already satisfied, skipping upgrade: colorama in c:\users\pawel\appdata\roaming\python\python37\site-packages (from pyocd) (0.4.1)
Requirement already satisfied, skipping upgrade: prettytable in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from pyocd) (0.7.2)
Requirement already satisfied, skipping upgrade: pyyaml<6.0,>=5.1 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from pyocd) (5.1.2)
Requirement already satisfied, skipping upgrade: pywinusb>=0.4.0; platform_system == "Windows" in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from pyocd) (0.4.2)
Requirement already satisfied, skipping upgrade: six<2.0,>=1.0 in c:\users\pawel\appdata\roaming\python\python37\site-packages (from pyocd) (1.12.0)
Requirement already satisfied, skipping upgrade: intervaltree<4.0,>=3.0.2 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from pyocd) (3.0.2)
Requirement already satisfied, skipping upgrade: cmsis-pack-manager>=0.2.7 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from pyocd) (0.2.9)
Requirement already satisfied, skipping upgrade: sortedcontainers<3.0,>=2.0 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from intervaltree<4.0,>=3.0.2->pyocd) (2.1.0)
Requirement already satisfied, skipping upgrade: appdirs>=1.4 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from cmsis-pack-manager>=0.2.7->pyocd) (1.4.3)
Requirement already satisfied, skipping upgrade: milksnake>=0.1.2 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from cmsis-pack-manager>=0.2.7->pyocd) (0.1.5)
Requirement already satisfied, skipping upgrade: cffi>=1.6.0 in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from milksnake>=0.1.2->cmsis-pack-manager>=0.2.7->pyocd) (1.13.2)
Requirement already satisfied, skipping upgrade: pycparser in c:\users\pawel\appdata\local\programs\python\python37\lib\site-packages (from cffi>=1.6.0->milksnake>=0.1.2->cmsis-pack-manager>=0.2.7->pyocd) (2.19)
You are using pip version 19.0.3, however version 19.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```

Someone have any idea what is wrong?

flit, 2019-12-24:
cc @schstm @jeromecoutant 

Sorry for the slow reply. Usually a failure to read IDCODE means the target device is not powered on or is in a deep sleep mode. Please make sure you are using a debug build of your application that does not put the device to sleep.

flit, 2020-01-10:
Please try setting the `connect_mode` option to `under-reset`. This will connect to the target while the hardware reset is held asserted. You can set this option by adding `-Oconnect_mode=under-reset` to the command line, or put it in a [config file](https://github.com/mbedmicro/pyOCD/blob/master/docs/configuration.md). Of course, you cannot use this option is you want to attach to the running system.

I also recommend that you use the combined `pyocd` tool instead of the (deprecated) `pyocd-gdbserver`. The basic functionality isn't any different, but `pyocd-gdbserver` will be dropped in version 1.0.

(Thanks to @schstm for this suggestion!)

flit, 2020-03-01:
@pawelm87 Were you able to connect by setting `connect_mode`? Can this issue be closed?