pyocd/pyOCD #774: Mem-AP: add option to provide base address

0xc0170, 2019-12-03:
I've received MCU that has invalid base address stored in BASE register. To be specific, I created yesterday an issue where address read is 0 and would traceback.  BASE register reading returns invalid address, it is 0. But MCU has it at

0xc0170, 2019-12-03:
Looking at the `CoreSightTarget` class, it contains list of aps (their numbers). As it is list, would it make sense to transform to an object that can hold more than just a number of AP? Each AP could have a debug base address defined.

flit, 2019-12-03:
Related to #769.

flit, 2019-12-03:
For the short term, it's not too hard to use a customized init sequence to accomplish this task. (That's one major reason why the named init sequences exist.)

Basically, in the target class (subclass of `CoreSightTarget`) you need to modify the standard init sequence to insert an init task before the `find_components` task that fixes the `rom_addr` attribute of each AP. At the point this new task runs, the APs will have been identified and instantiated, and their `init()` methods called. But the ROM tables have not been read yet, so you can correct the ROM table base address.

```py
    def create_init_sequence(self):
        seq = super(xxx, self).create_init_sequence()
        seq.insert_before('find_components', self._fixup_ap_base_addrs)
        return seq

    def _fixup_ap_base_addrs(self):
        self.dp.aps[0].base_addr = 0xe00fe000
        # fix other APs here…
```

Of course, this isn't tested, but it _should_ work. 😄 

See a class such as `pyocd.target.builtin.target_CY8C64xx.cy8c64xx` for a more complete example of a target class that overrides `create_init_sequence()` to modify the init tasks. This class uses a replacement `find_aps` task to set the DP's `valid_aps` list to a fixed selection of APs. (And to create a custom `CortexM` subclass for each core.)

0xc0170, 2019-12-04:
Yes, it works 👍  Thanks, will close this as resolved.

For anyone who has a similar issue, I'll paste here the code for reference:

```
    def create_init_sequence(self):
        seq = super(TargetNameClass, self).create_init_sequence()
        seq.replace_task('find_aps', self.find_aps)
        # after creating ap, we fix rom addr
        seq.insert_before('init_ap_roms',
            ('fixup_ap_base_addrs', self._fixup_ap_base_addrs),
            )
        return seq

    def _fixup_ap_base_addrs(self):
        self.dp.aps[self.AP_NUM].rom_addr = 0xe00fe000

    def find_aps(self):
        if self.dp.valid_aps is not None:
            return

        self.dp.valid_aps = (self.AP_NUM,)
```

closed 2019-12-04 by 0xc0170