pyocd/pyOCD #775: nRF52 APPROTECT, Cannot read_ap *or* write_ap with recent ST-Link firmwares (J33,J34,J35)

nocko, 2019-12-04:
I am enabling APPROTECT in a production programming setup using pyOCD:

``` python
from pyocd.core.helpers import ConnectHelper
from pyocd.core.memory_map import FlashRegion
from pyocd.flash.flash import Flash

with ConnectHelper.session_with_chosen_probe(target_override="nrf52") as session:
    target = session.board.target
    target.reset_and_halt()
    fl = FlashLoader(session)
    mmap = target.memory_map

    uicr2 = FlashRegion(start=0x10001200, end=0x1000120f, blocksize=0x10, is_testable=False)
    uicr2.flash = Flash(target, mmap.get_boot_memory().algo)
    uicr2.flash.region = uicr2
    mmap.add_region(uicr2)
    fl.add_data(APPROTECT_REG, b'\x00')
    fl.commit()
```
It's a little dance to dynamically add the *second page* (?) of the UICR to the memory map. I wasn't sure how to upstream this (poorly documented) memory region... it's not clear if it can be erased or why the address gap with the rest of the UICR... &c. This hack works well enough for a write once operation.

I also added a way to check if the weable is locked out:
``` python
def wearable_locked():
    session = start_uninit_session()
    target = session.board.target
    locked = not target.dp.read_ap(1 << 24 | 0xc)
    session.close()
    return locked
```
and several functions that should work while the wearable is locked (via CTRL-AP @1):
``` python
def recover():
    session = start_uninit_session()
    target = session.board.target
    target.dp.init()
    target.dp.power_up_debug()
    target.dp.write_ap(1 << 24 | 4, 1)
    while target.dp.read_ap(1 << 24 | 8):
        sleep(0.1)
    _ctrl_ap_reset(session)
    session.close()

def _ctrl_ap_reset(session=None):
    if not session:
        session = start_uninit_session()
    target = session.board.target
    target.dp.write_ap(1 << 24 | 0, 1)
    sleep(0.001)
    target.dp.write_ap(1 << 24 | 0, 0)
```

This works as intended with ST-Link v2 firmware 2J29S7, but with later firmwares (including all of those still available for download from ST (2J33xx, 2J34xx, 2J35xx) have stopped working.

The ST-Link has started returning JTAG_NO_DEVICE_CONNECTED when running either JTAG_WRITE_DAP_REG *or* JTAG_READ_DAP_REG *even when the address is on another unlocked AP*.

I looked into the pyOCD probe driver, but the ST-Link protocol is very high level. I can't imagine there's a way to work around this. It's still submitting the correct commands to the probe... I think it's a regression in the firmware; but I don't know how to proceed (I'm not a big ST customer).

It's currently sufficient for us to remain at 2J29S7 for our v2 probes, but v3 probes are similarly broken and I'm unaware of a working firmware version for that hardware.

Please advise.

flit, 2019-12-04:
Interesting, I haven't noticed this yet. I'll try to reproduce as soon as I can. And I'd love to help you upstream this support once we get past this issue.

cc @schstm

nocko, 2019-12-04:
Thanks. I'm willing to help. 

Which parts did you want upstream, the second UICR area... or integrating nrf52 support into the lock / unlock stuff I've seen (but not used) in commander?

If it's the latter, it'll be tough to maintain ST-Link support unless you have an *in* to get the firmware fixed (I think... we'll see what you think after you reproduce it).

flit, 2019-12-04:
All of the above! 😄 

Keep in mind that STLink is not often used with nRF52 devices. For instance, there is a DAPLink build that runs on the SAM3U2C on the nRF52-DK. Or you can use any other CMSIS-DAP compatible probe. I'd also like to add J-Link support at some point.

nocko, 2019-12-04:
Yeah, we've been all J-Link for development, but using pyOCD as a library is a better production programming solution for us (than paying for the J-Link SDK). Segger intentionally cripples the scripting capability of JLinkEXE to upsell it...

When we started these projects (that use pyOCD) we had a bunch of ST-Link sitting around... so we tested it and used them. I've ordered a few LPC-Link2 probes for testing... that's probably our medium term solution for the ST-Link badness.

When the new probes arrive, I'll look at upstreaming the lock/unlock.

40Grit, 2019-12-04:
This PR looks really interesting. Would love to see it get in.

@Nocko
We built a standalone debug board using the k20 hic.

See: https://os.mbed.com/components/Flidor/

@trowbridgec @millerc-ep

schstm, 2019-12-09:
Hello,
The ST-Link is designed and tested only on STM8 and STM32 products. ST won't investigate any issue which may occur on other targets.

nocko, 2019-12-12:
After diving into the code for inspiration, it seems clear that the flash security check and unlock for Kinetis targets is most probably broken on ST-Link as well.