pyocd/pyOCD #782: `pyocd flash` gets an uncaught exception

bikeNomad, 2019-12-17:
Trying to use `pyocd flash` to program an STM32F072VB processor. `pyocd gdb` works with the same settings.
I've tried both python 2 and python 3.
`pyocd -V` reports "0.23.0"

`pyocd.yaml`:
```yaml
# global options
debug.log_flm_info: true
frequency: 1000000
persist: true
enable_semihosting: false
vector_catch: all
target_override: stm32f051
chip_erase: chip
``` 
```text
$ pyocd flash -v ui-build/UserInterface.elf
0000278:INFO:session:Using config settings for probe 0670FF505454887767051341
0000278:INFO:board:Target type is stm32f051
0000311:INFO:dap:DP IDR = 0x0bb11477 (v1 MINDP rev0)
0000432:INFO:ap:AP#0 IDR = 0x04770021 (AHB-AP var2 rev0)
0000473:INFO:rom_table:AP#0 ROM table #0 @ 0xe00ff000 (designer=020 part=440)
0000505:INFO:rom_table:[0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0000521:INFO:rom_table:[1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0000533:INFO:rom_table:[2]<e0002000:BPU class=14 designer=43b part=00b>
0000544:INFO:cortex_m:CPU core #0 is Cortex-M0 r0p0
0000570:INFO:dwt:2 hardware watchpoints
0000580:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0000615:CRITICAL:__main__:uncaught exception: local variable 'addr' referenced before assignment
Traceback (most recent call last):
  File "/home/ned/.local/lib/python2.7/site-packages/pyocd/__main__.py", line 343, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/ned/.local/lib/python2.7/site-packages/pyocd/__main__.py", line 480, in do_flash
    file_format=self._args.format)
  File "/home/ned/.local/lib/python2.7/site-packages/pyocd/flash/loader.py", line 159, in program
    self._format_handlers[file_format](file_obj, **kwargs)
  File "/home/ned/.local/lib/python2.7/site-packages/pyocd/flash/loader.py", line 212, in _program_elf
    LOG.debug("Skipping segment LMA:0x%08x, VMA:0x%08x, size %d", addr,
UnboundLocalError: local variable 'addr' referenced before assignment

```

bikeNomad, 2019-12-19:
I have also tried this with the correct target type (after loading the correct CMSIS pack) with the same result.

flit, 2019-12-19:
Hi @bikeNomad , this was thankfully already fixed on master in commit 149556e. Sorry for the inconvenience! I'll shortly release version 0.24.0 that includes this fix.

flit, 2019-12-24:
Closing with release of 0.24.0.

closed 2019-12-24 by flit