pyocd/pyOCD #801: pyOCD does not autodetect LPC55S69(?)

korjaa, 2020-01-03:
When I'm flashing LPC55S69 with pyOCD v0.24.1 I need to use extra argument `-t` for example `pyocd flash /path/to/binary.hex -t lpc55s69`. Otherwise pyOCD detects LPC55S69 as some generic target and fails to flash.

flit, 2020-01-03:
That's probably because you are using the default firmware for the on-board debug probe, which is LPC-LinkII. If you change the firmware to DAPLink it should auto-detect.

korjaa, 2020-01-07:
Thanks for the insight, I think that is the case. I did not yet update interface firmware to DAPLink as it seemed to be a bit more involved in this LPC55S69.

korjaa, 2020-01-07:
How hard would it be to make work automatically with LPC-LinkII?

flit, 2020-03-01:
It's not possible to autodetect the target with the LPC-LinkII, as the same firmware is used on many NXP boards.

flit, 2020-03-01:
Closing this issue as there is no bug, the need to specify the target is simply a property of the LPC-LinkII firmware.

closed 2020-03-01 by flit