pyocd/pyOCD #802: issue flashing lpc1768

djj198, 2020-01-05:


danielRep, 2020-01-20:
Hi,

I am experience a similar issue with a Musca B1 board. I detect three different behaviours which probably can be related with the same issue that @djj198  reported. One, the gdbserver is initiated and pyocd is connect but the flashing operation doesn't start (it seems stuck), with the DAP LED blinking indefinetly. Second, the debug process passes all stages (w/o any outputted error) but the core seems to be in a wrong state. Third, the pyocd effectively outputs the following error, which is quite similar to @djj198 post. 

My openocd version is 0.24.1-dev15. 

`Started by GNU MCU Eclipse
0001289:INFO:server:Semihost server started on port 4445
0001290:INFO:gdbserver:GDB server started on port 3334
Started by GNU MCU Eclipse
0002328:INFO:gdbserver:One client connected!
0004534:ERROR:gdbserver:Unhandled exception in handle_message: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/gdbserver/gdbserver.py", line 379, in handle_message
    reply = handler(msg[msgStart:])
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/gdbserver/gdbserver.py", line 887, in handle_query
    return self.handle_remote_command(cmd)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/gdbserver/gdbserver.py", line 1009, in handle_remote_command
    self.target.reset_and_halt()
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/coresight/cortex_m.py", line 966, in reset_and_halt
    xpsr = self.read_core_register('xpsr')
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/coresight/cortex_m.py", line 1033, in read_core_register
    regValue = self.read_core_register_raw(regIndex)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/coresight/cortex_m.py", line 1047, in read_core_register_raw
    vals = self.read_core_registers_raw([reg])
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/coresight/cortex_m.py", line 1107, in read_core_registers_raw
    assert dhcsr_val & CortexM.S_REGRDY
AssertionError
0004546:INFO:gdbserver:Attempting to load Argon
0004547:INFO:gdbserver:Attempting to load FreeRTOS
0004548:INFO:gdbserver:Attempting to load Zephyr
0004549:INFO:gdbserver:Attempting to load RTX5
0006579:ERROR:gdbserver:Unhandled exception in handle_message: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/gdbserver/gdbserver.py", line 379, in handle_message
    reply = handler(msg[msgStart:])
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/gdbserver/gdbserver.py", line 598, in v_command
    return self.flash_op(data)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/gdbserver/gdbserver.py", line 714, in flash_op
    self.flash_loader.commit()
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/flash/loader.py", line 168, in commit
    keep_unwritten=self._keep_unwritten)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/flash/builder.py", line 417, in program
    self._build_sectors_and_pages(keep_unwritten)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/flash/builder.py", line 315, in _build_sectors_and_pages
    fill_end_of_page_gap()
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/flash/builder.py", line 259, in fill_end_of_page_gap
    self._enable_read_access()
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/flash/builder.py", line 221, in _enable_read_access
    self.flash.init(self.flash.Operation.VERIFY)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/flash/flash.py", line 235, in init
    self.target.reset_and_halt(Target.ResetType.SW)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/core/coresight_target.py", line 387, in reset_and_halt
    return self.selected_core.reset_and_halt(reset_type)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/coresight/cortex_m.py", line 966, in reset_and_halt
    xpsr = self.read_core_register('xpsr')
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/coresight/cortex_m.py", line 1033, in read_core_register
    regValue = self.read_core_register_raw(regIndex)
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/coresight/cortex_m.py", line 1047, in read_core_register_raw
    vals = self.read_core_registers_raw([reg])
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.24.1.dev15-py3.6.egg/pyocd/coresight/cortex_m.py", line 1107, in read_core_registers_raw
    assert dhcsr_val & CortexM.S_REGRDY
AssertionError
0006605:INFO:gdbserver:Client detached
0006606:INFO:gdbserver:Client disconnected!
Started by GNU MCU Eclipse
`

flit, 2020-03-22:
The LPC1768 bug is fixed by #821, and was regression after renaming `CATCH_CORE_RESET`.

@danielRep The Musca-B1 issue is quite different. I'm copying your report to a new issue.

flit, 2020-03-22:
Musca-B1 issue moved to #841.

flit, 2020-03-23:
Merged #821, so should be fixed on master.

closed 2020-03-23 by flit