pyocd/pyOCD #805: Not detecting probe AK-CMSIS-DAP

tlongeri, 2020-01-08:
I recently bought myself a shiny new [AK-CMSIS-DAP](https://www.artekit.eu/doc/guides/ak-cmsis-dap) from Artekit Labs, but I can't get pyOCD to detect it - I get "Waiting for a debug probe to be connected..." / "No available debug probes are connected". However, I was able to use it without issues using OpenOCD, so I suspect pyOCD is at fault here. I am using the latest pyOCD (v0.24.1) on Ubuntu 18.04.

I traced the problem to the USB interface string (not to be confused with the product string) not containing CMSIS-DAP. This is used as a criteria for filtering in `pyusb_backend`, filtering out the AK-CMSIS-DAP.

According to the [CMSIS-DAP reference](https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__ConfigUSB__gr.html), this is only required for composite devices:

> The CMSIS-DAP Firmware can be also part of a USB composite device. If this case, the HID Interface String must contain the sub-string CMSIS-DAP.

Correct me if I'm wrong, as I am not familiar with the USB protocol, but I gathered that "composite" means having more than one interface. If that is so, then the AK-CMSIS-DAP is not a composite device and should not be filtered out based on its interface string.

Here is the output of `lsusb -v` for the device:

```
Bus 001 Device 004: ID 10c4:8b54 Cygnal Integrated Products, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x10c4 Cygnal Integrated Products, Inc.
  idProduct          0x8b54 
  bcdDevice            1.00
  iManufacturer           1 ARTEKIT LABS
  iProduct                2 AK-CMSIS-DAP
  iSerial                 3 AK0B5F65
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              4 HID
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
```

flit, 2020-02-21:
Thanks, I hadn't seen the Artekit probe previously.

flit, 2020-03-10:
@tlongeri Could you please try testing with pyocd 0.25.0? This includes a change to relax interface name requirements.

tlongeri, 2020-03-10:
I am able to see the probe using pyocd list and opening a GDB server with pyocd gdb seems to work. Thanks!

closed 2020-03-10 by tlongeri

flit, 2020-03-11:
Great, thanks for testing again!